esphome/issues #831: esphome won't start - missing argument

lwqcz, 2019-11-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 onboard Wemos D1 Mini

**Affected component:**

https://esphome.io/guides/getting_started_command_line.html

**Description of problem:**
I've followed [CLI getting started](https://esphome.io/guides/getting_started_command_line.html) procedure and everything was fine.
When I've tried to start a text based wizard of dashboard - only this error was thrown thru stderr:
```
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 7, in <module>
    from esphome.__main__ import main
  File "/usr/local/lib/python3.5/dist-packages/esphome/__main__.py", line 10, in <module>
    from esphome import const, writer, yaml_util
  File "/usr/local/lib/python3.5/dist-packages/esphome/writer.py", line 7, in <module>
    from esphome.config import iter_components
  File "/usr/local/lib/python3.5/dist-packages/esphome/config.py", line 15, in <module>
    from esphome import core, core_config, yaml_util
  File "/usr/local/lib/python3.5/dist-packages/esphome/core_config.py", line 5, in <module>
    import esphome.codegen as cg
  File "/usr/local/lib/python3.5/dist-packages/esphome/codegen.py", line 11, in <module>
    from esphome.cpp_generator import (  # noqa
  File "/usr/local/lib/python3.5/dist-packages/esphome/cpp_generator.py", line 23, in <module>
    Type[bool], Type[int], Type[float], List[Any]]
  File "/usr/lib/python3.5/typing.py", line 552, in __getitem__
    dict(self.__dict__), parameters, _root=True)
  File "/usr/lib/python3.5/typing.py", line 512, in __new__
    for t2 in all_params - {t1} if not isinstance(t2, TypeVar)):
  File "/usr/lib/python3.5/typing.py", line 512, in <genexpr>
    for t2 in all_params - {t1} if not isinstance(t2, TypeVar)):
  File "/usr/lib/python3.5/typing.py", line 1077, in __subclasscheck__
    if super().__subclasscheck__(cls):
  File "/usr/lib/python3.5/abc.py", line 225, in __subclasscheck__
    for scls in cls.__subclasses__():
TypeError: descriptor '__subclasses__' of 'type' object needs an argument
```

**Problem-relevant YAML-configuration entries:**
```yaml
# no applicable

```

**Logs (if applicable):**

```
# no applicable

```

**Additional information and things you've tried:**

* I assumed that my Python3 env is working as the `esptool.py` or other tools are working correctly.

OttoWinter, 2019-11-08:
ESPHome 1.14+ requires Python 3.6+ - you have 3.5 installed.

Seems to be an error in setup.py - it's accepting py3.5 where it should reject that python version.

Please upgrade your python install to 3.6 or higher.

lwqcz, 2019-11-08:
I see :-) after upgrade to python3.6 and setup of env, everything is working fine I assume :-) thank you ...

Anyway ... Where can I see this requirement from a common user perspective at https://esphome.io/guides/getting_started_command_line.html ?
I didn't notice even use of `pip3`, but `pip` instead ...

OttoWinter, 2019-11-09:
@lwqcz It's all in a transition phase right now.

in 1.13.6, only python 2 was really supported (py3 worked, but was experimental).

in 1.14.x, both py2/3 are supported - hence the use of `pip` in the docs.

in 1.15.x, only py3 will be supported, then the docs will be updated to reflect that.

lwqcz, 2019-11-11:
🆗 I understand now, thank you a bunch for your great work and besides, I'll contribute too - I'm interested in support for Melexis's IR sensors and others 😎

closed 2019-11-11 by lwqcz