esphome/issues #832: Connecting to Multiple Networks Password issue

dheath73, 2019-11-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio Addon

**ESP (ESP32/ESP8266, Board/Sonoff):**

Node MCU

**Affected component:**

https://esphome.io/components/wifi.html?highlight=network

**Description of problem:**

I think I have found a bug

I have this in my config

```
networks:
    - ssid: !secret wifi_ssid_gif
      password: !secret wifi_password_gif
      priority: 10
    - ssid: !secret wifi_ssid_bc
      password: !secret wifi_password_bc
      priority: 20
```

And when I validate it i get this message returned

```
wifi:
  networks:
  - ssid: !secret 'wifi_ssid_gif'
    password: !secret 'wifi_password_bc'
    priority: 10.0
  - ssid: !secret 'wifi_ssid_bc'
    password: !secret 'wifi_password_bc'
    priority: 20.0
```

As you can see it states that the same password secret is being used for both my ssid's, am I doing it wrong or is this a bug please?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2019-11-08:
Is the password the same for both? If so, this is correct.

The way `!secret` is printed when dumping the config is a bit complex. It basically looks through all values, and if the value matches a secret string, it's replaced by the secret name.

dheath73, 2019-11-08:
Yes it is. Ok Thank you for your help and explanation.

Regards

David

closed 2019-11-09 by OttoWinter