esphome/issues #833: Deep Sleep behaviour

spattinson, 2019-11-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Home Assistant on Hassbian, MQTT, ESP32 is on a remote location behind another NAT across internet
**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32 DevKitC
**Affected component:**
Deep Sleep

**Description of problem:**
This is probably not a bug, but a request for clarification to back up documentation. I have a bme280  to take readings every hour and send via MQTT, running on batteries deep sleeping between readings. The below configuration worked fine while testing on my own network. I have set a *run_duration* of 0s which I took it to mean that it would run through measurements once and then go to sleep. I now have my doubts. On the remote network it does not report any values via  MQTT but it connects to MQTT every hour approx(see log below). I suspect that due to the remote location the MQTT connection takes longer and deep sleep is entered before the readings can be sent. Is this what would be expected?

I could make the *run_duration* longer if my suspicions are true, but I want to minimise the wake time to preserve batteries. I feel that it would be better to use *on_...:* automation to enter deep sleep. I could set up an automation in Home Assistant to fire a deep sleep message via MQTT on receipt of the readings and then use *on_message* trigger to enter deep sleep. This would also give me the option to disable deep sleep for OTA by disabling the automation in HA. Is this what would be recommended?

The other point I would like to clarify is the relationship between *sleep_duration* and *update_interval*. As the deep sleep time seems to be fairly imprecise (see time stamps on logs below) I was wondering  if it would be best practice to set the *update_interval* to some time lower than *sleep_duration* so that the measurement would always fire on wakeup? To put the question another way does esphome store the last measurement time in non-volatile memory so that it knows when the next one is due on wakeup from deep sleep?

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: bedford
  platform: ESP32
  board: esp32dev

wifi:
  ssid: 'OpenWrt'
  password: '********'
  fast_connect: true

mqtt:
  broker: ********.co.uk
  port: 23046
  username: ********
  password: ********
  birth_message:
  will_message:

logger:
  level: INFO

ota:
  password: *******

i2c:
  sda: 16
  scl: 17

sensor:
  - platform: bme280
    temperature:
      name: "bford temp"
      oversampling: 2x
    pressure:
      name: "bford pres"
      oversampling: 2x
    humidity:
      name: "bford humi"
      oversampling: 2x
    address: 0x76
    update_interval: 60 min

deep_sleep:
  run_duration: 0s
  sleep_duration: 60min

```

**Logs (if applicable):**

```
[01:49:32][I][mqtt:202]: MQTT Connected!
[02:49:13][I][mqtt:202]: MQTT Connected!
[03:48:56][I][mqtt:202]: MQTT Connected!
[04:48:38][I][mqtt:202]: MQTT Connected!
[05:48:21][I][mqtt:202]: MQTT Connected!
[06:48:05][I][mqtt:202]: MQTT Connected!
[07:47:47][I][mqtt:202]: MQTT Connected!
[08:48:07][I][mqtt:202]: MQTT Connected!
[09:47:50][I][mqtt:202]: MQTT Connected!
[10:47:46][I][mqtt:202]: MQTT Connected!
[11:47:30][I][mqtt:202]: MQTT Connected!
[12:52:20][I][mqtt:202]: MQTT Connected!
[13:56:07][I][mqtt:202]: MQTT Connected!
[14:55:52][I][mqtt:202]: MQTT Connected!

```
**Additional information and things you've tried:**
At some point I added blank MQTT birth and will messages to prevent Home Assistant marking the sensor as unavailable and making the graphs ugly. I don't know if that has any bearing on the issue?

I am using a non standard MQTT port for obfuscation purposes, what I am  doing is not secure, my NAT router redirects port 23046 to 1883 on the MQTT broker.

This sensor is in the ceiling of one of my tenant's flat at a remote location to hopefully detect humidity changes to see if a fix to the roof has worked to prevent leaks, its been an ongoing problem for a long time, I have just had contractors do some work, so this is like an early warning in case there is still a problem. I have limited access and don't want to disturb them unnecessarily. I've already been in and out a few times due to router issues and poor wifi signal strength. This is the reason for asking these questions rather than testing via trial and error. Thanks in advance.

spattinson, 2019-11-08:
What I was thinking of trying is:

Take out the blank birth/will - HA continuously reports the last known temp with no birth/will so I will more easily know if its really working or not. I am exporting to influxdb now so I can view graphs via another method.

Set update_interval to 45mins so it will be due at an hour

Deep sleep via MQTT message, automation in HA on receipt of readings

Set run_time to 5mins just in case there's connectivity issues, then esp will go to sleep by itself, but it will still give me a window to do OTA.

spattinson, 2019-11-09:
I have used an esp32 cam module and a bmp280 to set up a test environment with shorter sleep time so I can observe it easier. I used the following config:
```esphome:
  name: bedev
  platform: ESP32
  board: esp32dev

wifi:
  ssid: 'home'
  password: '********'
  fast_connect: true

mqtt:
  broker: 192.168.1.122
  username: scott
  password: ********
  on_message:
    - topic: bedev/ota_mode
      payload: 'ON'
      then:
        - deep_sleep.prevent: deep_sleep_1
    - topic: bedev/sleep_mode
      payload: 'ON'
      then:
        - deep_sleep.enter: deep_sleep_1

logger:
#  level: INFO

ota:
  password: ********

i2c:
  sda: 14
  scl: 15

sensor:
  - platform: bmp280
    temperature:
      name: "bedev_temp"
      oversampling: 2x
    pressure:
      name: "bedev_pres"
      oversampling: 2x
    address: 0x76
    update_interval: 45min

deep_sleep:
  id: deep_sleep_1
  run_duration: 60s
  sleep_duration: 60s
```
Added this to my automations.yaml:
```
- id: bedev_sleep
  alias: Send to sleep after mqtt receipt
  trigger:
    platform: mqtt
    topic: bedev/sensor/bedev_pres/state
  action:
    service: mqtt.publish
    data:
      topic: bedev/sleep_mode
      payload: 'ON'
```
It works how I expect and the esp32 goes to sleep immediately after it has sent the readings, the pressure reading is sent last so I used that to trigger sleep. 

In the above config I have run duration set to 60 secs, I will probably shorten that, so if there are network issues the esp32 will go to sleep eventually, and hope the issue has gone next time it wakes. The wifi signal is marginal at the deployment so if someone is using a microwave oven or something the wifi may be down

Deep sleep can be prevented by disabling the automation and publishing a retained mqtt message bedev/ota_mode ON. OTA can then be performed. Then another message is needed to return to normal mode payload OFF, and then simply re-enabling the automation puts the esp immediately back to sleep since there is already an mqtt message that fits the criteria.

Update interval has no effect when deep sleep is in use and it is longer than run_duration, which answers my other question. Esphome always sends the readings on boot.

I'd be interested in any further comments if I have missed something. Other than that this may be closed. Thanks!

OttoWinter, 2019-11-09:
- `run_time` is the ESP is running the `loop()` method after everything is set up.
So that time begins when everything is fully set up, including MQTT connection. 

- `update_interval` is not restored on boot - that means if you set update interval 45min that effectively means once per boot-cycle. If it's shorter than `run_time`, the sensor will report multiple readings each time it wakes up.

> As the deep sleep time seems to be fairly imprecise

That value is based on the internal oscillator (don't know if it's using another one for deep sleep, probably it does). Those oscillators are not known to be very accurate (and usually they don't have to either).

Otherwise I can't see any questions left - if there are, please post them here (sorry, don't have time reading through the entire wall of text)

closed 2019-11-09 by OttoWinter

spattinson, 2019-11-10:
Perfect. No thats great thanks