esphome/issues #834: Problems using DF-Player mini

S-Przybylski, 2019-11-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
esphome on Raspi-Docker 1.14.2
ESP: Wemos D1 mini with DF-Player mini
PIR Sensor SR501 on GPIO
SD Card with music (selected track is 30 minutes long!)

**Affected component:**
DF-Player mini automation

**Description of problem:**
Today i tried to enable the DF player mini. The main function taken out of the example are working.
Currently i have two issues:
1. I have trouble to work with the dfplayer.is_playing condition. 
How should it work: If the PIR sensor switches on again while the df player is playing, the event should be ignored. With my configuration the music is restarted every time the PIR switched to on.
2. I do not understand how to forward the is_playing condition towards HA. I tried a template sensor with some assumptions... but if i compile this configuration, i got an error.

Could you please support me getting this run?

**Problem-relevant YAML-configuration entries:**
```
# PIR SR501 ######################################
binary_sensor:
  - platform: gpio
    pin: D5
    name: "PIR Sensor"
    device_class: motion
# power off delay to suppress the death time of the sensor
    filters:
      - delayed_off: 3000ms
    on_press:
      then:
        if:
          condition:
            dfplayer.is_playing
          then:
            logger.log: 'Playback is active!'
          else: 
            - logger.log: 'No active playback! Starting playback!'
            - dfplayer.set_device: TF_CARD
            - dfplayer.set_volume: 20
            - dfplayer.play_folder:
                file: 001
                folder: 01

# doesn't work - compilation fails
  - platform: template
    name: "dfplayer_is_playing"
    lambda: |-
      if (dfplayer.is_playing) {
        return true;
      } else {
        return false;
      }
```
**Logs (if applicable):**

```
[22:47:13][D][binary_sensor:036]: 'PIR Sensor': Sending state OFF
[22:47:17][D][binary_sensor:036]: 'PIR Sensor': Sending state ON
[22:47:17][D][main:632]: No active playback! Starting playback!
[22:47:17][D][dfplayer:030]: Send Command 0x9 arg 0x02
[22:47:17][D][dfplayer:030]: Send Command 0x6 arg 0x14
[22:47:17][D][dfplayer:030]: Send Command 0xf arg 0x501
[22:47:17][V][dfplayer:072]: Received message cmd: 0x41 arg 0000
[22:47:17][V][dfplayer:097]: Ack ok
[22:47:17][V][dfplayer:072]: Received message cmd: 0x41 arg 0000
[22:47:17][V][dfplayer:097]: Ack ok
[22:47:23][D][binary_sensor:036]: 'PIR Sensor': Sending state OFF
[22:47:28][D][binary_sensor:036]: 'PIR Sensor': Sending state ON
[22:47:28][D][main:632]: No active playback! Starting playback!
[22:47:28][D][dfplayer:030]: Send Command 0x9 arg 0x02
[22:47:28][D][dfplayer:030]: Send Command 0x6 arg 0x14
[22:47:28][D][dfplayer:030]: Send Command 0xf arg 0x501
[22:47:28][V][dfplayer:072]: Received message cmd: 0x41 arg 0000
[22:47:28][V][dfplayer:097]: Ack ok
[22:47:28][V][dfplayer:072]: Received message cmd: 0x41 arg 0000
[22:47:28][V][dfplayer:097]: Ack ok
[22:47:28][W][dfplayer:064]: Expected end byte 0xEF, got 0xff
[22:47:38][D][binary_sensor:036]: 'PIR Sensor': Sending state OFF
[22:47:41][D][binary_sensor:036]: 'PIR Sensor': Sending state ON
[22:47:41][D][main:632]: No active playback! Starting playback!
[22:47:41][D][dfplayer:030]: Send Command 0x9 arg 0x02
[22:47:41][D][dfplayer:030]: Send Command 0x6 arg 0x14
[22:47:41][D][dfplayer:030]: Send Command 0xf arg 0x501
[22:47:41][V][dfplayer:072]: Received message cmd: 0x41 arg 0000
[22:47:41][V][dfplayer:097]: Ack ok
[22:47:42][V][dfplayer:072]: Received message cmd: 0x41 arg 0000
[22:47:42][V][dfplayer:097]: Ack ok
[22:47:48][D][binary_sensor:036]: 'PIR Sensor': Sending state OFF
```

OttoWinter, 2019-11-09:
As for 2., it would be `lambda: 'return id(my_df_player).is_playing();'` and give your dfplayer component the id `my_df_player`

OttoWinter, 2019-11-09:
CC: @glmnet (dfplayer contributor)

S-Przybylski, 2019-11-09:
Dear @OttoWinter 
thanks for you help. I 'll try to use that!

Regarding to dfplayer.set_eq Action:
If i send "eq_preset: ROCK" or other key words, HA responds that the key is not valid. I assume that the exchange of the value name and the function (lets say) "exchange defined state to related number" is somewhere broken.

This works within HA development tools --> services menu:
Service:   esphome.dfminiplayer_test_dfplayer_set_eq
Service Data (YAML, optional):  preset: 1

Documentation shows instead:
````
on_...:
  then:
    - dfplayer.set_eq:
        eq_preset: ROCK
    # Shorthand
    - dfplayer.set_eq: ROCK
````

glmnet, 2019-11-09:
Ahh silly me, the `is_playing` condition is being set only if you use the `play_file` not in many other situations.

glmnet, 2019-11-09:
@OttoWinter should this one be included in a point release?
Should I create the PR to dev or another branch?

S-Przybylski, 2019-11-11:
Dear @glmnet 
i think the best will be that you provide the patch to the dev system.

glmnet, 2019-11-11:
> Dear @OttoWinter
> thanks for you help. I 'll try to use that!
>
> Regarding to dfplayer.set_eq Action:
> If i send "eq_preset: ROCK" or other key words, HA responds that the key is not valid. I assume [...]

I forgot to replay about this... are you writing `ROCK` in HA Yamls? That will not work, `ROCK` is only valid in ESPHome Yaml after `.set_eq:`

closed 2019-11-12 by glmnet