esphome/issues #836: compile error using atm90e32 power_factor sensor, an runtime error onced fixed.

Diramu, 2019-11-10:
Docker and command line

ESP32

atm90e32

Using the new (latest dev only) sensor power_factor. You get a coompile error.
Fixing the compile error, https://github.com/esphome/esphome/pull/841
You then get a boot sequence error, which keeps looping.

**Problem-relevant YAML-configuration entries:**
```- platform: atm90e32
    cs_pin: 15
    phase_a:
      voltage:
        name: ${disp_name} CT7 Volts
        id: ic3aVolts
        accuracy_decimals: 1
        filters:
          - multiply: ${mult_v}
      current:
        name: ${disp_name} CT7 Amps
        id: ct7Amps
      power:
        name: ${disp_name} CT7 Pwr
        id: ct7Pwr
        filters:
          - multiply: ${mult_v}
      gain_voltage: ${gain_v}
      gain_ct: ${gain_ct30A}
      power_factor:                                                                                                                                                         
        name: ${disp_name} CT7 Pf                                                                                                                                           
        id: ct7Pf                     
```

**Logs (if applicable):**

compile error
```
src/main.cpp:1900:33: error: 'class esphome::atm90e32::ATM90E32Component' has no member named 'set_pf_sensor'
   atm90e32_atm90e32component_4->set_pf_sensor(2, ct12Pf);```

After fixing the compile error:
```
17:47:00]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[17:47:00]configsip: 0, SPIWP:0xee
[17:47:00]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[17:47:00]mode:DIO, clock div:2
[17:47:00]load:0x3fff0018,len:4
[17:47:00]load:0x3fff001c,len:952
[17:47:00]load:0x40078000,len:6084
[17:47:00]load:0x40080000,len:7944
[17:47:00]entry 0x40080310
[17:47:00][I][logger:156]: Log initialized
[17:47:00][C][ota:364]: There have been 0 suspected unsuccessful boot attempts.
[17:47:00][I][app:028]: Running through setup()...
[17:47:00][C][spi:020]: Setting up SPI bus...
[17:47:00][C][atm90e32:071]: Setting up ATM90E32Component...
[17:47:00][W][atm90e32:083]: Could not initialize ATM90E32 IC, check SPI settings
[17:47:00][E][component:092]: Component was marked as failed.
[17:47:00][C][atm90e32:071]: Setting up ATM90E32Component...
[17:47:00][W][atm90e32:083]: Could not initialize ATM90E32 IC, check SPI settings
[17:47:00][E][component:092]: Component was marked as failed.
[17:47:00][C][atm90e32:071]: Setting up ATM90E32Component...
[17:47:00][W][atm90e32:083]: Could not initialize ATM90E32 IC, check SPI settings
[17:47:00][E][component:092]: Component was marked as failed.
[17:47:00][C][atm90e32:071]: Setting up ATM90E32Component...
[17:47:00][W][atm90e32:083]: Could not initialize ATM90E32 IC, check SPI settings
[17:47:00][E][component:092]: Component was marked as failed.
[17:47:00][C][wifi:033]: Setting up WiFi...
[17:47:00]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
[17:47:00]Debug exception reason: Stack canary watchpoint triggered (loopTask) 
[17:47:00]Core 1 register dump:
[17:47:00]PC      : 0x40084b68  PS      : 0x00060836  A0      : 0x3ffb0110  A1      : 0x3ffb0050  
[17:47:00]A2      : 0x3ffb013c  A3      : 0x00000005  A4      : 0x8008a168  A5      : 0x3ffbe780  
[17:47:00]A6      : 0x00000003  A7      : 0x00060023  A8      : 0x00040000  A9      : 0x3ff42080  
[17:47:00]A10     : 0x3ff42000  A11     : 0x7000006b  A12     : 0x5c000008  A13     : 0xd0000070  
[17:47:00]A14     : 0x0070a000  A15     : 0xb0000000  SAR     : 0x00000020  EXCCAUSE: 0x00000001  
[17:47:00]EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
[17:47:00]
[17:47:00]Backtrace: 0x40084b68:0x3ffb0050 0x3ffb010d:0x3ffb0130 0x4009282e:0x3ffb0160 0x40092b84:0x3ffb0190 0x40092dd9:0x3ffb01b0 0x400871b1:0x3ffb01e0 0x40087428:0x3ffb0200 0x40112619:0x3ffb0250 0x4010fa66:0x3ffb0270 0x4010fe2c:0x3ffb0290 0x4010eddb:0x3ffb0310 0x4010f37e:0x3ffb0390 0x4010e761:0x3ffb0400 0x4015d103:0x3ffb0440 0x4015d2be:0x3ffb0490 0x4015d311:0x3ffb04c0 0x40112adf:0x3ffb04e0 0x40112c32:0x3ffb0500 0x400fd349:0x3ffb0520 0x400ee0dc:0x3ffb0540 0x400e0e05:0x3ffb0640 0x400e0e40:0x3ffb0670 0x400e01a8:0x3ffb06a0 0x401726dd:0x3ffb06c0 0x401727c5:0x3ffb06e0 0x400e2901:0x3ffb0700 0x400ea25d:0x3ffb0750 0x400fa403:0x3ffb1fb0 0x40088ff9:0x3ffb1fd0
[17:47:00]
[17:47:00]Rebooting...
```

Removing the use of the power_factor sensor   (12 of them), the esp32 boots correctly.

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

OttoWinter, 2019-11-12:
CC: @CircuitSetup (author power factor)

CircuitSetup, 2019-11-12:
I believe I forgot to change to the full names in sensor.py. This fixes the issue: https://github.com/esphome/esphome/pull/841

glmnet, 2020-01-17:
Fixed by esphome/esphome#841

closed 2020-01-17 by glmnet