esphome/issues #837: esp32 resets by itself

Oliviakrkk, 2019-11-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

esp32

**Affected component:**

esp32 ble

**Description of problem:**

esp32 keeps reseting. 

**Problem-relevant YAML-configuration entries:**

```
esphome:
  name: esp32
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "OLPN"
  password: 

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 

ota:
  password: 

esp32_ble_tracker:

binary_sensor:
  - platform: ble_presence
    mac_address: FF:FF:40:01:22:8C
    name: "ESP32 BLE Tracker iTag1"
  - platform: ble_presence
    mac_address: FF:FF:00:03:EC:58
    name: "ESP32 BLE Tracker iTag3"
  - platform: ble_presence
    mac_address: D0:F0:18:43:EB:7A
    name: "iNode1"
sensor:
  - platform: uptime
    name: "ESP32 LV Uptime Sensor"
  - platform: ble_rssi
    mac_address: D0:F0:18:43:EB:7A
    name: "iNode RSSI value"
  - platform: ble_rssi
    mac_address: FF:FF:40:01:22:8C
    name: "iTag1 RSSI value"
  - platform: ble_rssi
    mac_address: FF:FF:00:03:EC:58
    name: "iTag3 RSSI value"
  - platform: xiaomi_hhccjcy01
    mac_address: 'C4:7C:8D:6A:DC:3B'
    temperature:
      name: "Xiaomi MiFlora Fikus benjamin Temperature"
    moisture:
      name: "Xiaomi MiFlora Fikus benjamin Moisture"
    illuminance:
      name: "Xiaomi MiFlora Fikus benjamin Illuminance"
    conductivity:
      name: "Xiaomi MiFlora Fikus benjamin Soil Conductivity"
  - platform: xiaomi_hhccjcy01
    mac_address: '80:EA:CA:88:B4:21'
    temperature:
      name: "Xiaomi MiFlora Mentha1 Temperature"
    moisture:
      name: "Xiaomi MiFlora Mentha1 Moisture"
    illuminance:
      name: "Xiaomi MiFlora Mentha1 Illuminance"
    conductivity:
      name: "Xiaomi MiFlora Mentha1 Soil Conductivity"
  - platform: xiaomi_hhccjcy01
    mac_address: '80:EA:CA:88:AE:E2'
    temperature:
      name: "Xiaomi MiFlora Mentha2 Temperature"
    moisture:
      name: "Xiaomi MiFlora Mentha2 Moisture"
    illuminance:
      name: "Xiaomi MiFlora Mentha2 Illuminance"
    conductivity:
      name: "Xiaomi MiFlora Mentha2 Soil Conductivity"
```

**Logs (if applicable):**

```
[10:14:42][D][sensor:092]: 'ESP32 LV Uptime Sensor': Sending state 414.05499 s with 0 decimals of accuracy
[10:14:48]/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:706 (xQueueCreateCountin
gSemaphore)- assert failed!
[10:14:48]abort() was called at PC 0x4008e0c7 on core 0
[10:14:48]
[10:14:48]Backtrace: 0x40092094:0x3ffe9900 0x400922c5:0x3ffe9920 0x4008e0c7:0x3ffe9940 0x40116111:0x3ffe9960 0x4011615e:0x3ffe9980 0x400dcbc
0:0x3ffe99a0 0x4008453e:0x3ffe99e0 0x400f5eed:0x3ffe9a30 0x40138559:0x3ffe9a80 0x4013d5d1:0x3ffe9ac0 0x401428ba:0x3ffe9ae0 0x4012e12f:0x3ffe
9b00 0x4008e7dd:0x3ffe9b30
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40092094: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/p
anic.c:707
WARNING Decoded 0x400922c5: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:
707
WARNING Decoded 0x4008e0c7: xQueueCreateCountingSemaphore at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/c
omponents/freertos/queue.c:2520
WARNING Decoded 0x40116111: static_init_prepare() at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/component
s/cxx/cxx_guards.cpp:210 (discriminator 1)
WARNING Decoded 0x4011615e: __cxa_guard_acquire at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/
cxx/cxx_guards.cpp:210 (discriminator 1)
WARNING Decoded 0x400dcbc0: esphome::esp_idf_log_vprintf_(char const*, __va_list_tag) at /home/olivia/esphome/esp32/src/esphome/core/log.cpp
:60 (discriminator 1)
WARNING Decoded 0x4008453e: esp_log_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/log/lo
g.c:215
WARNING Decoded 0x400f5eed: _udp_recv at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mdns/mdns_
networking.c:123 (discriminator 2)
WARNING Decoded 0x40138559: udp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/
src/core/udp.c:401
WARNING Decoded 0x4013d5d1: ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/
src/core/ipv4/ip4.c:740
WARNING Decoded 0x401428ba: ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c:184
WARNING Decoded 0x4012e12f: tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:483
WARNING Decoded 0x4008e7dd: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[10:14:48]
[10:14:48]Rebooting...
```

**Additional information and things you've tried:**

esphome version  1.14.2

glmnet, 2019-11-11:
Have you done any test / research?
Did yo check removing the uptime sensor? (Seems like the one causing the issue)

OttoWinter, 2019-11-12:
This looks like an out-of-memory issue. Please try with fewer xiaomi devices declared in your config to verify.

Oliviakrkk, 2019-11-12:
Hi

OK I will try both. 
First I will reduce xiaomi devices to 1 and then remove uptime counter.

I will get back with the results in some time.

jakehigg, 2019-11-12:
I have hit this same issue as well.  I attempted removing most of my BLE sensors, and this did not resolve the issue.  I then reduced down to a single BLE sensor and still had the issue.  Lastly i removed all BLE components and this resolved the problem.  It appears to me that when BLE is enabled the wifi drops.  If i just have the BLE tracker component enabled, in my logs I get a single wifi disconnect "reason=Auth Expired" and the device seems stable.  When I add a single Xiaomi CGG1 sensor things go down hill and constant wifi drops.  hope this helps.

jakehigg, 2019-11-12:
upon further investigation, it appears that a single Xiaomi CGG1 sensor works.  I have disabled the device tracker binary sensors and it appears to have resolved the wifi issues.  i did have a few wifi drop in setup() with auth expired, but it appears to have stabalized.  it is strange that after the auth expired i also got "bssid=00:00:00:00:00:00 reason=AP Not Found" the actual BSSID in the logs was all zeros.

Oliviakrkk, 2019-11-12:
Not good. I have Esp32 for presence detection and MiFlora ::

jakehigg, 2019-11-12:
try removing all presence detection and leaving the miflora.  i have been running for 30 minutes stable without the presence detection components.  I am suspecting that this could be related to the code refactoring that happened with the Xiaomi products in 1.14.0.  I am in the same boat.  I need presence detection to run logic that will determine if the ESP32 can still "see" the device, if it does it publishes an mqtt message with the sensor value.

brandond, 2019-11-12:
My understanding is that the BT and WiFi stacks use the same hardware, so the more time it spends doing things like scanning for BT packets, the more likely it is that it'll drop off the WiFi connection due to missing too many beacon intervals. I don't know how it's intended to work, but people sure seem to have a lot of trouble with it.

OttoWinter, 2019-11-12:
@brandond Correct. Yeah I'm reverting the defaults to 1.13.x (see also explanation in https://github.com/esphome/esphome/pull/855).

However, I believe the disconnects are not related to this issue. The new defaults would only explain why the device is dropping off wifi, but would not explain why the device is crashing in `xQueueCreateCountingSemaphore`.

brandond, 2019-11-12:
If you look at the stack trace, the interesting bit is at
https://github.com/espressif/esp-idf/blob/master/components/mdns/mdns_networking.c#L123 which is a call to whatever the HOOK_MALLOC_FAILED macro resolves to, with a comment on the following line of "missed packet - no memory". Given that the stack after that goes into esp_log_write, the hook seems to involve writing a log entry, which in turn tries to queue a semaphore. However, since the system is out of memory, it can't create the new semaphore and crashes on an assertion.

tl;dr it's out of memory and the network stack blows up

Oliviakrkk, 2019-11-14:
I removed some sensors and now it was stable for a day. Closing issue. 

Thanks!

closed 2019-11-14 by Oliviakrkk