esphome/issues #838: DHCP client problems, temporary solution with manual fixed IP

PieBru, 2019-11-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Arch Linux on I7 + HASSIO 0.100.3 + ESPHome 1.14.2 (all just updated)

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 (Wemos D1 Mini Pro and others)

**Affected component:**

DHCP client

**Description of problem:**
Continuous API disconnections every couple of minutes.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  device_name: d1mini_test1

esphome:
  name: $device_name
  platform: ESP8266
  board: d1_mini_pro

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # **Temporary solution**
  manual_ip:
    static_ip: 172.16.2.128
    gateway: 172.16.1.1
    subnet: 255.255.0.0
    dns1: 172.16.1.1
    dns2: 8.8.8.8

logger:

api:
  password: !secret default_password

ota:
  password: !secret default_password

# Wemos D1 Mini builtin LED
status_led:
  pin:
    number: D4
    inverted: true
```

**Logs (if applicable):**

```
Shown on https://community.home-assistant.io/t/solved-esphome-dhcp-client-problems-temporary-solution-with-manual-fixed-ip/147312/10
```

**Additional information and things you've tried:**
Please see https://community.home-assistant.io/t/solved-esphome-dhcp-client-problems-temporary-solution-with-manual-fixed-ip/147312/10 about narrowing down of the problem, as it initially appeared as a Wifi issue.

OttoWinter, 2019-11-10:
I doubt that's the case. DHCP client only is active while WiFi connection is set up. After that, it essentially behaves the same way as with a manual IP config.

Please also post your entire config/log here. In your community forums thread you posted you have mqtt/api active at the same time, but I only see one of them here.

What might be an issue though is mDNS, there have been some reports that the latest arduino SDK broke mDNS.

Please try the following: `arduino_version: 2.4.2` in your top-level `esphome:` config.

PieBru, 2019-11-10:
> In your community forums thread you posted you have mqtt/api active at the same time, but I only see one of them here.

This was initially, then I disabled MQTT to narrow down the problem, which didn't change.
The last problematic ESPHome config I used is the one I reported here.

`arduino_version: 2.4.2` doesn't solve :(
The error appears continuously between one and two minutes from `INFO Successfully connected to d1mini_test1.local`

```YAML
substitutions:
  device_name: d1mini_test1

esphome:
  name: $device_name
  platform: ESP8266
  board: d1_mini_pro
  arduino_version: 2.4.2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  #fast_connect: true
  #reboot_timeout: 45min
  #manual_ip:
  #  static_ip: 172.16.2.128
  #  gateway: 172.16.1.1
  #  subnet: 255.255.0.0
  #  dns1: 172.16.1.1
  #  dns2: 8.8.8.8

logger:

api:
  password: !secret default_password

ota:
  password: !secret default_password

# Wemos D1 Mini builtin LED
status_led:
  pin:
    number: D4
    inverted: true
```

```LOG
...
Checking size /data/d1mini_test1/.pioenvs/d1mini_test1/firmware.elf
DATA:    [====      ]  44.2% (used 36176 bytes from 81920 bytes)
PROGRAM: [===       ]  31.7% (used 331528 bytes from 1044464 bytes)
========================= [SUCCESS] Took 13.91 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of d1mini_test1.local
INFO  -> 172.16.2.128
INFO Uploading /data/d1mini_test1/.pioenvs/d1mini_test1/firmware.bin (335680 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from d1mini_test1.local using esphome API
WARNING Error resolving IP address of d1mini_test1.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname). Re-Trying in 1 seconds
INFO Connecting to d1mini_test1.local:6053 (172.16.2.128)
INFO Successfully connected to d1mini_test1.local
[12:16:07][I][app:100]: ESPHome version 1.14.2 compiled on Nov 10 2019, 12:15:33
[12:16:07][C][status_led:019]: Status LED:
[12:16:07][C][status_led:020]:   Pin: GPIO2 (Mode: OUTPUT, INVERTED)
[12:16:08][C][wifi:409]: WiFi:
[12:16:08][C][wifi:277]:   SSID: [redacted]
[12:16:08][C][wifi:278]:   IP Address: 172.16.2.128
[12:16:08][C][wifi:280]:   BSSID: [redacted]
[12:16:08][C][wifi:281]:   Hostname: 'd1mini_test1'
[12:16:08][C][wifi:285]:   Signal strength: -43 dB ▂▄▆█
[12:16:08][C][wifi:289]:   Channel: 6
[12:16:08][C][wifi:290]:   Subnet: 255.255.0.0
[12:16:08][C][wifi:291]:   Gateway: 172.16.1.1
[12:16:08][C][wifi:292]:   DNS1: 172.16.1.4
[12:16:08][C][wifi:293]:   DNS2: 172.16.1.8
[12:16:08][C][logger:175]: Logger:
[12:16:08][C][logger:176]:   Level: DEBUG
[12:16:08][C][logger:177]:   Log Baud Rate: 115200
[12:16:08][C][logger:178]:   Hardware UART: UART0
[12:16:08][C][ota:029]: Over-The-Air Updates:
[12:16:08][C][ota:030]:   Address: d1mini_test1.local:8266
[12:16:08][C][ota:032]:   Using Password.
[12:16:08][C][api:095]: API Server:
[12:16:08][C][api:096]:   Address: d1mini_test1.local:6053
[12:16:09][D][api.connection:579]: Client 'Home Assistant 0.101.3 (172.16.1.4)' connected successfully!
[12:17:55][I][ota:046]: Boot seems successful, resetting boot loop counter.
ERROR Error while reading incoming messages: Error while receiving data: [Errno 104] Connection reset by peer
WARNING Disconnected from API: Error while receiving data: [Errno 104] Connection reset by peer
INFO Connecting to d1mini_test1.local:6053 (172.16.2.128)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 1 seconds
INFO Connecting to d1mini_test1.local:6053 (172.16.2.128)
INFO Successfully connected to d1mini_test1.local
ERROR Error while reading incoming messages: Error while receiving data: [Errno 104] Connection reset by peer
WARNING Disconnected from API: Error while receiving data: [Errno 104] Connection reset by peer
INFO Connecting to d1mini_test1.local:6053 (172.16.2.128)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 1 seconds
INFO Connecting to d1mini_test1.local:6053 (172.16.2.128)
INFO Successfully connected to d1mini_test1.local
ERROR Error while reading incoming messages: Error while receiving data: [Errno 104] Connection reset by peer
WARNING Disconnected from API: Error while receiving data: [Errno 104] Connection reset by peer
INFO Connecting to d1mini_test1.local:6053 (172.16.2.128)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 1 seconds
INFO Connecting to d1mini_test1.local:6053 (172.16.2.128)
INFO Successfully connected to d1mini_test1.local
...
```

Thanks,
Piero

PieBru, 2019-11-10:
While `arduino_version: 2.4.2` doesn't solve, the status LED blinks only 3-4 times, less insistently than before.
To rout hardware doubts this is my test setup, it's pretty basic:
[image]

PieBru, 2019-11-10:
And the DHCP/mDNS server is an up-to-date OpenWRT, as shown here: https://community.home-assistant.io/t/solved-esphome-dhcp-client-problems-temporary-solution-with-manual-fixed-ip/147312/18

PieBru, 2019-11-10:
More tests using the same ESPHome build system:
- Same problem with ESP8266 Wemos D1 Mini v3.0.0
- No problem with ESP-WROOM-32 (Doit devkit v1)

So the issue appears only on the ESP8266 platform, presumably on all boards.

RGN01, 2019-11-10:
I'm experiencing the same (or very similar) issues on NodeMCU devices - but only two of the perhaps 10 that I have. Both of these have Dallas DS18b20 temperature sensors and none of the others do. I wonder if this is in some way significant?

PieBru, 2019-11-11:
@RGN01 
If you can, I suggest you to try with a bare-bone NodeMCU (without DS18B20, physically and in the yaml) with `debug:` enabled in the yaml, and then try adding the DS18B20 to spot out differences in the ESP behavior and logs.

PieBru, 2019-11-11:
Found a dusty DS18B20.
In my case with my bare-bone test wemos boards it does **not** influence the problem.

RGN01, 2019-11-11:
@PieBru Thanks - I'm very limited in how much time I can spend on this during the week but I have done some basic testing. Before I started I observed the 'resets' (if that is what they are) and they occurred about 55 - 65 seconds apart

I started by commenting out the Dallas components and the timing component. This seemed to make a major difference. I then realised that the default Dallas update period is 60 seconds so I  uncommented the Dallas components but left the timing component disabled. 

This seems to have worked and my working assumption at this point is the time: component was causing the problem.

The part I commented out is:
```
time:
  - platform: sntp
    on_time:
      - seconds: 0
        minutes: /1
        then:
          - component.update: ds18b20
```

I'll need to monitor it for longer but the signs are good - it is a lot more stable and I've not seen a 'reset' since I did this.

The other thing I did when I did this was enable debug - I'll try disabling that later this week, leaving the rest the same.

I hope this helps.

RGN01, 2019-11-11:
Just to clarify, this is on NodeMCU boards that I've been running unchanged for about a year - apart from ESPHome version updates.

RGN01, 2019-11-12:
Correction. It was not the time: aspect that prevented the 'resets' - it is monitoring the unit with 'show logs' . I noticed this a few minutes after writing the earlier message. While monitoring it with that, it appears to be rock solid, stop monitoring and the problem reoccurs.

Armed with that, I'm now testing the Dallas theory and it does seem to be a major contributing cause, if not the cause. 

Before / problematic:
```
esphome:
  name: shed_multi
  platform: ESP8266
  board: nodemcuv2
  
wifi:
  ssid: !secret wifi_name
  password: !secret wifi_password
  domain: !secret wifi_domain

# Enable logging
debug: 
logger:
  level: DEBUG

# Enable Home Assistant API
api:

ota:

i2c:
  sda: D1
  scl: D2
  scan: True

dallas:
  - pin: D3
    id: "ds18b20"

binary_sensor:

  - platform: gpio
    pin:
      number: D4
      mode: INPUT_PULLUP
      inverted: FALSE
    id: shed_door
    name: "Shed Door"
    device_class: opening

  - platform: gpio
    pin:
      number: D5
      mode: INPUT_PULLUP
      inverted: FALSE
    id: shed_window_1
    name: "Shed Window 1"
    device_class: opening

  - platform: gpio
    pin:
      number: D6
      mode: INPUT_PULLUP
      inverted: FALSE
    id: shed_window_2
    name: "Shed Window 2"
    device_class: opening

  - platform: gpio
    pin:
      number: D7
      mode: INPUT_PULLUP
      inverted: FALSE
    id: shed_window_3
    name: "Shed Window 3"
    device_class: opening

output:
  - platform: gpio
    pin: D8
    id: gpio15   
    
# GPIO set as an output so that it can be used for reed switches but not interfere with boot
# if D4 is low at boot, goes to flash mode
  - platform: gpio
    pin: D0
    id: gpio16   

light:
  - platform: binary
    name: "Shed LED"
    id: shed_led
    output: gpio15    

sensor:
  - platform: wifi_signal
    name: "WiFi - Shed Multi"
    update_interval: 60s

  - platform: dallas
    address: 0xD680000007CE4128
    name: "Garage external"
  - platform: dallas
    address: 0x5E0316504B0AFF28
    name: "Shed internal"
  - platform: dallas
    address: 0x9E031647CAFAFF28
    name: "Hotbox"
  - platform: dallas
    address: 0x790316521536FF28
    name: "HW Return House End"
  - platform: dallas
    address: 0xAC041654CAFEFF28
    name: "Garage internal"
  - platform: dallas
    address: 0xF50316522561FF28
    name: "HW feed house end"
  - platform: dallas
    address: 0xFB03165186E1FF28
    name: "HW return shed end"
  - platform: dallas
    address: 0x250416553733FF28
    name: "HW feed shed end"

  - platform: bme280
    temperature:
      name: "BME280 Temperature"
      oversampling: 16x
    pressure:
      name: "BME280 Pressure"
    humidity:
      name: "BME280 Humidity"
    address: 0x76
    update_interval: 60s
    
  - platform: adc
    pin: A0
    name: "Brightness"
    update_interval: 10s
    unit_of_measurement: "Light"
    accuracy_decimals: 1
    filters:
      - calibrate_linear:
          # Map 0.0 (from sensor) to 0.0 (true value)
          - 0.0 -> 0.0
          - 0.75 -> 100.0
      - sliding_window_moving_average:
          window_size: 6
          send_every: 6
```
 
After / OK:
```
esphome:
  name: shed_multi
  platform: ESP8266
  board: nodemcuv2
  
wifi:
  ssid: !secret wifi_name
  password: !secret wifi_password
  domain: !secret wifi_domain

# Enable logging
debug: 
logger:
  level: DEBUG

# Enable Home Assistant API
api:

ota:

i2c:
  sda: D1
  scl: D2
  scan: True

binary_sensor:

  - platform: gpio
    pin:
      number: D4
      mode: INPUT_PULLUP
      inverted: FALSE
    id: shed_door
    name: "Shed Door"
    device_class: opening

  - platform: gpio
    pin:
      number: D5
      mode: INPUT_PULLUP
      inverted: FALSE
    id: shed_window_1
    name: "Shed Window 1"
    device_class: opening

  - platform: gpio
    pin:
      number: D6
      mode: INPUT_PULLUP
      inverted: FALSE
    id: shed_window_2
    name: "Shed Window 2"
    device_class: opening

  - platform: gpio
    pin:
      number: D7
      mode: INPUT_PULLUP
      inverted: FALSE
    id: shed_window_3
    name: "Shed Window 3"
    device_class: opening

output:
  - platform: gpio
    pin: D8
    id: gpio15   
    
# GPIO set as an output so that it can be used for reed switches but not interfere with boot
# if D4 is low at boot, goes to flash mode
  - platform: gpio
    pin: D0
    id: gpio16   

light:
  - platform: binary
    name: "Shed LED"
    id: shed_led
    output: gpio15    

sensor:
  - platform: wifi_signal
    name: "WiFi - Shed Multi"
    update_interval: 60s

  - platform: bme280
    temperature:
      name: "BME280 Temperature"
      oversampling: 16x
    pressure:
      name: "BME280 Pressure"
    humidity:
      name: "BME280 Humidity"
    address: 0x76
    update_interval: 60s
    
  - platform: adc
    pin: A0
    name: "Brightness"
    update_interval: 10s
    unit_of_measurement: "Light"
    accuracy_decimals: 1
    filters:
      - calibrate_linear:
          # Map 0.0 (from sensor) to 0.0 (true value)
          - 0.0 -> 0.0
          - 0.75 -> 100.0
      - sliding_window_moving_average:
          window_size: 6
          send_every: 6
```
 The only difference is removing the Dallas components.

OttoWinter, 2019-11-12:
Thanks for testing with the old SDK.

> While arduino_version: 2.4.2 doesn't solve, the status LED blinks only 3-4 times, less insistently than before.

Interesting. If you don't have anything else in your config (other than status_led), that means the device is probably in OTA safe mode - meaning it's crashing a bunch of times until going into safe mode. But that will need more info because it could also be just because of how many times you've uploaded a firmware before.

@RGN01 I don't think the dallas component can influence this - at least directly. There have been no changes to that component for a long time, which rules out a regression there. I think you're much rather affected by https://github.com/esphome/issues/issues/455.

As for @PieBru, that doesn't necessarily look like #455. It looks more like https://github.com/esphome/issues/issues/810.

Can you try this for me: Do not set a manual IP, let it use DHCP as normal. But set `use_address` to the address its getting over DHCP (you have to look that up manually) and set it using `use_address`. That will remove mDNS from the equation and help pinpoint the issue. In HA, you will need to re-add the device by entering the IP address in the integrations screen instead of using the auto-config.

```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: IP_OF_DEVICE
  # No manual_ip
```

PieBru, 2019-11-12:
The "ESPHome API disconnect" problem seems related, probably consequent, to a DHCP client lease request the ESPHome wemos issues every 1 minute or so to my OpenWRT up-to-date DHCP server, even if the wemos DHCP lease isn't expired yet (at least on the DHCP server side).
To stress this behavior, on the DHCP server I defined the leases expires every 2 minutes, but the wemos never reaches this limit, requesting a new lease every minute.

To help pinpoint the problem I minimized the test device to a factory vanilla [Wemos D1 mini v3.0.0](https://wiki.wemos.cc/products:retired:d1_mini_v3.0.0), but in my tests it affects other similar ESP8266-based boards.

To reproduce the issue I use this minimal yaml:
```YAML
esphome:
  name: d1mini_test2
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

This reproduces the every-1-minute DHCP client request.

While writing the above I received Otto's reply notify, so I tried adding only `use_address: IP_OF_DEVICE` to the above yaml, and **nothing changed**.

OttoWinter, 2019-11-12:
@PieBru I may have misunderstood something. Did you have this issue before 1.14? I thought you meant it only started to appear in 1.14.x, but looking through this thread I can see no reference to that.

If it is related to DHCP, it's likely an upstream issue. ESPHome just uses the normal dhcp client functions exposed through the ESP SDK, which is closed-source and so not possible to debug/fix.

I'm not sure about how DHCP is supposed to work - but could it be due to the DHCP lease time? I mean 2 minutes sounds like a very small value - mine is on the order of months.

I just checked with an ESP8266 of mine and my DHCP server, and it only requests a DHCP lease once it connects, and not every minute. So I suspect it may have to do with the short lease time.

Not that it means the lease time itself is the issue - it's probably the SDK. But as mentioned there's no way we can fix the SDK if it does have a DHCP client issue.

PieBru, 2019-11-12:
> @PieBru I may have misunderstood something. Did you have this issue before 1.14? I thought you meant it only started to appear in 1.14.x, but looking through this thread I can see no reference to that.

I tested this also on 1.13.6, please see https://community.home-assistant.io/t/solved-esphome-dhcp-client-problems-temporary-solution-with-manual-fixed-ip/147312/10

> If it is related to DHCP, it's likely an upstream issue. ESPHome just uses the normal dhcp client functions exposed through the ESP SDK, which is closed-source and so not possible to debug/fix.

Using Espurna (PlatformIO) I don't have this issue, so there should be a way to get rid of it. Espurna by default uses Arduino Core 2.3.0 (platformIO 1.5.0).

> Not that it means the lease time itself is the issue - it's probably the SDK. But as mentioned there's no way we can fix the SDK if it does have a DHCP client issue.

I will try with Arduino Core 2.3.0 and/or a longer lease time, and report back here.

Thank you much.
Piero

PieBru, 2019-11-12:
This issue seems an upstream problem, solved by `arduino_version: 2.5.2`, here's why:

Arduino version recap:
1. Building with `arduino_version: 2.3.0` gives too much errors.
2. Building with `arduino_version: 2.4.2` doesn't solve.
3. **Building with `arduino_version: 2.5.2` seems to solve this issue.**

Correction: the latest Espurna-dev built with `arduino_version: 2.3.0` has the same issue, but I din't notice it as in my config it does not act as an API server.

Changing the DHCP lease time to a loooonger one has no effect on this issue.

OttoWinter, 2019-11-12:
Uhm... 2.5.2 is the *default* for esphome v1.14...

brandond, 2019-11-12:
> To stress this behavior, on the DHCP server I defined the leases expires every 2 minutes, but the wemos never reaches this limit, requesting a new lease every minute.

This is standard behavior. DHCP leases are renewed at half of the lease time. If the client waited until 2 minutes to renew, the lease would technically be expired before the client renewed it.

See: https://en.wikipedia.org/wiki/Dynamic_Host_Configuration_Protocol#Reliability
> Clients begin to attempt to renew their leases once half the lease interval has expired. They do this by sending a unicast DHCPREQUEST message to the DHCP server that granted the original lease. If that server is down or unreachable, it will fail to respond to the DHCPREQUEST. However, in that case the client repeats the DHCPREQUEST from time to time, so if the DHCP server comes back up or becomes reachable again, the DHCP client will succeed in contacting it and renew the lease.

RGN01, 2019-11-12:
> > @RGN01 I don't think the dallas component can influence this - at least directly. There have been no changes to that component for a long time, which rules out a regression there. I think you're much rather affected by #455.
> 

Thank you. Unfortunately the symptoms of #455 are not close to what I am seeing and the are two guaranteed methods of stopping the issue occurring:
1. remove the Dallas components or
2. leave Debug logging on and the PC collecting the debug data 

I'll keep experimenting but am away for the next 3 weeks so won't be able to do much more until the 1st week of December at the earliest.

PieBru, 2019-11-12:
@brandond thanks for pointing out the DHCP lease half-time policy, I totally missed that.
However, my other network nodes - mainly Linux boxes - request the DHCP renew few seconds before the lease expires, and doesn't cause any perceivable TCP/IP disconnection. This is why the wemos behavior raised my attention.
I defined 2 minutes for a stress test and then leaved it so short to have a frequently updated list of active LAN nodes, I queried it via OpenWRT API. I don't use this trick anymore, but 2 minutes leases didn't hurt the other (mostly) Linux systems and remained in the config.

My 2 minutes DHCP lease may be uncommon, but AFAIK is legal and in this case may be considered as good a stress-test to pinpoint some unpredictable and rare problems.

I definitely must make more tests and researches to understand what happened with this DHCP+wemos+arduino_version mix of tests. Side note: before compiling I always do a redundant "clean build".

So my concern now is: **The "ESPHome API disconnect" may cause problems at every DHCP client renew request?**

ESP8266 wifi is closed-source, but it doesn't disconnect and it doesn't seem to be the cause of this "ESPHome API disconnect" behavior.
The networking stack - including Ethernet, DHCP, DNS, TCP/UDP, etc. - are in the Extensa/Arduino toolkits, which fortunately are open-source. 
This is why, from the ESPHome point of view, this may be an upstream issue. Anyway this is why if it is a real problem it can be solved.

PieBru, 2019-11-13:
Thanks all, here is a recap after the above issue origin narrowing.

Test environment: HASS 0.101.3, ESPHome 1.14.2, a bunch of Wemos D1 Mini models (and probably most ESP8266 boards with arduino_version up to 2.5.2).

With this simple yaml:
```YAML
esphome:
  name: d1mini_test2
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
logger:

debug:
    
api:
  password: !secret default_password

ota:
  password: !secret default_password
```

I observed these log errors:
[image]

Those errors and warnings are repeated continuously every minute by stressing the DHCP lease time to 2 minutes, which is legal and may be useful to spot DHCP clients problems. OpenWRT screenshot:
[image]

- A "normal" or "relaxed" DHCP lease reduces the log error messages rate but does not eliminate it.
- **This issue seems somewhat related to the ESP8266 DHCP client lease renewal**, as it can be bypassed by assigning fixed IP to all ESP8266 devices.
- **ESP32 ESPHome devices are not affected** (used the same yaml except `platform: ESP32` and `board: esp32doit-devkit-v1`).
- The problem may reside in the upstream networking stack inside the Extensa/Arduino toolkits (which includes Ethernet, DHCP, DNS, TCP/UDP, etc.). Fortunately it's open-source, but I didn't yet dig into it.
- I didn't dig/experiment if ESPHome or the HASS API client code can solve/mitigate/bypass this ESP8266-only DHCP-related problem.
- I didn't test `arduino_version: 2.6.0` or the corresponding `arduino_version: espressif8266@#.#.#` as I haven't a PlatformIO which supports it.

IMO:
1. This behavior should be considered a "problem" **only** if it can anyhow hurt something in the HA server and/or its user additions, or in the ESPHome device "autonomous" work. In this case, the fact it can be mitigated (to me it sounds as "randomly dispersed") using longer DHCP leases makes more difficult to spot those "problems", and can make it worst with the actual fast-paced growing of the user base.
2. If it is **not** a "problem", it would be nice if the logs doesn't show errors or warnings, or at least explain this behavior is benign, or temporary until ...

Hope this helps,
Piero

PieBru, 2019-11-13:
To make some test with 2.6.0, I tried using this [ESP tcp server example](https://circuits4you.com/2018/11/26/esp8266-nodemcu-tcp-socket-server-arduino-example/) in Arduino IDE, first with esp8266 board support v2.5.2, upgraded then to v2.6.0 .

I used both netcat and telnet on the TCP client side.

In all tests, no disconnections are signaled on the server or clients side while the ESP server renews its DHCP lease.

I'm not fully convinced of the effectiveness of this test, but that's it.
I hope this can help analyzing the issue by debugging it in ESPHome with the help of the "2 minutes" DHCP leases.

fribse, 2019-11-18:
I have 60 min DHCP lease, and with the 1.14.0 and 1.14.2 I couldn't get a stable connection to any of my ESP8266 devices. I can't see logfiles in the webinterface, and the webinterface shows them as offline, and they can't be updated.
It sometimes fails to update HA with status, and doesn't respond properly to commands the other way around, so I'm back at 1.13.6 which is still very stable.
I'll do an experiement with an update to 1.14.3 today, to see if anything has changed, but I won't update all the devices until it is absolutely stable.

fribse, 2019-11-18:
Ok, I can see that it now uses <ESPAsyncTCP-esphome> 1.2.2 where it used 1.2.1 previously.
It seems to go online as it should. If I add the webserver it also loads 1.2.1.
I have all my esp's on static IP's, not that it changed anything in 1.14.0 and 1.14.2. Before compiling I did a 'Clean build'.

fribse, 2019-11-22:
With static IP's and ESPHome 1.14.3 I still see repeatedly markings of offline in the GUI, despite that it can show logs when I click on that part.

RGN01, 2019-12-01:
> With static IP's and ESPHome 1.14.3 I still see repeatedly markings of offline in the GUI, despite that it can show logs when I click on that part.

Me too. I've been away for two weeks so not able to debug but have now updated HASS.IO to 0.102 and ESPHome to 1.14.3 and still have the same issues. I have tried both fixed IPs and DHCP with no difference. WiFi signal is strong (-57 and -56dB). I have tried different versions of Arduino_version with no apparent difference. There are two strange, perhaps pertinent attributes:
1. this only happens on my NodeMCUs running 1-wire (DS18B20) sensors, not the others (i.e this happens on 2 of my approximately 10 NodeMCUs)
2. When I am 'Showing Logs' from the ESPHome GUI this problem does not occur. When I stop logging it reoccurs after about 60 seconds

Unfortunately I do not have backups / images of the HASS.IO from before this problem occurred so I cannot easily revert to the earlier version of ESPHome. I really don't know what to try next except change the two affected NodeMCUs from ESP32 devices - but that will entail making new PC boards so I'm hoping to avoid that!

PieBru, 2019-12-01:
Maybe loosely related, just to desperately trying to clear more "random" causes.

The Tasmota wiki suggests some generic checks:
- https://github.com/arendst/Tasmota/wiki/FAQ#device-disconnects-from-wi-fi-often
- https://github.com/arendst/Tasmota/wiki/FAQ#wi-fi-stops-working
- https://github.com/arendst/Tasmota/wiki/FAQ#frequent-mqtt-reconnects

Once upon a time I experienced some issues with powerline ethernet adapters. These weird issues were caused by LED bulbs dropper capacitor, which introduced interference disturbing on the wifi frequency spectrum. Some nearby switching or "dropping" power supplies may cause these issues.

RGN01, 2019-12-01:
Very interesting, @PieBru - thank you. I will follow those Tasmota suggestions - I have nothing to lose. 

The two devices in question are in two separate buildings, close to two separate WiFi Access Points, using two different high quality power supplies (each capable of 1A at a steady 3V3) and in close proximity to other ESPHome devices that are not displaying this behaviour. As both started doing this at the same time after the upgrade I think it reasonable to assume it is not a hardware issue but I'll swap out one of them anyway (the other one is not easy to get to!) 

Thanks again!

RGN01, 2019-12-06:
This is almost certainly related to the Dallas 1-wire and DS18B20 in some way.

I set up a new D1 Mini with just the basic config, placed it in the location near one of the problematic ones and it was fine. Moved the DS18b20 functionality to it and the 'unavailable' issue moved from the NodeMCU that was affected to the D1 Mini.

I guess I'll replace the NodeMCU with an ESP32 as I believe they are not affected by this.

RGN01, 2019-12-07:
I've now got the DS18B20s on and ESP-32S and it is working fine. I still have the other functions on the original NodeMCU and that, too, is working fine. 

The toxic combination (for me, at least) therefore seems to be ESPHome, ESP8266 and the Dallas 1-wire DS18b20 devices (perhaps other 1-wire devices but I've not tested those)

I hope this helps identify the problem.

OttoWinter, 2019-12-07:
If it's really the dallas component, it might have to do with disabling interrupts (which is done for dallas to get good 1-wire timings, otherwise a lot of data is garbage).

In that case I would recommend trying to update to ESPHome dev (for now, will be in next patch release too) and setting arduino_version to the latest:

```yaml
esphome:
  # ...
  arduino_version: 2.6.2
```

jhiller227, 2019-12-16:
I am also having stability problems with my esp8266 devices all of my esp32 are fine. The devices that still have 1.13.6 are fine but every esp8266 that has 1.14.3 is unstable. I am running hasio 0.103.0 had have tried arduino 2.4.2 and 2.5.2 with the same result. the devices will run about 40 minutes then disconnect, serial login shows 'abort called'. I have tried to downgrade to 1.13.6 but the gui will not open. error 404. Any ideas?

QbaF, 2020-04-25:
still. 
ESP-WROOM-32; HA 0.108.8; ESPHome 1.14.3
Changing arduino_version to 1.0.4 does work.

GigaVik, 2020-10-28:
Welcome all.  I can confirm the problem when using nodemcu esp8266 + rf transmitter 433mhz.  the problem was detected on 3 different nodemcu esp8266, solution: either disable the rf module, or set a static ip.

closed 2021-03-11 by stale[bot]