esphome/issues #839: No RMT support for Neopixelbus Light

icefire555, 2019-11-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** Docker/Unraid

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP32 NODEMCU

**Affected component:** https://esphome.io/components/light/neopixelbus.html

**Description of problem:**
I am unable to use the RMT method on an ESP32.

**Problem-relevant YAML-configuration entries:**
```yaml
If I use "test" for my method an error will tell me all the available methods.
"
light:
 - platform: neopixelbus
   method: test
   type: BRG
   name: "GameRoom Lights"
   id: ledstrip
   num_leds: 295
   pin: GPIO2
   effects:
    - random:
    - strobe:
    - flicker:
    - addressable_rainbow:
    - addressable_color_wipe:
    - addressable_scan:
    - addressable_twinkle:
    - addressable_random_twinkle:
    - addressable_fireworks:
    - addressable_flicker:
"
I get an error 
"Unknown value 'TEST', valid options are 'ESP32_I2S_0', 'ESP32_I2S_1', 'ESP32_RMT_0', 'ESP32_RMT_1', 'ESP32_RMT_2', 'ESP32_RMT_3', 'ESP32_RMT_4', 'ESP32_RMT_5', 'ESP32_RMT_6', 'ESP32_RMT_7', 'BIT_BANG'."

If I change just the method to "method: ESP32_RMT_0" 
I get an error which I'll put in the log section.

```

**Logs (if applicable):**

```
INFO Reading configuration /config/gameroomlights.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 11, in <module>
    load_entry_point('esphome', 'console_scripts', 'esphome')()
  File "/usr/src/app/esphome/__main__.py", line 551, in main
    return run_esphome(sys.argv)
  File "/usr/src/app/esphome/__main__.py", line 529, in run_esphome
    config = read_config()
  File "/usr/src/app/esphome/config.py", line 808, in read_config
    res = load_config()
  File "/usr/src/app/esphome/config.py", line 668, in load_config
    return _load_config()
  File "/usr/src/app/esphome/config.py", line 656, in _load_config
    result = validate_config(config)
  File "/usr/src/app/esphome/config.py", line 561, in validate_config
    validated = comp.config_schema(input_conf)
  File "/usr/local/lib/python3.6/dist-packages/voluptuous/validators.py", line 208, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
  File "/usr/local/lib/python3.6/dist-packages/voluptuous/validators.py", line 283, in _exec
    v = func(v)
  File "/usr/local/lib/python3.6/dist-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.6/dist-packages/voluptuous/schema_builder.py", line 817, in validate_callable
    return schema(data)
  File "/usr/src/app/esphome/components/neopixelbus/light.py", line 77, in validate_method_pin
    pins_ = method_pins[method]
KeyError: 'ESP32_RMT_0'

```

**Additional information and things you've tried:**
I've also tried using https://github.com/Makuna/NeoPixelBus/wiki/ESP32-NeoMethods to test other methods. But it only accepts the ones listed above.

glmnet, 2019-11-11:
Seems like some definitions are missing here https://github.com/esphome/esphome/blob/f8d98ac4948b0b14702cd5a79fe2967dce452e6e/esphome/components/neopixelbus/light.py#L72

closed 2019-11-12 by OttoWinter

icefire555, 2019-11-13:
I'm still seeing the same error.
I updated the ESP Docker on unraid, and told ESPhome to do a clean build before validating the functional code so that it could update all the libraries and it didn't help.

`INFO Reading configuration /config/gameroomlights.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 11, in <module>
    load_entry_point('esphome', 'console_scripts', 'esphome')()
  File "/usr/src/app/esphome/__main__.py", line 551, in main
    return run_esphome(sys.argv)
  File "/usr/src/app/esphome/__main__.py", line 529, in run_esphome
    config = read_config()
  File "/usr/src/app/esphome/config.py", line 808, in read_config
    res = load_config()
  File "/usr/src/app/esphome/config.py", line 668, in load_config
    return _load_config()
  File "/usr/src/app/esphome/config.py", line 656, in _load_config
    result = validate_config(config)
  File "/usr/src/app/esphome/config.py", line 561, in validate_config
    validated = comp.config_schema(input_conf)
  File "/usr/local/lib/python3.6/dist-packages/voluptuous/validators.py", line 208, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
  File "/usr/local/lib/python3.6/dist-packages/voluptuous/validators.py", line 283, in _exec
    v = func(v)
  File "/usr/local/lib/python3.6/dist-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.6/dist-packages/voluptuous/schema_builder.py", line 817, in validate_callable
    return schema(data)
  File "/usr/src/app/esphome/components/neopixelbus/light.py", line 77, in validate_method_pin
    pins_ = method_pins[method]
KeyError: 'ESP32_RMT_0'`

OttoWinter, 2019-11-13:
You're still running 1.14.2, which does not include the patch.

Please wait until 1.14.3 is released.