esphome/issues #840: Bang Bang Climate for a water heater has Heat/Cool option in HA instead of Auto

spattinson, 2019-11-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant 0.100.3 esphome 1.14.2, hassbian

NOdeMCU esp8266
**Affected component:**

https://esphome.io/components/climate/bang_bang.html

**Description of problem:**
I have no cool_action defined in my esphome config yet in the HA frontend I have options Heat, Off, Heat/Cool. I believe that I should have Auto instead of Heat/Cool.

**Problem-relevant YAML-configuration entries:**
```yaml
climate:
  - platform: bang_bang
    sensor: hot_water_temperature
    name: Hot Water Boiler
    id: hot_water_boiler
    visual:
      min_temperature: 40
      max_temperature: 70
      temperature_step: 1
    default_target_temperature_low: 48 °C
    default_target_temperature_high: 52 °C
    heat_action:
      - switch.turn_on: hw_control
    idle_action:
      - switch.turn_off: hw_control
    away_config:
      default_target_temperature_low: 5 °C
      default_target_temperature_high: 8 °C

```

**Logs (if applicable):**

```
[20:11:06][C][bang_bang.climate:149]: Bang Bang Climate 'Hot Water Boiler'
[20:11:06][C][bang_bang.climate:150]:   Supports HEAT: YES
[20:11:06][C][bang_bang.climate:151]:   Supports COOL: NO
[20:11:06][C][bang_bang.climate:152]:   Supports AWAY mode: YES
[20:11:06][C][bang_bang.climate:153]:   Default Target Temperature Low: 48.0°C
[20:11:06][C][bang_bang.climate:154]:   Default Target Temperature High: 52.0°C
[20:11:06][C][bang_bang.climate:149]: Bang Bang Climate 'Hot Water Immersion'
[20:11:06][C][bang_bang.climate:150]:   Supports HEAT: YES
[20:11:06][C][bang_bang.climate:151]:   Supports COOL: NO
[20:11:06][C][bang_bang.climate:152]:   Supports AWAY mode: YES
[20:11:06][C][bang_bang.climate:153]:   Default Target Temperature Low: 48.0°C
[20:11:06][C][bang_bang.climate:154]:   Default Target Temperature High: 52.0°C
[20:11:06][C][dallas.sensor:070]: DallasComponent:
[20:11:06][C][dallas.sensor:071]:   Pin: GPIO5 (Mode: INPUT)
[20:11:06][C][dallas.sensor:072]:   Update Interval: 60.0s
[20:11:06][D][dallas.sensor:077]:   Found sensors:
[20:11:06][D][dallas.sensor:080]:     0x8502169177D30428

[20:42:12][D][climate:172]: 'Hot Water Boiler' - Sending state:
[20:42:12][D][climate:175]:   Mode: AUTO
[20:42:12][D][climate:177]:   Action: OFF
[20:42:12][D][climate:180]:   Current Temperature: 52.38°C
[20:42:12][D][climate:184]:   Target Temperature: Low: 52.00°C High: 52.00°C
[20:42:12][D][climate:189]:   Away: OFF
[20:42:14][D][climate:010]: 'Hot Water Boiler' - Setting
[20:42:14][D][climate:020]:   Target Temperature Low: 48.00
[20:42:14][D][climate:023]:   Target Temperature High: 52.00
[20:42:14][D][climate:172]: 'Hot Water Boiler' - Sending state:
[20:42:14][D][climate:175]:   Mode: AUTO
[20:42:14][D][climate:177]:   Action: OFF
[20:42:14][D][climate:180]:   Current Temperature: 52.38°C
[20:42:14][D][climate:184]:   Target Temperature: Low: 48.00°C High: 52.00°C
[20:42:14][D][climate:189]:   Away: OFF

```
In .storage the configuration file that esphome generated contains:
```
{
    "data": {
        "climate": [
            {
                "key": 661990594,
                "name": "Hot Water Boiler",
                "object_id": "hot_water_boiler",
                "supported_modes": [
                    1,
                    0,
                    3
                ],
                "supports_away": true,
                "supports_current_temperature": true,
                "supports_two_point_target_temperature": true,
                "unique_id": "hwclimatehot_water_boiler",
                "visual_max_temperature": 70.0,
                "visual_min_temperature": 40.0,
                "visual_temperature_step": 1.0
            },
            {
                "key": 221469590,
                "name": "Hot Water Immersion",
                "object_id": "hot_water_immersion",
                "supported_modes": [
                    1,
                    0,
                    3
                ],
                "supports_away": true,
                "supports_current_temperature": true,
                "supports_two_point_target_temperature": true,
                "unique_id": "hwclimatehot_water_immersion",
                "visual_max_temperature": 60.0,
                "visual_min_temperature": 40.0,
                "visual_temperature_step": 1.0
            }
        ],
        "device_info": {
            "compilation_time": "Nov  9 2019, 09:42:53",
            "esphome_version": "1.14.2",
            "has_deep_sleep": false,
            "mac_address": "84:F3:EB:0A:9A:0E",
            "model": "PLATFORMIO_NODEMCUV2",
            "name": "hw",
            "uses_password": false
        },
```
I Believe the supported modes should be 0, 1 and 2 so that Auto appears in the front end instead of Heat/Cool. From the Logs esphome thinks it is in Auto mode.

**Additional information and things you've tried:**

Its working fine as Heat/Cool, its a display usability issue only.

One other thing I noticed is that the graph that is displayed in frontend is not coloured to show when heating is active like it was when I had it configured as a generic_thermostat in HA. The graph IS coloured when it is set to Heat though, I thought Heat = Automatic for a heater when presented with a choice of Off, Heat, and Heat/Cool.

glmnet, 2019-11-10:
This is correct behavior per HA design. I still can't believe they decided to name "Heat / Cool", this "mode" instead of "Auto", which is how many devices call this mode.
I hadn't realized that your heat only, or cool only device goes to "Heat / Cool" mode when the device only supports one mode, this is just.... 
And of course is confusing lots of people for sure.
If you want to learn more you'll have to search for this issue at HA issue tracker or forums.
I'm closing this as you say it is just a display usability issue only.

spattinson, 2019-11-11:
I have another climate integration using vaillant component. That has heat/off/auto.see these two screenshots of vaillant component "idle" and "heating"  home assistant does not know when esphome is heating or idle. In the second screenshot i set a manual setpoint, the climate has changed from auto to heat and the state has changed from idle to heating, heat, manual. I have a feeling that having high/low setpoints implies heat/cool capability in home assistant even if there is no cool_action. my vailant component has only one setpoint not high/low and handles hysteresis itself by modulating boiler. I get it that bang bang works like this. I will live with it though, wait for someone else who can define the issue more precisely. Thanks
[image]
[image]

glmnet, 2019-11-11:
I understand your frustration, I even feel it myself.

You'll find the tip of this iceberg here:
https://github.com/home-assistant/architecture/issues/260

All you say is expected behavior:
The Vaillant when in auto mode is not accepting a Target temperature (set point) while ESPHome accepts a Target temperature in Heat / Cool.
Two point target temperature is unrelated.
Climate action (heating, cooling) should be informed by ESPHome to HA, see https://github.com/esphome/esphome/pull/720, if not open an issue.

Remember this is open source software and not everyone is going to review your issue, you are supposed to investigate and contribute, there is lots of information about this in HA community.
I'm reopening this issue so "someone else who can define the issue more precisely" sees this, but I encourage you to close it.

spattinson, 2019-11-11:
Thanks. my understanding is improved. I think I will set a schedule of temperatures rather than use away mode at night. By doing that I am able to emulate auto mode, and I can alter a setpoint with the full knowledge that it will return to 'auto' mode at the next scheduled change. Thanks I do appreciate all the hard work put into this project. Its awesome. I should improve my programming skills so that I can contribute too. Closing.

closed 2019-11-11 by spattinson