esphome/issues #841: Crash at boot due to exception if using the DHT sensor library

heckler, 2019-11-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

- ESPHome version 1.14.2
- Windows 10 version 1809 (build 17763.805)
- Windows System for Linux (WSL) - Ubuntu 18.04.2 LTS
- Python 3.6.8, pip 9.0.1

**ESP (ESP32/ESP8266, Board/Sonoff):**

- ESP8266, NodeMCU "v3" LoLin

**Affected component:**

- [DHT sensor library](https://esphome.io/components/sensor/dht.html)

**Description of problem:**

Unable to utilize the DHT sensor library; as soon as I add it to my configuration, the device crashes at boot and enters a reboot loop.  I've managed to reproduce this issue with a very minimal YAML (below) and confirmed the behavior in 3 different ESP8266 boards (two "LoLin" and one V2).

**Problem-relevant YAML-configuration entries:**

```yaml
esphome:
  name: uptime_test
  platform: ESP8266
  board: nodemcuv2
 
logger:
  level: VERBOSE
  
sensor:
  - platform: dht
    pin: D5
    update_interval: 60s
```

**Note:** this is the minimal YAML config which reproduces this issue.  I've omitted all other settings (WiFi, etc.) on purpose, so that I could capture the logs with the bare minimal config which would reproduce this issue.  Logs below:

**Logs (if applicable):**

```
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
[I][logger:156]: Log initialized
[I][app:028]: Running through setup()...
[V][app:029]: Sorting components by setup priority...
[C][dht:011]: Setting up DHT...
[I][app:058]: setup() finished successfully!
[W][dht:058]: Invalid readings! Please check your wiring (pull-up resistor, pin number) and consider manually specifying the DHT model using the model option.
Fatal exception 29(StoreProhibitedCause):
epc1=0x40201d1d, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000018, depc=0x00000000

 Exception (29):
epc1=0x40201d1d epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

>>>stack>>>
ctx: cont
sp: 3ffffd10 end: 3fffffc0 offset: 01a0
    3ffffeb0:  00000000 00013dbb 3ffee4f0 40203490
    3ffffec0:  3ffe85cf 3ffe85cf 00000050 00000000
    3ffffed0:  3ffef3b4 3ffef44c 7fc00000 402017d4
    3ffffee0:  3fffff10 3fffff00 401003de 4f9db22d
    3ffffef0:  7fc00000 7fc00000 00000010 40203944
    3fffff00:  3ffef49c 3ffef44c 3ffee4f0 402071dc
    3fffff10:  3fffdad0 00000000 3ffee4f0 40100210
    3fffff20:  00000000 00000000 4bc6a7f0 00000000
    3fffff30:  00000050 00000002 401003de 4d4fdf3b
    3fffff40:  00000000 4020efa2 3ffef3ec 00000050
    3fffff50:  3fffdad0 00000000 3ffee4f0 40201f5c
    3fffff60:  3ffee554 00000000 3ffef404 40203c76
    3fffff70:  3ffeee78 feefeffe feefeffe feefeffe
    3fffff80:  3ffef3b4 feefeffe feefeffe 3ffee728
    3fffff90:  3fffdad0 00000000 3ffee6f8 40203c94
    3fffffa0:  3fffdad0 00000000 3ffee6f8 402051dc
    3fffffb0:  feefeffe feefeffe 3ffe8530 401006ed
<<<stack<<<

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
```

**Note:** the stack dump above had weird formatting when I copied it from Putty; I've removed leading spaces from each log line in order to format it. No other changes have been made to the logs.

**Additional information and things you've tried:**

Tried commenting-out the DHT sensor block, and then the device boots normally and just outputs these logs, as expected:

```
SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
[I][logger:156]: Log initialized
[I][app:028]: Running through setup()...
[V][app:029]: Sorting components by setup priority...
[I][app:058]: setup() finished successfully!
[I][app:100]: ESPHome version 1.14.2 compiled on Nov 11 2019, 13:52:20
[C][logger:175]: Logger:
[C][logger:176]:   Level: VERBOSE
[C][logger:177]:   Log Baud Rate: 115200
[C][logger:178]:   Hardware UART: UART0
```

Also, I've collected the crash logs using putty via the USB-UART connection.  It seems from other issue threads that there should be a way to run `esphome config.yaml logs` over a serial line in order to get better-formatted exception logs, but I haven't found the relevant information on how to specify a serial port for the `logs` command.  Instructions are welcome! :-)

glmnet, 2019-11-11:
ahh yes you are not definining

````yaml

    temperature:
      name: "Living Room Temperature"
    humidity:
      name: "Living Room Humidity"

````

and this is causing the crashes..
@Otto should those be non optional at vol checks or otherwise null checks needs to be added on C++ side.

heckler, 2019-11-11:
Oh, I see!  Let me explain how I fell into this trap ...  When I stumbled upon this issue, I had two dht sensors in the config, but only one of them had both temperature and humidity, the other one had only humidity.  When I saw it crashing I started removing config entries to the bare minimum which would still reproduce the crash.  Hence the bare bones config above ...

I can confirm that putting back both temperature.name and humidity.name on both of my dht sensor entries makes the generated code happy and all **works fine now**.

I like @glmnet idea of making these config entries mandatory to prevent this mistake ;)

closed 2019-11-12 by OttoWinter