esphome/issues #843: PZEM-004T report frequency/power factor as Hz/PF

Fusseldieb, 2019-11-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Not relevant

**ESP (ESP32/ESP8266, Board/Sonoff):**

Not relevant

**Affected component:**

https://esphome.io/components/sensor/pzem004t.html

**Description of problem:**
I don't know if I should ask this as a bug or feature request, but basically it would be nice to report PZEMs frequency as "Hz" instead of nothing. The thing is, when Home Assistant has a unit_of_measurement is displays useful graphs, but when it has none, it makes a lot of slices, rendering histories with a lot of changes useless, which is this case.

[image]

Not only that, but the browser lags _massively_ trying to make this "slice-graph", instead of the normal line-graph. It takes 40+ seconds for it to load, while others with the same amount of info take less than 3 sec.

[image]

All this can be solved using `unit_of_measurement`s. I would really appreciate it.

I would suggest Hz for frequency and PF for power factor.

**Problem-relevant YAML-configuration entries:**
```yaml
```

**Logs (if applicable):**

```
```

**Additional information and things you've tried:**
Theoretically I could make a template sensor inside HA, but I think this should be done inside the conponent itself.

OttoWinter, 2019-11-12:
For frequency you're right, that should be Hz by default.

For power factor, I don't agree so much. Technically, power factor does not have a unit (it's a ratio), the only valid unit would be `1` mathematically speaking.

Fusseldieb, 2019-11-12:
Wow! tHaNkS so much for this fast PR! You're awesome @OttoWinter :)

Well, about the Power Factor. "1" as a unit would confuse/annoy people, for sure. But PF is just the name itself. It just needs to be *something* for the graph to show up.

closed 2019-11-12 by OttoWinter