esphome/issues #845: D1 Mini OTA upload "An existing connection was forcibly closed by the remote host"

dseven, 2019-11-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Windows 10, pip (both python 2.7 and 3.7)

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266/D1 Mini

**Affected component:**
Not sure?

**Description of problem:**
Attempting to upload OTA to a D1 Mini consistently fails as below (see logs section). Upload over serial works fine. Logs OTA works fine. The device is talking to HA fine. OTA upload to a NodeMCU ESP32S using the same method / environment has worked recently.

**Problem-relevant YAML-configuration entries:**
```
esphome:
  name: d1_mini_1
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "XXX"
  password: "XXX"

logger:
api:
ota:
```

**Logs (if applicable):**

```
INFO Reading configuration .\d1_mini_1.yaml...
Found multiple options, please choose one:
  [1] COM8 (USB-SERIAL CH340 (COM8))
  [2] Over The Air (d1_mini_1.local)
(number): 2
INFO Resolving IP address of d1_mini_1.local
INFO Address not available when adding 169.254.203.153 to multicast group, it is expected to happen on some systems
INFO Address not available when adding 169.254.222.51 to multicast group, it is expected to happen on some systems      INFO Address not available when adding 192.168.2.1 to multicast group, it is expected to happen on some systems
INFO  -> 192.168.10.112
INFO Uploading .\d1_mini_1\.pioenvs\d1_mini_1\firmware.bin (364544 bytes)
Uploading: [==                                                          ] 3%
ERROR Error sending data: [WinError 10054] An existing connection was forcibly closed by the remote host
```

**Additional information and things you've tried:**
This is the first D1 Mini that I've tried to use. Hope I'm not doing something stupid :/

Oh, and it fails the same way if I specify "use_address" in the config. It seems to connect OK, and start uploading, but then the device drops the connection for some reason.

dseven, 2019-11-13:
It just occurred to me to hook up a serial connection. Here's what it says:

```[D][ota:072]: Starting OTA Update from 192.168.10.63...
[D][ota:243]: OTA in progress: 0.3%
[W][ota:233]: Error writing binary data to flash: 0 != 1024!
[W][ota:276]: Update end failed! Error: ERROR[2]: Flash Erase Failed
```

dseven, 2019-11-13:
https://community.home-assistant.io/t/ota-update-error-104-flash-erase-failed/140717 looks like the same issue(?)

OttoWinter, 2019-11-13:
Looks like you have an issue with the flash chip on the device. Erasing some flash sectors is failing with an error.

Flash erase is handled by the arduino framework - so it would be a bug there if anything.

If you have another ESP8266 chip around, try with that. If it works, then it's likely a hardware fault. Also post what hardware board it is you have exactly - there are many wemos d1 minis around (and mostly fake ones).

brandond, 2019-11-13:
Maybe try the reduced tx power fix too? So many low quality D1 Mini clones.

dseven, 2019-11-13:
This is the hardware I'm trying to use:

https://www.amazon.com/dp/B07BK435ZW/ref=cm_sw_em_r_mt_dp_U_HhaZDbDMYXFKQ

I have two of them, but haven't soldered headers on the other one yet. I'll try it a bit later. 

I'm struggling to see how it could be a bad flash chip when I'm (consistently) able to upload to it over USB/Serial..... unless it is some issue with the WiFi radio interfering. That seems like a stretch. but I'll try the fix ... got a link handy? Google's failing me so-far....

dseven, 2019-11-13:
I think I figured out my dumb mistake - I was using board type "d1_mini", where it should be "d1_mini_lite". Seems to work now .. well, it has worked once, at least.

I guess the flash size is detected when using USB/serial, but not when using OTA ... ??

dseven, 2019-11-13:
Seems fine now. Apologies for the noise.

closed 2019-11-13 by dseven

brandond, 2019-11-13:
Yeah, flashing via USB and OTA use completely different methods. USB directly overwrites the user code, while OTA stores a temporary copy in flash, reboots into a bootloader which then copies the temporary code into place, and then reboots again.

And yes, there are known issues with wifi interfering with flash access. I'm glad it's not what you were running into though, as it's harder to fix.
https://github.com/esphome/issues/issues/455#issuecomment-552976310

delfer, 2019-11-20:
Have the same errors.

On esphome:
```
Uploading: [=                                                           ] 2% 
ERROR Error sending data: [Errno 104] Connection reset by peer
```

On serial:
```
D][ota:072]: Starting OTA Update from 192.168.1.6...
[D][ota:243]: OTA in progress: 0.3%
[W][ota:233]: Error writing binary data to flash: 0 != 536!
[W][ota:276]: Update end failed! Error: ERROR[2]: Flash Erase Failed

[D][ota:072]: Starting OTA Update from 192.168.1.6...
[D][ota:243]: OTA in progress: 0.1%
[W][ota:233]: Error writing binary data to flash: 0 != 536!
[W][ota:276]: Update end failed! Error: ERROR[2]: Flash Erase Failed

[D][ota:072]: Starting OTA Update from 192.168.1.6...
[D][ota:243]: OTA in progress: 0.1%
[W][ota:233]: Error writing binary data to flash: 0 != 1024!
[W][ota:276]: Update end failed! Error: ERROR[2]: Flash Erase Failed
```
I have `board = esp07`, and after found in table http://docs.platformio.org/en/latest/platforms/espressif8266.html#nodemcu that it have 4Mb flash.
But when fleshing by serial port it writes:
```
Auto-detected Flash size: 1MB
```
I just changed board to `esp01_1m` and now OTA works fine!

OttoWinter, 2019-11-20:
@delfer Oh ok that might have to do with weird flash chips (puya) or a fake device (there are almost no genuine wemos/nodemcu devices buyable anymore) with less flash than advertised.