esphome/issues #847: Custom Sensor Component with multiple output values does not create the UI elements

manju-rn, 2019-11-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32s
**Affected component:**

https://esphome.io/components/sensor/custom.html
Bonus: Sensors With Multiple Output Values

**Description of problem:**
When a Custom Component is created with the multiple output values, the UI elements for the same is not created. I mean the integration with the Home Assistant does not show up. If only one component is used, the UI shows up perfectly.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
```
// Not working
sensor:
- platform: custom
  lambda: |-
    iTagSensor *itag_sensor = new iTagSensor();
    App.register_component(itag_sensor);
    return {itag_sensor->button_sensor, itag_sensor->battery_sensor};
    
  sensors:
  - name: "iTagSensor"
  - name: "iTagBattery"
// The iTagSensor and iTagBattery are not shown up. In the logs, only when the iTag button is clicked, the values are shown up in the logs

// Working
sensor:
- platform: custom
  lambda: |-
    iTagSensor *itag_sensor = new iTagSensor();
    App.register_component(itag_sensor);
    return {itag_sensor};
    
  sensors:
  - name: "iTagSensor"

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
[14:53:47][D][sensor:092]: '': Sending state 0.00000  with 0 decimals of accuracy
[14:53:48][D][sensor:092]: '': Sending state 0.00000  with 0 decimals of accuracy
[14:53:49][D][sensor:092]: 'iTagSensor': Sending state 1.00000  with 0 decimals of accuracy
[14:53:50][D][sensor:092]: 'iTagBattery': Sending state 99.0  with 0 decimals of accuracy

```

**Additional information and things you've tried:**

brandond, 2019-11-14:
In the non working examples, your lambda is returning Sensors, not Components. You have to register the component, not the sensors.

OttoWinter, 2019-11-14:
You need to return the sensor component also, see below:

```cpp
return {itag_sensor, itag_sensor->button_sensor, itag_sensor->battery_sensor};
```

Otherwise ESPHome only knows of the button and battery sensor, not the parent itag_sensor.

closed 2019-11-14 by OttoWinter

manju-rn, 2019-11-14:
okay thanks.  The docs were incorrect then.  I create a PR #414