esphome/issues #848: PCF8575 not working since INPUT_PULLUP removed 1.14.1

Neil-M0NFI, 2019-11-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio running on Ubuntu server

**ESP (ESP32/ESP8266, Board/Sonoff):**

D1 mini 

**Affected component:**

https://esphome.io/components/pcf8574.html

Using PCF8575 with all 16 GPIO as inputs with 10k pull-up resistors. 

**Description of problem:**

I made a change to remove the second wifi network and had to remove INPUT_PULLUP to compile. Once uploaded the inputs give unreliable switching states. When I tested the hardware before installation I found I had to use INPUT_PULLUP to give reliable switching. 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

i2c:
  sda: GPIO4
  scl: GPIO5
  scan: True

pcf8574:
  - id: 'pcf8575_1'
    address: 0x20
    pcf8575: True

binary_sensor:
  - platform: status
    name: "Alarm Status"

  - platform: gpio
    device_class: door
    name: "Door 1"

    pin:
      pcf8574: pcf8575_1
      number: 0
      mode: INPUT_PULLUP
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2019-11-14:
Might be related to https://github.com/esphome/issues/issues/809#issuecomment-549124604

Neil-M0NFI, 2019-11-14:
That was exactly the problem, thank you and sorry for not seeing the other issue before I opened this one.

closed 2019-11-14 by Neil-M0NFI