esphome/issues #849: rf transmitter sends to long sync bit with NodeMCU-ESP32

KarlNap, 2019-11-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io V0.100.2 on RaspberryPi 3B+ 
**ESP (ESP32/ESP8266, Board/Sonoff):**

AZDelivery NodeMCU32 with ESP-WROOM-32
https://cdn.shopify.com/s/files/1/1509/1638/files/ESP_-_32_NodeMCU_Developmentboard_Datenblatt_a3bf98d8-6a53-4d26-8f1c-c61b1c82af39.pdf?76837
**Affected component:**

remote_transmitter

**Description of problem:**
I use the remote_transmitter to switch a radio socket. The sync bit transmitted over 433MHz is too long.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: dev
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: !secret ssid
  password: !secret wifi
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: xxx
    # Set this to the IP address of the router. Often ends with .1
    gateway: xxx
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

remote_transmitter:
  pin: GPIO5
    # RF uses a 100% carrier signal
  carrier_duty_percent: 100%
  
switch:
  - platform: template
    name: "socket 1"
    id: socket_1
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '001001100001011100011100'
          protocol:
            pulse_length: 374
            sync: [1,6]
            zero: [1,3]
            one: [3,1]
            inverted: false
          repeat: 
            times: 3
    turn_off_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '001011001010111100101100'
          protocol:
            pulse_length: 374
            sync: [1,6]
            zero: [1,3]
            one: [3,1]
            inverted: false
          repeat: 
            times: 3
  - platform: template
    name: "socket 2"
    id: socket_2
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '001001111111011011110101'
          protocol:
            pulse_length: 374
            sync: [1,6]
            zero: [1,3]
            one: [3,1]
            inverted: false
          repeat: 
            times: 3
    turn_off_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '001010010110101111010101'
          protocol:
            pulse_length: 374
            sync: [1,6]
            zero: [1,3]
            one: [3,1]
            inverted: false
          repeat: 
            times: 3
```
I use 
**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

I use an [RFSniffer](https://github.com/ninjablocks/433Utils/tree/master/RPi_utils) on a Raspberry Pi to read the signal. 
```
GENERAL
=======
Received 2496284
Bitlength 24
Protocol 1
Delay(pulse length in us) 374
(delay provided from rc-switch)

STATISTICS
==========
Duration of data bits in pulse train :  36702
Data bit duration = 1529 and standard deviation = 157.75
Coefficient of variation of data-bit duration = 10.32 % (should be less than 10%)
Do not use the rest of the information if big coefficient of variation
Long-to-short duration ratio for data bits (rounded) = 3
Sync bit (in multiples of the pulseLength) = 1 34

Proposed protocol for RCswitch
{ 374, { 1, 34 }, { 1, 3 }, { 3, 1 }, false }
```

Everything is correct except the sync bit with 1, 34. I have also tried to find a solution by reducing the sync bit settings. Unfortunately without success. How can I best proceed here to find the error? The same yaml configuration runs with the same transmitter on a Heltech `heltec_wifi_lora_32`

OttoWinter, 2019-11-14:
This is because of the default value of `wait_time` in repeat.

Try this:

```yaml
      - remote_transmitter.transmit_rc_switch_raw:
          code: '001011001010111100101100'
          protocol:
            pulse_length: 374
            sync: [1,6]
            zero: [1,3]
            one: [3,1]
            inverted: false
          repeat: 
            times: 3
            wait_time: 0s
```

KarlNap, 2019-11-14:
Thanks, that solved my problem.

closed 2019-11-14 by KarlNap