esphome/issues #850: "Cannot connect to MQTT broker: Int or String expected" (hass.io addon)

Joshfindit, 2019-11-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
Home Assistant `0.101.3`
ESPHome Addon `1.15.0-dev`

**ESP (ESP32/ESP8266, Board/Sonoff):**

Happens to all of them
- D1 Mini
- ESP32 Dev

**Affected component:**

https://esphome.io/components/mqtt.html

**Description of problem:**

Devices show online, I can upload code, I can't get the debug info after the upload is complete (but the device accepts the upload fine).
The device reboots, attaches to the MQTT server (also running as a hass.io addon), and accepts commands as expected.

But I can't get the log by pressing "SHOW LOGS" as this happens:

```
INFO Reading configuration /config/esphome/lamp2.yaml...
INFO Starting log output from lamp2/debug
ERROR Cannot connect to MQTT broker: Int or String expected
```
(Note: the error line is the same when uploading is successful)

Recently started happening, not sure when as I just ignored it for a while.

**Problem-relevant YAML-configuration entries:**
```yaml
# "Bed light"

substitutions:
  devicename: lamp2
  upper_devicename: Lamp 2
  ledpin: GPIO19
  pushbutton_pin: GPIO14
  touchpin: GPIO27
  touchname:  "ESP32 Touch Pad GPIO27"

esphome:
  name: $devicename
  platform: ESP32
  board: esp32dev
  on_boot:
    priority: 90
    then:
      - switch.turn_off: onboard_led

mqtt:
  broker: 192.168.7.20
  port: 1883
  username: REDACTED
  password: REDACTED

wifi:
  ssid: "REDACTED"
  password: "REDACTED"

# Enable Home Assistant API
# api:

# Enable logging
logger:
  level: DEBUG

# Enable OTA programming
ota:

# ------ Done basic setup

# ----

light:
  - platform: monochromatic
    name: $upper_devicename
    id: mainlight
    output: output_component1
    default_transition_length: 500ms
    effects:
      - flicker:
          name: Flicker
          alpha: 95%
          intensity: 1.5%
      - strobe:
          name: Strobe
          colors:
            - state: True
              brightness: 100%
              red: 100%
              green: 100%
              blue: 100%
              white: 100%
              duration: 80ms
            - state: False
              duration: 80ms

esp32_touch:
  setup_mode: false
  
  

binary_sensor:
    - platform: esp32_touch
      name: $touchname
      id: maintouch
      pin: $touchpin
      threshold: 500
 
      on_multi_click:
        - timing:
            - ON for at most 0.1s
            - OFF for at least 0.3s
          then:
            - logger.log: "Single Short Clicked"
            - light.turn_off:
                id: mainlight
                transition_length: 0s
      on_press:
        - while:
            condition:
              binary_sensor.is_on: maintouch
            then:
              - light.dim_relative:
                  id: mainlight
                  relative_brightness: 2%
                  transition_length: 0.05s
              - delay: 0.05s

output:
  - platform: ledc
    id: output_component1
    pin: $ledpin
    frequency: 25000Hz
    # bit_depth: 10

switch:
  - platform: gpio
    pin: GPIO2
    name: "Onboard LED"
    id: onboard_led
    internal: false
    inverted: yes

```

**Logs (if applicable):**

```
No log available
```

**Additional information and things you've tried:**

Reinstalling the addon

*To be honest, I suspect this is a networking glitch between the addons, but it being Hass.io, I don't have tools to do troubleshooting at that level.

mkono87, 2019-11-16:
Yeah same issue here after doing the latest update, have to switch to api for now.

ssieb, 2019-11-18:
That fix is incorrect.  It's the port that's the problem, not the host.  The port value has the type "esphome.yaml_util.ESPInt".

mkono87, 2019-11-18:
True I am not sure the standard 1883 port

brandond, 2019-11-18:
According to the yaml file in the original post, the default port is being used:
```yaml
mqtt:
  broker: 192.168.7.20
  port: 1883
  username: REDACTED
  password: REDACTED
```

You should be able to just remove it until the issue is fixed.

mkono87, 2019-11-18:
But then the device is rendered useless unless I use Api: for now.

ssieb, 2019-11-18:
The OP used the default port, but for others that don't (like @mkono87), there's no workaround other than switching to using the api or serial instead.

OttoWinter, 2019-11-20:
Uh yes, true - the issue is not fixed yet. Sorry about that.

closed 2019-12-04 by OttoWinter