esphome/issues #851: Ble scan settings

Oliviakrkk, 2019-11-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

esp32

**Affected component:**

esp32_ble_tracker

**Description of problem:**

Scan parameters are different from those set in config

**Problem-relevant YAML-configuration entries:**
```yaml
esp32_ble_tracker:
  scan_parameters:
     window: 30ms
```

**Logs (if applicable):**

```
[14:24:46][C][esp32_ble_tracker:442]: BLE Tracker:
[14:24:46][C][esp32_ble_tracker:443]:   Scan Duration: 300 s
[14:24:46][C][esp32_ble_tracker:444]:   Scan Interval: 512 ms
[14:24:46][C][esp32_ble_tracker:445]:   Scan Window: 48 ms
[14:24:46][C][esp32_ble_tracker:446]:   Scan Type: ACTIVE
```

**Additional information and things you've tried:**

ESPHome version 1.14.2

Oliviakrkk, 2019-11-14:
To be honest I have a really big issue with ble presence detection.  I have ble scanner app on the Android phone and it shows iNode with ~70-80 RSSI but esp32 does not see it at all. Is that normal?

I tried to play with scan settings but whatever I set it is worse than not setting at all and running on default...Which is:
```
[13:39:05][C][esp32_ble_tracker:442]: BLE Tracker:
[13:39:05][C][esp32_ble_tracker:443]:   Scan Duration: 300 s
[13:39:05][C][esp32_ble_tracker:444]:   Scan Interval: 512 ms
[13:39:05][C][esp32_ble_tracker:445]:   Scan Window: 320 ms
[13:39:05][C][esp32_ble_tracker:446]:   Scan Type: ACTIVE
```

Also why window is different than the one from config file?

brandond, 2019-11-14:
The configured interval and window are both divided by 0.625 for some reason:

https://github.com/esphome/esphome/blob/dev/esphome/components/esp32_ble_tracker/__init__.py#L104

OttoWinter, 2019-11-14:
Yeah it's an issue in the log config dump - the values themselves are right.

esp-idf accepts these values in units of 0.625 milliseconds - hence the division

closed 2019-11-17 by OttoWinter