esphome/issues #852: Tuya dimmer turns off immideately

Deinara, 2019-11-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Ubuntu Server host.
Running ESPHome in a Docker container, with Home Assistant in another Docker container.
ESPHome 1.14.2
Home Assistant 0.101.3

**ESP (ESP32/ESP8266, Board/Sonoff):**
I believe a version of ESP8266 labelled as:
P/N: TYWE3S
Model: WF-DS01

It's a EU-size tuya dimmer, marked as YDM_WIFI_SV5
[image]

**Affected component:**
https://esphome.io/components/light/tuya.html

**Description of problem:**
The dimming function itself works as expected, I can set any value between 0 and 100% using the Home Assistant interface. 
When it is on, I can use the plus and minus buttons on the device to change the brightness of the bulb attached. I can also turn the lamp off using the middle (power) button.

However, when the light is off and I press the power button once, the light bulb instantly turns on at full brightness (despite what the setting was when it was turned off) and starts gradually dimming to 0% instantly.

Weird things I have noticed:
- The light changes to ON (twice), then OFF in the logs even though the button is pressed only once.
- When the dimmer is power cycled, the middle (power) button sometimes correctly turns on the bulb, without instantly starting to dim it. But the original behaviour returns after this first time.
- You'll notice the last line in the logs mentions ```Brightness: 100%``` even though the light was just dimmed to 0%. Despite the line saying 100%, the light is off (and didn't flicker on between being dimmed and turning off completely).
- Every other line in the log seems to mention ```'sh_tuyd_15b9_dimmer' Setting:``` without actually showing any setting.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  device_name: "sh_tuyd_15b9"

esphome:
  name: ${device_name}
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: [redacted]
  password: [redacted]
  reboot_timeout: 23h

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  password: [redacted]
  reboot_timeout: 23h

ota:
  password: [redacted]

# My dimmer used the hardware serial port on the alternate pins
uart:
  rx_pin: GPIO03
  tx_pin: GPIO01
  baud_rate: 9600

# Register the Tuya MCU connection
tuya:

light:
  - platform: "tuya"
    id: dimmer
    name: "${device_name}_dimmer"
    dimmer_datapoint: 2
    switch_datapoint: 1
    min_value: 30
    
sensor:
# Extra sensor to keep track of plug uptime
  - platform: uptime
    name: ${device_name}_uptime
    icon: mdi:clock-outline
```

**Logs (if applicable):**
I did not connect the device via USB.
```
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from sh_tuyd_15b9.local using esphome API
INFO Connecting to sh_tuyd_15b9.local:6053 (192.168.1.127)
INFO Successfully connected to sh_tuyd_15b9.local
[16:43:52][I][app:100]: ESPHome version 1.14.2 compiled on Nov 14 2019, 16:43:20
[16:43:52][C][wifi:409]: WiFi:
[16:43:52][C][wifi:277]:   SSID: [redacted]
[16:43:52][C][wifi:278]:   IP Address: 192.168.1.127
[16:43:52][C][wifi:280]:   BSSID: [redacted]
[16:43:52][C][wifi:281]:   Hostname: 'sh_tuyd_15b9'
[16:43:52][C][wifi:285]:   Signal strength: -43 dB ▂▄▆█
[16:43:52][C][wifi:289]:   Channel: 7
[16:43:52][C][wifi:290]:   Subnet: 255.255.255.0
[16:43:52][C][wifi:291]:   Gateway: 192.168.1.1
[16:43:52][C][wifi:292]:   DNS1: 192.168.1.1
[16:43:52][C][wifi:293]:   DNS2: (IP unset)
[16:43:52][C][uart:145]: UART Bus:
[16:43:52][C][uart:147]:   TX Pin: GPIO1
[16:43:52][C][uart:150]:   RX Pin: GPIO3
[16:43:52][C][uart:152]:   Baud Rate: 9600 baud
[16:43:52][C][uart:153]:   Stop bits: 1
[16:43:52][C][uart:155]:   Using hardware serial interface.
[16:43:52][C][tuya:024]: Tuya:
[16:43:52][C][tuya:026]:   GPIO MCU configuration not supported!
[16:43:52][C][tuya:029]:   Datapoint 1: switch (value: OFF)
[16:43:52][C][tuya:031]:   Datapoint 2: int value (value: 0)
[16:43:53][C][uptime.sensor:030]: Uptime Sensor 'sh_tuyd_15b9_uptime'
[16:43:53][C][uptime.sensor:030]:   Unit of Measurement: 's'
[16:43:53][C][uptime.sensor:030]:   Accuracy Decimals: 0
[16:43:53][C][uptime.sensor:030]:   Icon: 'mdi:clock-outline'
[16:43:53][C][logger:175]: Logger:
[16:43:53][C][logger:176]:   Level: DEBUG
[16:43:53][C][logger:177]:   Log Baud Rate: 0
[16:43:53][C][logger:178]:   Hardware UART: UART0
[16:43:53][C][light:177]: Light 'sh_tuyd_15b9_dimmer'
[16:43:53][C][light:179]:   Default Transition Length: 0.0s
[16:43:53][C][light:180]:   Gamma Correct: 1.00
[16:43:53][C][tuya.light:027]: Tuya Dimmer:
[16:43:53][C][tuya.light:029]:    Dimmer has datapoint ID 2
[16:43:53][C][tuya.light:031]:    Switch has datapoint ID 1
[16:43:53][C][ota:029]: Over-The-Air Updates:
[16:43:53][C][ota:030]:   Address: sh_tuyd_15b9.local:8266
[16:43:53][C][ota:032]:   Using Password.
[16:43:53][C][api:095]: API Server:
[16:43:53][C][api:096]:   Address: sh_tuyd_15b9.local:6053
[16:44:30][D][sensor:092]: 'sh_tuyd_15b9_uptime': Sending state 54.43400 s with 0 decimals of accuracy
[16:45:00][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:00][D][light:273]:   State: ON
[16:45:00][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:00][D][light:277]:   Brightness: 100%
[16:45:00][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:00][D][light:277]:   Brightness: 100%
[16:45:00][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:00][D][light:273]:   State: ON
[16:45:00][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:00][D][light:277]:   Brightness: 100%
[16:45:00][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:00][D][light:273]:   State: OFF
[16:45:00][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:00][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:00][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:00][D][light:277]:   Brightness: 100%
[16:45:00][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:00][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:00][D][light:277]:   Brightness: 96%
[16:45:00][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:00][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:00][D][light:277]:   Brightness: 95%
[16:45:00][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:00][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:00][D][light:277]:   Brightness: 94%
[16:45:00][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:

[Snipped for redundancy]

[16:45:07][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:07][D][light:277]:   Brightness: 5%
[16:45:07][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:07][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:07][D][light:277]:   Brightness: 4%
[16:45:07][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:07][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:07][D][light:277]:   Brightness: 3%
[16:45:07][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:07][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:07][D][light:277]:   Brightness: 2%
[16:45:07][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:07][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:07][D][light:277]:   Brightness: 1%
[16:45:07][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:07][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[16:45:07][D][light:277]:   Brightness: 100%
```

**Additional information and things you've tried:**
As the issue seems like it could be related to 1% meaning 100% as suggested in this comment:
[https://github.com/esphome/feature-requests/issues/352#issuecomment-541526433](https://github.com/esphome/feature-requests/issues/352#issuecomment-541526433)
I set `min_value: 20` and later increased it to 30, to no avail

brandond, 2019-11-14:
Sounds like we need an invert option for dimmers to flip the scale.

Can you try the dev branch and share the logs? I don't think it will fix it but I'm curious to see the output.

Deinara, 2019-11-14:
I'll try out the dev branch in a bit.

For now I've been messing around with the Tuya datapoints:
The Tuya component reports:
```[tuya:024]: Tuya:
[tuya:026]:   GPIO MCU configuration not supported!
[tuya:029]:   Datapoint 1: switch (value: OFF)
[tuya:031]:   Datapoint 2: int value (value: 0)
```
So I used those datapoints in the YAML.

However, I've changed those later. The config was as follows:
```yaml
light:
  - platform: "tuya"
    id: dimmer
    name: "${device_name}_dimmer"
    dimmer_datapoint: 2
    switch_datapoint: 3
```
And now the device works! ..Sorta
Pressing the power button, the light fades in, and if i press it again, it fades out. Nice!
It even remembers the brightness setting.
I can also change brightness from home assistant. However, the on/off state is not reported to home assistant, and I can not turn the bulb on or off from home assistant.

So I changed it again to 
```yaml
light:
  - platform: "tuya"
    id: dimmer
    name: "${device_name}_dimmer"
    dimmer_datapoint: 3
    switch_datapoint: 1
```
This time, the on/off state was reported to home assistant, but not the dimming value.
The operation of the buttons was good as well this time.

So it seems the switch datapoint definitely needs to be 1, the dimmer datapoint needs to be 2, but combining those together seems to mess things up.

Deinara, 2019-11-14:
Alright, ran the dev build as requested.
There is indeed no change in behaviour from the normal build.
I used datapoint 1 and 2 again this time.

The log output is below:

```
INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from sh_tuyd_15b9.local using esphome API
INFO Connecting to sh_tuyd_15b9.local:6053 (192.168.1.127)
INFO Successfully connected to sh_tuyd_15b9.local
[21:01:04][I][app:100]: ESPHome version 1.15.0-dev compiled on Nov 14 2019, 21:00:18
[21:01:04][C][wifi:415]: WiFi:
[21:01:04][C][wifi:283]:   SSID: [redacted]
[21:01:04][C][wifi:284]:   IP Address: 192.168.1.127
[21:01:04][C][wifi:286]:   BSSID: [redacted]
[21:01:04][C][wifi:287]:   Hostname: 'sh_tuyd_15b9'
[21:01:04][C][wifi:291]:   Signal strength: -49 dB ▂▄▆█
[21:01:04][C][wifi:295]:   Channel: 7
[21:01:04][C][wifi:296]:   Subnet: 255.255.255.0
[21:01:04][C][wifi:297]:   Gateway: 192.168.1.1
[21:01:04][C][wifi:298]:   DNS1: 192.168.1.1
[21:01:04][C][wifi:299]:   DNS2: (IP unset)
[21:01:04][C][uart:140]: UART Bus:
[21:01:04][C][uart:142]:   TX Pin: GPIO1
[21:01:04][C][uart:145]:   RX Pin: GPIO3
[21:01:04][C][uart:147]:   Baud Rate: 9600 baud
[21:01:04][C][uart:148]:   Stop bits: 1
[21:01:04][C][uart:150]:   Using hardware serial interface.
[21:01:04][C][uptime.sensor:030]: Uptime Sensor 'sh_tuyd_15b9_uptime'
[21:01:04][C][uptime.sensor:030]:   Unit of Measurement: 's'
[21:01:04][C][uptime.sensor:030]:   Accuracy Decimals: 0
[21:01:04][C][uptime.sensor:030]:   Icon: 'mdi:clock-outline'
[21:01:04][C][logger:175]: Logger:
[21:01:04][C][logger:176]:   Level: DEBUG
[21:01:04][C][logger:177]:   Log Baud Rate: 0
[21:01:04][C][logger:178]:   Hardware UART: UART0
[21:01:04][C][light:177]: Light 'sh_tuyd_15b9_dimmer'
[21:01:04][C][light:179]:   Default Transition Length: 0.0s
[21:01:04][C][light:180]:   Gamma Correct: 1.00
[21:01:04][C][tuya.light:027]: Tuya Dimmer:
[21:01:04][C][tuya.light:029]:    Dimmer has datapoint ID 2
[21:01:04][C][tuya.light:031]:    Switch has datapoint ID 1
[21:01:05][C][ota:029]: Over-The-Air Updates:
[21:01:05][C][ota:030]:   Address: sh_tuyd_15b9.local:8266
[21:01:05][C][ota:032]:   Using Password.
[21:01:05][C][api:095]: API Server:
[21:01:05][C][api:096]:   Address: sh_tuyd_15b9.local:6053
[21:01:05][C][tuya:023]: Tuya:
[21:01:05][C][tuya:032]:   Datapoint 1: switch (value: OFF)
[21:01:05][C][tuya:034]:   Datapoint 2: int value (value: 0)
[21:01:05][C][tuya:044]:   GPIO Configuration: status: pin 14, reset: pin 0 (not supported)
[21:01:05][C][tuya:046]:   Product: 'bSXSSFArVKtc4DyC1.0.0'
[21:01:05][C][tuya:023]: Tuya:
[21:01:05][C][tuya:032]:   Datapoint 1: switch (value: OFF)
[21:01:05][C][tuya:034]:   Datapoint 2: int value (value: 0)
[21:01:05][C][tuya:044]:   GPIO Configuration: status: pin 14, reset: pin 0 (not supported)
[21:01:05][C][tuya:046]:   Product: 'bSXSSFArVKtc4DyC1.0.0'
[21:01:10][D][api.connection:583]: Client 'Home Assistant 0.101.3 (192.168.1.25)' connected successfully!
[21:01:14][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:14][D][light:273]:   State: ON
[21:01:14][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:14][D][light:277]:   Brightness: 100%
[21:01:14][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:14][D][light:277]:   Brightness: 100%
[21:01:14][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:14][D][light:273]:   State: ON
[21:01:14][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:14][D][light:277]:   Brightness: 100%
[21:01:14][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:14][D][light:273]:   State: OFF
[21:01:14][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:14][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:14][D][light:277]:   Brightness: 100%
[21:01:14][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:14][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:14][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:14][D][light:277]:   Brightness: 96%
[21:01:14][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:14][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:14][D][light:277]:   Brightness: 95%
[21:01:14][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:14][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:14][D][light:277]:   Brightness: 94%
[21:01:14][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:14][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:14][D][light:277]:   Brightness: 93%
[21:01:15][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:

[Removed for brevity]

[21:01:21][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:21][D][light:277]:   Brightness: 6%
[21:01:21][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:21][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:21][D][light:277]:   Brightness: 5%
[21:01:21][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:21][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:21][D][light:277]:   Brightness: 4%
[21:01:21][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:21][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:21][D][light:277]:   Brightness: 3%
[21:01:21][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:21][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:21][D][light:277]:   Brightness: 2%
[21:01:21][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:21][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:21][D][light:277]:   Brightness: 1%
[21:01:21][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:21][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[21:01:21][D][light:277]:   Brightness: 100%
```

brandond, 2019-11-14:
There are a couple things to note here.
```
[21:01:05][C][tuya:023]: Tuya:
[21:01:05][C][tuya:032]:   Datapoint 1: switch (value: OFF)
[21:01:05][C][tuya:034]:   Datapoint 2: int value (value: 0)
[21:01:05][C][tuya:044]:   GPIO Configuration: status: pin 14, reset: pin 0 (not supported)
[21:01:05][C][tuya:046]:   Product: 'bSXSSFArVKtc4DyC1.0.0'
```

First of all, the product ID is interesting - the Tuya docs require this to be a JSON string, which this is not. So that's odd.

Second, the device has requested ESP-managed IO configuration, with a configuration reset switch on GPIO0 and a status LED on GPIO 14. You can set these up manually if you'd like; it suggests that there are a button and light that you're not currently using.

Third, I think you are on the right track with using datapoints 1 and 2 and the values being reversed. If you turn the log level up to VERBOSE you can see the datapoints and values that Tuya is sending when you operate the switch and slider locally, and figure out what values ESPHome should be sending.

Deinara, 2019-11-14:
I'm not sure what the product ID is supposed to be other than this, or if I can extract it some other way?

But the other light is interesting. There was a status indicator LED with the stock Tuya firmware that I had sort of forgotten about, but I set ESPHome's status_led to GPIO14, and sure enough... Thanks!

So for the verbose output:
I changed the datapoints again to prevent the weird behaviour.

### Pressing the power button to turn on
I can tell here that there is some issue with datapoint 2.
Update to 0 equals a brightness of 100%, update to 1 equals a brightness of 1%, update to 5 equals a brightness of 2%...
While the percentages make sense (mapping 256 to 100%), an update to 0 should not equal 100% I think.
```
[22:25:31][V][tuya:093]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)]
[22:25:31][V][tuya:111]: MCU Heartbeat (0x01)
[22:25:32][V][tuya:093]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)]
[22:25:32][V][tuya:111]: MCU Heartbeat (0x01)
[22:25:33][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[01.01.00.01.01 (5)]
[22:25:33][V][tuya:205]: Datapoint 1 update to 1
[22:25:33][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[02.02.00.04.00.00.00.00 (8)]
[22:25:33][V][tuya:205]: Datapoint 2 update to 0
[22:25:33][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[22:25:33][D][light:277]:   Brightness: 100%
[22:25:33][V][tuya:244]: Datapoint 3 set to 0
[22:25:33][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[01.01.00.01.01 (5)]
[22:25:33][V][tuya:205]: Datapoint 1 update to 1
[22:25:33][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[02.02.00.04.00.00.00.02 (8)]
[22:25:33][V][tuya:205]: Datapoint 2 update to 2
[22:25:33][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[22:25:33][D][light:277]:   Brightness: 1%
[22:25:33][V][tuya:244]: Datapoint 3 set to 0
[22:25:33][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[01.01.00.01.01 (5)]
[22:25:33][V][tuya:205]: Datapoint 1 update to 1
[22:25:33][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[02.02.00.04.00.00.00.05 (8)]
[22:25:33][V][tuya:205]: Datapoint 2 update to 5
[22:25:33][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[22:25:33][D][light:277]:   Brightness: 2%
[22:25:33][V][tuya:244]: Datapoint 3 set to 0
[22:25:33][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[01.01.00.01.01 (5)]
[22:25:33][V][tuya:205]: Datapoint 1 update to 1
[22:25:33][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[02.02.00.04.00.00.00.08 (8)]
[22:25:33][V][tuya:205]: Datapoint 2 update to 8
[22:25:33][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[22:25:33][D][light:277]:   Brightness: 3%
[22:25:33][V][tuya:244]: Datapoint 3 set to 0

[Removed for brevity]

[22:25:40][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[01.01.00.01.01 (5)]
[22:25:40][V][tuya:205]: Datapoint 1 update to 1
[22:25:40][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[02.02.00.04.00.00.00.F3 (8)]
[22:25:40][V][tuya:205]: Datapoint 2 update to 243
[22:25:40][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[22:25:40][D][light:277]:   Brightness: 95%
[22:25:40][V][tuya:244]: Datapoint 3 set to 0
[22:25:40][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[01.01.00.01.01 (5)]
[22:25:40][V][tuya:205]: Datapoint 1 update to 1
[22:25:40][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[02.02.00.04.00.00.00.F6 (8)]
[22:25:40][V][tuya:205]: Datapoint 2 update to 246
[22:25:40][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[22:25:40][D][light:277]:   Brightness: 96%
[22:25:40][V][tuya:244]: Datapoint 3 set to 0
[22:25:40][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[01.01.00.01.01 (5)]
[22:25:40][V][tuya:205]: Datapoint 1 update to 1
[22:25:40][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[02.02.00.04.00.00.00.FF (8)]
[22:25:40][V][tuya:205]: Datapoint 2 update to 255
[22:25:40][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[22:25:40][D][light:277]:   Brightness: 100%
[22:25:40][V][tuya:244]: Datapoint 3 set to 0
[22:25:40][V][tuya:093]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)]
[22:25:40][V][tuya:111]: MCU Heartbeat (0x01)
[... 73 lines cut ...]
[22:26:21][V][tuya:244]: Datapoint 3 set to 0
[22:26:21][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[01.01.00.01.00 (5)]
[22:26:21][V][tuya:205]: Datapoint 1 update to 0
[22:26:21][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[02.02.00.04.00.00.00.05 (8)]
[22:26:21][V][tuya:205]: Datapoint 2 update to 5
[22:26:21][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[22:26:21][D][light:277]:   Brightness: 2%
[22:26:21][V][tuya:244]: Datapoint 3 set to 0
[22:26:21][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[01.01.00.01.00 (5)]
[22:26:21][V][tuya:205]: Datapoint 1 update to 0
[22:26:21][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[02.02.00.04.00.00.00.02 (8)]
[22:26:21][V][tuya:205]: Datapoint 2 update to 2
[22:26:21][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[22:26:21][D][light:277]:   Brightness: 1%
[22:26:21][V][tuya:244]: Datapoint 3 set to 0
[22:26:21][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[01.01.00.01.00 (5)]
[22:26:21][V][tuya:205]: Datapoint 1 update to 0
[22:26:21][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[02.02.00.04.00.00.00.00 (8)]
[22:26:21][V][tuya:205]: Datapoint 2 update to 0
[22:26:21][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[22:26:21][D][light:277]:   Brightness: 100%
[22:26:21][V][tuya:244]: Datapoint 3 set to 0
[22:26:21][V][tuya:093]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)]
[22:26:21][V][tuya:111]: MCU Heartbeat (0x01)
```

brandond, 2019-11-14:
> I'm not sure what the product ID is supposed to be other than this, or if I can extract it some other way?

No, it's just not what I expected. The docs say it should be JSON with a couple different attributes in it. It's also sending a protocol version of 0, which I haven't seen outside of battery powered devices that seem to use a completely different set of command codes.

> I wouldn't know about another button, but the other light is interesting. There was a status indicator LED with the stock Tuya firmware that I had sort of forgotten about, but I set ESPHome's status_led to GPIO14, and sure enough... Thanks!

What do the docs say for configuring the device? Normally there's a button you can press that will reset the config and prepare it to connect up to your wifi and Tuya account. The button that is used to do this should be wired in to GPIO0. 

This bit is weird:
```
[22:25:33][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[02.02.00.04.00.00.00.00 (8)]
[22:25:33][V][tuya:205]: Datapoint 2 update to 0
[22:25:33][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[22:25:33][D][light:277]:   Brightness: 100%

[22:25:33][V][tuya:093]: Received Tuya: CMD=0x07 VERSION=0 DATA=[02.02.00.04.00.00.00.02 (8)]
[22:25:33][V][tuya:205]: Datapoint 2 update to 2
[22:25:33][D][light:264]: 'sh_tuyd_15b9_dimmer' Setting:
[22:25:33][D][light:277]:   Brightness: 1%
```

A value of 0 should not be treated as 100%. I don't see anywhere in the code that's doing this intentionally. I'm not super familiar with how C++ handles division and casting, but it might be this line:
https://github.com/esphome/esphome/blob/dev/esphome/components/tuya/light/tuya_light.cpp#L13

brandond, 2019-11-14:
Also @OttoWinter the debug logging for light state publishe is confusing. The state value is suppressed if nothing is changing, but the intro ' Setting:' line and other attributes are printed unconditionally.
https://github.com/esphome/esphome/blob/dev/esphome/components/light/light_state.cpp#L272

Deinara, 2019-11-15:
> Normally there's a button you can press that will reset the config and prepare it to connect up to your wifi and Tuya account.

The device physically has no other buttons. To reset or configure the device, I believe the dim up and dim down buttons had to be pressed at the same time. Could this action result in a pulse of the GPIO0 for example?
If I get around to it tonight, I'll try to test this theory.

brandond, 2019-11-15:
Yeah, give it a try, grab verbose logs at the same time if you can.

Deinara, 2019-11-15:
I can't seem to be able to get any response on GPIO0, I've tried numerous button combinations.
Doesn't it say using the pin is unsupported?
`[21:01:05][C][tuya:044]:   GPIO Configuration: status: pin 14, reset: pin 0 (not supported)`
Or does that mean that ESPHome doesn't know what to do with it by itself?

brandond, 2019-11-15:
yeah, the "unsupported" message means that we don't configure any of the GPIO stuff for you based on the configuration sent by the MCU. It's not specific to that pin.

From the protocol docs:
> For example, the Wi-Fi module sends 55 aa 00 02 00 00 01. The MCU returns 55 aa 03 02 00 00 04 (cooperative processing by the MCU and Wi-Fi module) or 55 aa 03 02 00 02 05 00 0b (processing by the Wi-Fi module). “05” and “00” indicate the I/O interfaces 5 and 0 that are connected to the Wi-Fi indicator and Wi-Fi reset button, respectively.

.. but again, there are a few other things here that don't line up exactly with the protocol docs so who knows.

OttoWinter, 2019-11-16:
> A value of 0 should not be treated as 100%. I don't see anywhere in the Tuya code that's doing this intentionally. @OttoWinter do you have any idea where this might be coming from?

I think this is coming from color normalization here:

https://github.com/esphome/esphome/blob/6682c43dfaeb1c006943ae546145e5f22262cadb/esphome/components/light/light_color_values.h#L164
(called from here: https://github.com/esphome/esphome/blob/1bf55c130beddf405e82efb933cb4665276f2247/esphome/components/light/light_state.cpp#L433)

Basically, if you set brightness to 0% that means the light is OFF. But when the light is off, brightness is reset to 100% (I believe this comes from HA - if it were not reset to 100%, the next time you press ON the state will be ON+0% brightness, which HA interface renders as OFF - so you can never turn the light on again).

> Also @OttoWinter the debug logging for light state publish is confusing. The state value is suppressed if nothing is changing, but the intro ' Setting:' line and other attributes are printed unconditionally.

True, I think the main reason behind it was to make logs less cluttered, as most calls to brightness will be through `light.turn_on`. I'd be ok with changing it to print it always if the optional is set.

mightydread, 2019-12-13:
any update on the issue? I have the same problem. I am glad to provide with anything that might help..

brandond, 2019-12-13:
I haven't had much time to work on this lately, but properly scaling the min and max values, and handling min > max as inverted, are both on my to-do list.

mightydread, 2019-12-16:
@brandond  Sure! here are the `DEBUG` logs on first boot

```
[15:13:40][I][app:100]: ESPHome version 1.14.3 compiled on Dec 16 2019, 15:11:48
[15:13:40][C][status_led:019]: Status LED:
[15:13:40][C][status_led:020]:   Pin: GPIO14 (Mode: OUTPUT)
[15:13:40][C][wifi:415]: WiFi:
[15:13:40][C][wifi:283]:   SSID: 

[redacted]
[15:13:40][C][wifi:284]:   IP Address: 10.0.0.23
[15:13:40][C][wifi:286]:   BSSID: 

[redacted]
[15:13:40][C][wifi:287]:   Hostname: 'mighty_dimmer'
[15:13:40][C][wifi:291]:   Signal strength: -64 dB ▂▄▆█
[15:13:40][C][wifi:295]:   Channel: 1
[15:13:40][C][wifi:296]:   Subnet: 255.255.255.0
[15:13:40][C][wifi:297]:   Gateway: 10.0.0.1
[15:13:40][C][wifi:298]:   DNS1: 10.0.0.10
[15:13:40][C][wifi:299]:   DNS2: (IP unset)
[15:13:40][C][uart:140]: UART Bus:
[15:13:40][C][uart:142]:   TX Pin: GPIO1
[15:13:40][C][uart:145]:   RX Pin: GPIO3
[15:13:40][C][uart:147]:   Baud Rate: 9600 baud
[15:13:40][C][uart:148]:   Stop bits: 1
[15:13:40][C][uart:150]:   Using hardware serial interface.
[15:13:40][C][logger:175]: Logger:
[15:13:40][C][logger:176]:   Level: DEBUG
[15:13:40][C][logger:177]:   Log Baud Rate: 0
[15:13:40][C][logger:178]:   Hardware UART: UART0
[15:13:40][C][light:177]: Light 'mighty_dimmer'
[15:13:40][C][light:179]:   Default Transition Length: 0.0s
[15:13:40][C][light:180]:   Gamma Correct: 1.00
[15:13:40][C][tuya.light:027]: Tuya Dimmer:
[15:13:40][C][tuya.light:029]:    Dimmer has datapoint ID 2
[15:13:40][C][tuya.light:031]:    Switch has datapoint ID 1
[15:13:40][C][captive_portal:169]: Captive Portal:
[15:13:40][C][ota:029]: Over-The-Air Updates:
[15:13:40][C][ota:030]:   Address: mighty_dimmer.local:8266
[15:13:40][C][api:095]: API Server:
[15:13:40][C][api:096]:   Address: mighty_dimmer.local:6053
[15:13:40][C][tuya:023]: Tuya:
[15:13:40][C][tuya:032]:   Datapoint 1: switch (value: OFF)
[15:13:40][C][tuya:034]:   Datapoint 2: int value (value: 0)
[15:13:40][C][tuya:044]:   GPIO Configuration: status: pin 14, reset: pin 0 (not supported)
[15:13:40][C][tuya:046]:   Product: 'bSXSSFArVKtc4DyC1.0.0'
```
And here is the `VERBOSE` log when pressing the on button:
```
[15:19:39][I][app:100]: ESPHome version 1.14.3 compiled on Dec 16 2019, 15:17:45
[15:19:39][C][status_led:019]: Status LED:
[15:19:39][C][status_led:020]:   Pin: GPIO14 (Mode: OUTPUT)
[15:19:39][C][wifi:415]: WiFi:
[15:19:39][C][wifi:283]:   SSID: 

[redacted]
[15:19:39][C][wifi:284]:   IP Address: 10.0.0.23
[15:19:39][C][wifi:286]:   BSSID: 

[redacted]
[15:19:39][C][wifi:287]:   Hostname: 'mighty_dimmer'
[15:19:39][C][wifi:291]:   Signal strength: -63 dB ▂▄▆█
[15:19:39][V][wifi:293]:   Priority: 0.0
[15:19:39][C][wifi:295]:   Channel: 6
[15:19:39][C][wifi:296]:   Subnet: 255.255.255.0
[15:19:39][C][wifi:297]:   Gateway: 10.0.0.1
[15:19:39][C][wifi:298]:   DNS1: 10.0.0.10
[15:19:39][C][wifi:299]:   DNS2: (IP unset)
[15:19:39][C][uart:140]: UART Bus:
[... 1592 lines cut ...]
[15:25:35][V][tuya:235]: Datapoint 2 update to 246
[15:25:35][D][light:264]: 'mighty_dimmer' Setting:
[15:25:35][D][light:277]:   Brightness: 96%
[15:25:35][V][tuya:275]: Datapoint 2 set to 246
[15:25:35][V][tuya:279]: Not sending unchanged value
[15:25:35][V][tuya:275]: Datapoint 1 set to 1
[15:25:35][V][tuya:279]: Not sending unchanged value
[15:25:35][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=0 DATA=[01.01.00.01.01 (5)] INIT_STATE=5
[15:25:35][V][tuya:235]: Datapoint 1 update to 1
[15:25:35][D][light:264]: 'mighty_dimmer' Setting:
[15:25:35][V][tuya:275]: Datapoint 2 set to 246
[15:25:35][V][tuya:279]: Not sending unchanged value
[15:25:35][V][tuya:275]: Datapoint 1 set to 1
[15:25:35][V][tuya:279]: Not sending unchanged value
[15:25:35][V][tuya:098]: Received Tuya: CMD=0x07 VERSION=0 DATA=[02.02.00.04.00.00.00.FF (8)] INIT_STATE=5
[15:25:35][V][tuya:235]: Datapoint 2 update to 255
[15:25:35][D][light:264]: 'mighty_dimmer' Setting:
[15:25:35][D][light:277]:   Brightness: 100%
[15:25:35][V][tuya:275]: Datapoint 2 set to 255
[15:25:35][V][tuya:279]: Not sending unchanged value
[15:25:35][V][tuya:275]: Datapoint 1 set to 1
[15:25:35][V][tuya:279]: Not sending unchanged value
[15:25:35][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[15:25:35][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
```

mightydread, 2019-12-16:
Hope this helps. If I can be of any more help please let me know.

poldim, 2020-01-08:
Chiming in with what I've noticed:

- Similar dimmer hardware with the same product ID (`bSXSSFArVKtc4DyC1.0.0`).
- The dimmer is controlling a light with 6x 6w dimmable LED bulbs
- The light was previously controlled by a [~20+ year old rotary dimmer](https://i.imgur.com/b6z9RT8.jpg).  This dimmer provided fine low-level dimming
- Initially ran Tasmota and it seemed to be able to dim to a lower brightness on the low end, but very poor distribution as 60% of the lighting control happened in the first 20% of the dimmer slider and the remaining 40% happened in the last 40% of the slider, with the rest being a dead band.
- The instructions about `gamma_correct` say they are 1.0 by default on the MCU but did not say you cannot overwrite them with a new value.  Attempting to do this creates erratic continuous blinking of the lights so I reverted back.
- When adjusting the brightness via the slider, it will jump back some percentage.  This occurs in both dimming up & down. [GIF for reference.](https://imgur.com/ZBGlUxs)
- Since the device has no physical buttons to provide input on full on and full off, I cannot find out the actual min/max values.

[image]

## Logs
```
[04:01:46][I][app:101]: ESPHome version 1.15.0-dev compiled on Jan  8 2020, 04:00:56
[04:01:46][C][wifi:415]: WiFi:
[04:01:46][C][wifi:283]:   SSID: [redacted]
[04:01:46][C][wifi:284]:   IP Address: 10.30.30.180
[04:01:46][C][wifi:286]:   BSSID: [redacted]
[04:01:46][C][wifi:287]:   Hostname: 'esphome_dining_room_dimmer'
[04:01:46][C][wifi:291]:   Signal strength: -58 dB ▂▄▆█
[04:01:46][C][wifi:295]:   Channel: 6
[04:01:46][C][wifi:296]:   Subnet: 255.255.255.0
[04:01:46][C][wifi:297]:   Gateway: 10.30.30.1
[04:01:46][C][wifi:298]:   DNS1: 10.30.30.1
[04:01:46][C][wifi:299]:   DNS2: (IP unset)
[04:01:46][C][uart:140]: UART Bus:
[04:01:46][C][uart:142]:   TX Pin: GPIO1
[04:01:46][C][uart:145]:   RX Pin: GPIO3
[04:01:46][C][uart:147]:   Baud Rate: 9600 baud
[04:01:46][C][uart:148]:   Stop bits: 1
[04:01:46][C][uart:150]:   Using hardware serial interface.
[04:01:46][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'physical_button'
[04:01:46][C][gpio.binary_sensor:016]:   Pin: GPIO0 (Mode: INPUT_PULLUP, INVERTED)
[04:01:46][C][logger:175]: Logger:
[04:01:46][C][logger:176]:   Level: VERBOSE
[04:01:46][C][logger:177]:   Log Baud Rate: 0
[04:01:46][C][logger:178]:   Hardware UART: UART0
[04:01:46][C][light:177]: Light 'Dining Chandelier'
[04:01:46][C][light:179]:   Default Transition Length: 0.0s
[04:01:46][C][light:180]:   Gamma Correct: 1.00
[04:01:46][C][tuya.light:027]: Tuya Dimmer:
[04:01:46][C][tuya.light:029]:    Dimmer has datapoint ID 2
[04:01:46][C][tuya.light:031]:    Switch has datapoint ID 1
[04:01:46][C][restart:021]: Restart Switch 'Dining Chandelier REBOOT'
[04:01:46][C][restart:021]:   Icon: 'mdi:restart'
[04:01:46][C][web_server:123]: Web Server:
[04:01:46][C][web_server:124]:   Address: esphome_dining_room_dimmer.local:80
[04:01:46][C][ota:029]: Over-The-Air Updates:
[04:01:46][C][ota:030]:   Address: esphome_dining_room_dimmer.local:8266
[04:01:46][C][api:095]: API Server:
[04:01:46][C][api:096]:   Address: esphome_dining_room_dimmer.local:6053
[04:01:46][C][tuya:023]: Tuya:
[04:01:46][C][tuya:032]:   Datapoint 1: switch (value: OFF)
[04:01:46][C][tuya:034]:   Datapoint 2: int value (value: 21)
[04:01:46][C][tuya:044]:   GPIO Configuration: status: pin 14, reset: pin 0 (not supported)
[04:01:46][C][tuya:046]:   Product: 'bSXSSFArVKtc4DyC1.0.0'
[04:01:46][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[04:01:46][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
[04:02:25][V][tuya:115]: MCU Heartbeat (0x01)
[04:02:26][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[04:02:26][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
[04:02:26][V][tuya:115]: MCU Heartbeat (0x01)
[04:02:27][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[04:02:27][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
[04:02:27][V][tuya:115]: MCU Heartbeat (0x01)
[04:02:28][V][tuya:261]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[(0)] INIT_STATE=5
[04:02:28][V][tuya:098]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
[04:02:28][V][tuya:115]: MCU Heartbeat (0x01)
[04:02:29][D][light:264]: 'Dining Chandelier' Setting:
[... 359 lines cut ...]
  # default_transition_length: 1s
  min_value: 0
  max_value: 255

binary_sensor:
- platform: gpio
  pin:
    number: GPIO0
    mode: INPUT_PULLUP
    inverted: True
  id: physical_button
  internal: true
  on_press:
    - light.toggle: dimmer

switch:
- platform: restart
  name: '${friendly_name} REBOOT'

# status_led:
#   pin:
#     number: GPIO14
#     inverted: True

```

LeandroSPereira, 2020-01-13:
Good evening, I also got a dimmer module like this one and I want to run ESPHOME on it, today with TASMOTA it has bad dimmer control, I have tried every possible `` dimmerrange`` and it does not respond satisfactorily.

I will continue studying the same, if you have any satisfactory configuration with ESPHOME, I will post here, and I ask if someone can work well with it, also position us.

Thank you all.

kainzoW, 2020-03-08:
I am confused with this issue. Is it to do with HA or ESPHome? There seems to be a setting that makes dimmer work, but the dimmer status is not updated in HA.
I am happy to contribute to testing, but I am fairly inexpirienced.

Chuckame, 2020-03-25:
Hi,

This is working with my tuya switch (2 way) with this config:
```
esphome:
  name: switch_tuya2_1
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "xxx"
  password: "xxx"

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:

ota:
```

kainzoW, 2020-03-25:
@Chuckame This bug is specific to Tuya dimmer.
The issue is that dimmer needs:
switch_datapoint:
dimmer_datapoint:
settings and there is the setting that makes dimmer to work but not communicate with HA or the setting that makes dimmer to misbehave but the communication with HA works.

wil62, 2020-04-17:
Hello, same issue than the original post for me, when i turn on the light on the device, the light turns on and turns off imediately. Secondly the brightness value is not registered, each times i turn on the light by HA or the device, the brightness is 100%, is this normal for the second point?

ssypchenko, 2020-04-18:
Hello, same issue than the original post for me as well :( 

Please help us to find a solution. It is something wrong in this part: 
`tuya:

light:
  - platform: "tuya"
    id: ${devicename}
    name: ${fullname}
    switch_datapoint: 1
    dimmer_datapoint: 2
    restore_mode: ALWAYS_OFF`

because without it if switch it manually, it works perfectly.

markbooth73, 2020-06-30:
Having the same issues as original poster, just to add when the dimmer had the original firmware you had to press the down button for >10 sec to put it into pairing mode.
What I did find if i press the power button (middle) the light comes on they fades to off.

If I tap the button twice in quick succession the light stays on.
my config
```
esphome:
  name: esp_dim01
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "IOT_Net"
  password: "redacted"
  power_save_mode: none
  
  manual_ip:  
   static_ip: 192.168.0.240
   gateway: 192.168.0.2
   subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp Dim01 Fallback Hotspot"
    password: "zcgCOBfC1eho"

# Enable logging

logger:
  baud_rate: 0
  level: VERBOSE
  
api:

ota:

web_server:

time:
  - platform: homeassistant

# Uart definition to talk to MCU dimmer
uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  stop_bits: 1
  baud_rate: 9600

# Register the Tuya MCU connection
tuya:

sensor:
  - platform: wifi_signal
    name: "dim_1 WiFi Signal Sensor"
    update_interval: 60s

# Primary Light object exposed to HA
light:
  - platform: "tuya"
    name: "dim1"
    dimmer_datapoint: 2
    switch_datapoint: 1
    min_value: 25
    max_value: 255

switch:
  - platform: restart
    name: "${switch_id} Restart"
```

ssypchenko, 2020-06-30:
I moved to Tasmota. Its dimmer works correctly. It is not so easy to set up as well. However, it works right.

aaliddell, 2020-11-21:
Not stale, still issue, pending PRs

bbadrignans, 2020-11-30:
For me setting max_value to 254 somehow fix the behavior of the button.
My model is YDM_WIFI_SV4.

The YAML : 

```
substitutions:
  node_name: dimmersalon
  device_verbose_name: "Lustre salon"

esphome:
  name: $node_name
  platform: ESP8266
  #WF-DS01
  board: esp01_1m

packages:
  device_base: !include common/common.yaml

# Make sure logging is not using the serial port
logger:
  baud_rate: 0

# My dimmer used the hardware serial port on the alternate pins
uart:
  rx_pin: GPIO03
  tx_pin: GPIO01
  baud_rate: 9600

# Register the Tuya MCU connection
tuya:

light:
  - platform: "tuya"
    id: dimmer
    name: "salon"
    dimmer_datapoint: 2
    switch_datapoint: 1
    min_value: 0
    max_value: 254
  
status_led:
  pin: GPIO14

```

The logs when I push the button shows that it does not really fix the underlying issue.

```
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][D][light:274]:   State: ON
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][D][light:278]:   Brightness: 100%
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][D][light:278]:   Brightness: 100%
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][D][light:274]:   State: ON
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][W][light:468]: 'salon' - Brightness value 1.00 is out of range [0.0 - 1.0]!
[18:48:29][D][light:278]:   Brightness: 100%
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][D][light:274]:   State: OFF
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][W][light:468]: 'salon' - Brightness value 1.00 is out of range [0.0 - 1.0]!
[18:48:29][D][light:278]:   Brightness: 100%
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][D][light:278]:   Brightness: 97%
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][D][light:278]:   Brightness: 96%
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][D][light:278]:   Brightness: 94%
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][D][light:278]:   Brightness: 94%
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][D][light:278]:   Brightness: 100%
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][D][light:274]:   State: ON
[18:48:29][D][light:265]: 'salon' Setting:
[18:48:29][W][light:468]: 'salon' - Brightness value 1.00 is out of range [0.0 - 1.0]!
[18:48:29][D][light:278]:   Brightness: 100%
[18:48:30][D][light:265]: 'salon' Setting:
[18:48:30][D][light:278]:   Brightness: 100%
```

markbooth73, 2020-12-02:
Hi Benoit,

Applied your "fix" and whilst the switch behaviour is more manageable now it is not perfect.

My logs as follows.
```
INFO Reading configuration /config/esphome/esp_dining_room_dimmer.yaml...
INFO Detected timezone 'GMT' with UTC offset 0 and daylight savings time from 28 March 01:00:00 to 31 October 02:00:00
INFO Starting log output from 192.168.0.240 using esphome API
INFO Connecting to 192.168.0.240:6053 (192.168.0.240)
INFO Successfully connected to 192.168.0.240
[15:43:25][I][app:105]: ESPHome version 1.16.0-dev compiled on Nov 30 2020, 19:10:14
[15:43:25][C][wifi:443]: WiFi:
[15:43:25][C][wifi:303]:   SSID: [redacted]
[15:43:25][C][wifi:304]:   IP Address: 192.168.0.240
[15:43:25][C][wifi:306]:   BSSID: [redacted]
[15:43:25][C][wifi:307]:   Hostname: 'esp_dining_room_dimmer'
[15:43:25][C][wifi:311]:   Signal strength: -59 dB ▂▄▆█
[15:43:25][C][wifi:315]:   Channel: 11
[15:43:25][C][wifi:316]:   Subnet: 255.255.255.0
[15:43:25][C][wifi:317]:   Gateway: 192.168.0.2
[15:43:25][C][wifi:318]:   DNS1: (IP unset)
[15:43:25][C][wifi:319]:   DNS2: (IP unset)
[15:43:25][C][uart_esp8266:075]: UART Bus:
[15:43:25][C][uart_esp8266:077]:   TX Pin: GPIO1
[15:43:25][C][uart_esp8266:080]:   RX Pin: GPIO3
[15:43:25][C][uart_esp8266:081]:   RX Buffer Size: 256
[15:43:25][C][uart_esp8266:083]:   Baud Rate: 9600 baud
[15:43:25][C][uart_esp8266:084]:   Data Bits: 8
[15:43:25][C][uart_esp8266:085]:   Parity: NONE
[15:43:25][C][uart_esp8266:086]:   Stop bits: 1
[15:43:25][C][uart_esp8266:088]:   Using hardware serial interface.
[15:43:25][C][uptime.sensor:030]: Uptime Sensor 'Dining Room Dimmer Uptime'
[15:43:25][C][uptime.sensor:030]:   Unit of Measurement: 's'
[15:43:25][C][uptime.sensor:030]:   Accuracy Decimals: 0
[15:43:25][C][uptime.sensor:030]:   Icon: 'mdi:timer-outline'
[15:43:25][C][logger:185]: Logger:
[15:43:25][C][logger:186]:   Level: DEBUG
[15:43:25][C][logger:187]:   Log Baud Rate: 0
[15:43:25][C][logger:188]:   Hardware UART: UART0
[15:43:25][C][light:178]: Light 'Dining Room Dimmer'
[15:43:25][C][light:180]:   Default Transition Length: 0.0s
[15:43:25][C][light:181]:   Gamma Correct: 1.00
[15:43:25][C][tuya.light:034]: Tuya Dimmer:
[15:43:25][C][tuya.light:036]:    Dimmer has datapoint ID 2
[15:43:25][C][tuya.light:038]:    Switch has datapoint ID 1
[15:43:25][C][restart:021]: Restart Switch 'Dining Room Restart'
[15:43:25][C][restart:021]:   Icon: 'mdi:restart'
[15:43:25][C][status:034]: Status Binary Sensor 'Dining Room Dimmer Status'
[15:43:25][C][status:034]:   Device Class: 'connectivity'
[15:43:25][C][homeassistant.time:010]: Home Assistant Time:
[15:43:25][C][homeassistant.time:011]:   Timezone: 'GMT0BST-1,M3.4.0/1,M10.5.0/2'
[15:43:25][C][version.text_sensor:021]: Version Text Sensor 'Dining Room Dimmer ESPHome Version'
[15:43:25][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[15:43:25][C][web_server:132]: Web Server:
[15:43:25][C][web_server:133]:   Address: 192.168.0.240:80
[15:43:25][C][ota:029]: Over-The-Air Updates:
[15:43:25][C][ota:030]:   Address: 192.168.0.240:8266
[15:43:25][C][api:095]: API Server:
[15:43:25][C][api:096]:   Address: 192.168.0.240:6053
[15:43:26][C][wifi_signal.sensor:009]: WiFi Signal 'Dining Room Dimmer WiFi Signal'
[15:43:26][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[15:43:26][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[15:43:26][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[15:43:26][C][wifi_info:009]: WifiInfo IPAddress 'IP Dining Room Dimmer'
[15:43:26][C][tuya:033]: Tuya:
[15:43:26][C][tuya:042]:   Datapoint 1: switch (value: OFF)
[15:43:26][C][tuya:044]:   Datapoint 2: int value (value: 0)
[15:43:26][C][tuya:054]:   GPIO Configuration: status: pin 14, reset: pin 0 (not supported)
[15:43:26][C][tuya:056]:   Product: 'bSXSSFArVKtc4DyC1.0.0'
[... 278 lines cut ...]
[15:43:45][D][light:265]: 'Dining Room Dimmer' Setting:
[15:43:45][D][light:265]: 'Dining Room Dimmer' Setting:
[15:43:45][D][light:278]:   Brightness: 6%
[15:43:45][D][light:265]: 'Dining Room Dimmer' Setting:
[15:43:45][D][light:265]: 'Dining Room Dimmer' Setting:
[15:43:45][D][light:278]:   Brightness: 6%
[15:43:45][D][light:265]: 'Dining Room Dimmer' Setting:
[15:43:45][D][light:265]: 'Dining Room Dimmer' Setting:
[15:43:45][D][light:278]:   Brightness: 4%
[15:43:45][D][light:265]: 'Dining Room Dimmer' Setting:
[15:43:45][D][light:265]: 'Dining Room Dimmer' Setting:
[15:43:45][D][light:278]:   Brightness: 3%
[15:43:45][D][light:265]: 'Dining Room Dimmer' Setting:
[15:43:45][D][light:265]: 'Dining Room Dimmer' Setting:
[15:43:45][D][light:278]:   Brightness: 2%
[15:43:45][D][light:265]: 'Dining Room Dimmer' Setting:
[15:43:45][D][light:265]: 'Dining Room Dimmer' Setting:
[15:43:45][D][light:278]:   Brightness: 1%
[15:43:45][D][light:265]: 'Dining Room Dimmer' Setting:
[15:43:45][D][light:265]: 'Dining Room Dimmer' Setting:
[15:43:45][D][light:278]:   Brightness: 100%
```
I tried going back to Tasmota v9.1.0 Imogen but got all kinds of strange behaviours (even worse than ESPHome) so reverted back.
So in short many thanks all that are, have tried to fix this so far....
Regards,

ssypchenko, 2021-05-13:
Hi guys, 
Any way to fix this?

rb0135, 2021-09-22:
Hi All.

Any more ideas? I had my dimmer on Tasmota 8.5 and worked fine, however, any version above this has the same issues as does esphome (dimmer not working, pressing the on button causes lights to come on then go off straight away). 

Im trying to move my devices away from Tasmota to esphome.

Willing to test.

One question: Do you need a Binary Sensor section in the yaml? I noticed most of you in your above configurations dont have that section, while some configs Ive seen do.

Thanks,
Rob

rb0135, 2021-09-23:
Hi All.

So to add some more information. Here are some tests, my config and logs.

1) I needed to add the min_value and max_value options.
2) Doing step #1 above, made the physical buttons turn on the lights (but had to hold the button for 3 seconds, rather than just touching it).
3) Home Assistant work perfectly. IE: I could switch them on and use the dimmer function.
4) However, both 3 and 4, the brightness was always 100% when turned on. The unit did not retain the previous Dimmer setting.
5) Turning off both physical button and homeassitant worked.
6) I added some extra code on_turn_on as I want the dimmer set to 5% when turned on. However, this stuffed the physical buttons from working.
7) Home Assistant did work perfect with the added code and when turning the lights on, it set the brightness to 5%.
8) The physical button to turn the lights off worked fine.

So, it is interesting that adding some code to set the dimmer level after turn on totally makes the unit fail to turn on with physical touch of the button. What is more interesting, is in the log (shown below), it does register the ON command, it does register 5% command, but then sends OFF to the MCU. It doesnt send the ON to the MCU.

Does this mean there might be some timing issue? It seems ESPHome must be sending the OFF instead of the ON.. maybe?

Here is my config (relevant part) and logs. 

`#captive_portal:

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

tuya:

light:
  - platform: "tuya"
    name: "LoungeLights"
    id: dimmer
    dimmer_datapoint: 2
    switch_datapoint: 1
    min_value: 0
    max_value: 254
    restore_mode: ALWAYS_OFF

    on_turn_on:
      then:
        - light.control:
            id: dimmer
            brightness: 5%

status_led:
  pin:
    number: GPIO14
    
web_server:
  port: 80`

Log when touching the physical button:
`[07:56:57][I][app:102]: ESPHome version 2021.9.1 compiled on Sep 24 2021, 07:44:40
[07:56:57][C][status_led:019]: Status LED:
[07:56:57][C][status_led:020]:   Pin: GPIO14 (Mode: OUTPUT)
[07:56:57][C][wifi:501]: WiFi:
[07:56:57][C][wifi:361]:   SSID: [redacted]
[07:56:57][C][wifi:362]:   IP Address: 10.35.2.130
[07:56:57][C][wifi:364]:   BSSID: [redacted]
[07:56:57][C][wifi:365]:   Hostname: 'sw_loungelights'
[07:56:57][C][wifi:369]:   Signal strength: -43 dB ▂▄▆█
[07:56:57][V][wifi:371]:   Priority: 0.0
[07:56:57][C][wifi:373]:   Channel: 6
[07:56:57][C][wifi:374]:   Subnet: 255.255.255.0
[07:56:57][C][wifi:375]:   Gateway: 10.35.2.254
[07:56:57][C][wifi:376]:   DNS1: 10.35.2.2
[07:56:57][C][wifi:377]:   DNS2: (IP unset)
[07:56:57][C][uart_esp8266:097]: UART Bus:
[07:56:57][C][uart_esp8266:099]:   TX Pin: GPIO1
[07:56:57][C][uart_esp8266:102]:   RX Pin: GPIO3
[... 176 lines cut ...]
[07:55:51][D][tuya:268]: Datapoint 2 update to 5
[07:55:51][D][light:035]: 'LoungeLights' Setting:
[07:55:51][D][light:050]:   Brightness: 2%
[07:55:51][D][tuya:499]: Setting datapoint 1 to 0
[07:55:51][V][tuya:507]: Not sending unchanged value
[07:55:51][V][tuya:106]: Received Tuya: CMD=0x07 VERSION=0 DATA=[01.01.00.01.00 (5)] INIT_STATE=5
[07:55:51][D][tuya:260]: Datapoint 1 update to OFF
[07:55:51][D][light:035]: 'LoungeLights' Setting:
[07:55:51][D][tuya:499]: Setting datapoint 1 to 0
[07:55:51][V][tuya:507]: Not sending unchanged value
[07:55:51][V][tuya:106]: Received Tuya: CMD=0x07 VERSION=0 DATA=[02.02.00.04.00.00.00.02 (8)] INIT_STATE=5
[07:55:51][D][tuya:268]: Datapoint 2 update to 2
[07:55:51][D][light:035]: 'LoungeLights' Setting:
[07:55:51][D][light:050]:   Brightness: 1%
[07:55:51][D][tuya:499]: Setting datapoint 1 to 0
[07:55:51][V][tuya:507]: Not sending unchanged value
[07:55:51][V][tuya:106]: Received Tuya: CMD=0x07 VERSION=0 DATA=[01.01.00.01.00 (5)] INIT_STATE=5
[07:55:51][D][tuya:260]: Datapoint 1 update to OFF
[07:55:51][D][light:035]: 'LoungeLights' Setting:
[07:55:51][V][tuya:106]: Received Tuya: CMD=0x07 VERSION=0 DATA=[02.02.00.04.00.00.00.00 (8)] INIT_STATE=5
[07:55:51][D][tuya:268]: Datapoint 2 update to 0
[07:55:51][D][light:035]: 'LoungeLights' Setting:
[07:55:51][D][light:050]:   Brightness: 100%
[07:55:51][D][tuya:499]: Setting datapoint 1 to 0
[07:55:51][V][tuya:507]: Not sending unchanged value`

aaliddell, 2021-09-23:
There's a set of race conditions that lead to what you are seeing, where the MCU and ESP race to change states, which you may also have added a race with HA also. I proposed a fix for this in https://github.com/esphome/esphome/pull/1289, with the comments there explaining why this happens. I've been running that PR's code since Sep 2020 without any of the issues, but unfortunately that PR seems to have been closed now without a different fix in place :shrug:

rb0135, 2021-09-23:
> There's a set of race conditions that lead to what you are seeing, where the MCU and ESP race to change states, which you may also have added a race with HA also. I proposed a fix for this in [esphome/esphome#1289](https://github.com/esphome/esphome/pull/1289), with the comments there explaining why this happens. I've been running that PR's code since Sep 2020 without any of the issues, but unfortunately that PR seems to have been closed now without a different fix in place 🤷

Thanks @aaliddell. Interesting read that you listed in the points and that makes 100% perfect sense...  You mention you are running the PR Code without any issues.. What code are you referring to?

Thanks for commenting.
Rob

aaliddell, 2021-09-23:
I ran esphome built from the code in that pull request, uploaded the image to the ESP and then haven't had to touch it since. The high level steps are:

- Clone https://github.com/aaliddell/esphome
- Checkout branch `fix-tuya-dimmer`
- Run esphome with `python3 -m esphome dashboard` etc from repo root
- Nuild and install the image to the dimmer (don't upload to any other things you might have)
- Let us know if it works :smile: 

You can then switch back to mainline esphome for everything else, but just ensure you ignore the warnings to update the dimmer since that'd overwrite what you did above.

rb0135, 2021-09-23:
> I ran esphome built from the code in that pull request, uploaded the image to the ESP and then haven't had to touch it since. The high level steps are:
> 
> * Clone https://github.com/aaliddell/esphome
> * Checkout branch `fix-tuya-dimmer`
> * Run esphome with `python3 -m esphome dashboard` etc from repo root
> * Nuild and install the image to the dimmer (don't upload to any other things you might have)
> * Let us know if it works 😄
> 
> You can then switch back to mainline esphome for everything else, but just ensure you ignore the warnings to update the dimmer since that'd overwrite what you did above.

Ohh.. thanks for that.. I see now... I havent done the custom compile of ESPhome, but thanks for posting the steps... Something to play with today. Will certainly report back.

Thanks for your assistance.
Rob

rb0135, 2021-09-24:
@aaliddell - Hi Adam. Well, after a bit of phaffing around (getting python stuff sorted.. I hate python with a passion.. Im a C, C#, C++ guy... another story) I was able to compile a bin file with my yaml code above.

Uploaded it to the dimmer, and, it works 100%... It turns on/off from the Dimmer unit itself, Dim Up/Dim Down work perfectly, and Home Assistant works as previous. Even controlling the Dimmer itself registers in HA. And my code to preset the lights to 5% at turn on works a treat. 

Thank you very much for the heads up on your changes. It is a shame they havent been included yet into the mainstream

aaliddell, 2021-09-24:
Great to hear, I'm glad it's fixed everything for you :+1: 

Well if the maintainers want a fix then the code is available as ever. If it's actually going to get merged then I can try to rebase it against master, but I'm not going to do that if it'll sit idle for another year and then get closed :confused:

rb0135, 2021-09-25:
> Great to hear, I'm glad it's fixed everything for you 👍
> 
> Well if the maintainers want a fix then the code is available as ever. If it's actually going to get merged then I can try to rebase it against master, but I'm not going to do that if it'll sit idle for another year and then get closed 😕

Thanks Adam... I have found an issue in HA where the dimmer function jumps, so over the "100%" there seems to be only 4 settings. Im presuming that is my min/max values I put into the script. Will revisit that some day (physical buttons work great though).

I used to do a lot of "volunteer" programming for various projects as the "owners" of these projects always scream out for help. However, the more I did, the less the maintainers would merge fixes from those helpers, so I gave up helping...

mvanlijden, 2021-12-15:
Any update on a merge of https://github.com/esphome/esphome/pull/1289 to the main tree?
This item has been open for over **two years** with 17 participants, someone has reported a fix for the issue over a year ago already and somehow this has never been accepted by the devs. How is this possible?
The quick answer from @OttoWinter of not having a Tuya device is not really helping users. There are devs here working on code, users willing to test. How can you ignore such a request...

funkybassmandan, 2022-02-26:
I'm trying to install the fix as described by @rb0135 , but due to my inexperience, am drawing a blank!
Would it be possible to give a more detailed description of its implementation for us noobs?
Thanks.

inspoadam, 2023-10-08:
Hi all. Reviving this thread as I have burned a lot of time on this issue. I have finally managed to get something approximating working by using the fix provided by @aaliddell (thanks) and reducing max_value from 255 to 254. I am still getting some unpredictable results though, which I would be happy to work with someone on for the benefit of myself and others having the same problem.

The important bit of my config is as follows:

...

esp8266:
  board: esp01_1m
...

status_led:
  pin: 
    number: GPIO14
    inverted: false

# tuya MCO Connection
tuya:

light:
  - platform: "tuya"
    name: State
    id: ${device_name}
    dimmer_datapoint: 2
    switch_datapoint: 1
    min_value: 10
    max_value: 254
    restore_mode: RESTORE_DEFAULT_OFF
    icon: "mdi:lightbulb"

  
For those of you struggling (as I did) to implement the fix provided by @aaliddell, here is a slightly more verbose version:

1. From a console, create a folder to store the fork and open the folder.
2. Clone https://github.com/aaliddell/esphome [git clone https://github.com/aaliddell/esphome]
3. Checkout branch fix-tuya-dimmer [cd esphome, git checkout fix-tuya-dimmer]
4. Run esphome with python3 -m esphome dashboard etc from repo root [python3 -m esphome esphome dashboard] - boy did this take a long time to work out!!!
5. Build and install the image to the dimmer (don't upload to any other things you might have) [from your browser, open http://0.0.0.0:6052 (if using defaults), click on the + to create a new node and follow the wizard).
6. Edit your config as required (if you are using ESPHome already, this is no different) and update your device OTA.

One thing I did notice is that although this is an old version of ESPHome, the "default" config created is slightly different from the current default, everything seems to work fine (including substitutions). e.g.

Fork version (v1.16.0-dev)

esphome:
  name: guest-big-light
  platform: ESP8266
  board: esp01_1m

Current version embedded in HA (2023.9.3)

esphome:
  name: "guest-big-light"
  friendly_name: Guest Big Room

esp8266:
  board: esp01_1m

I am sure there will be things that don't work correctly, but I haven't found anything yet.

As far as my light is concerned, The main problem has been resolved, but I have noticed that when connecting directly to the web server on the device, if I change the brightness while the light is off, the state toggle does not change. Also, I am getting inconsistent behaviour when turning the light on using the switch. Sometimes it goes to the last brightness and others, it comes on at 100%.

I am also seeing the following log entry whenever the brightness is set to 100% (255). The reason for this is pretty obvious (max_value is set to 254) and the interface is trying to set it to 255.

17:07:55 | [W] | [light:219] | 'State' - Brightness value 1.00 is out of range [0.0 - 1.0]!

One of the things that is sticking in my head is the race condition that was identified and I do feel like setting max_value to 254 is a workaround, rather than a fix as there seem to be something funky going on with brightness when transitioning from off to on.

slaroy, 2023-11-14:
I'm seeing the same problem on a Treatlife DS02 Dimmer. I would love to have the fix merged in so I can take advantage of the libretiny code for this BK7231T-based device.

inspoadam, 2023-11-14:
> I'm seeing the same problem on a Treatlife DS02 Dimmer. I would love to have the fix merged in so I can take advantage of the libretiny code for this BK7231T-based device.

@slaroy please take a look at https://github.com/esphome/issues/issues/1448#issuecomment-1781991725 as I managed to use the code provided by (the other) Adam and integrate it with the current version of ESP Home, without requiring a merge. Hope you manage to get a working config.

BTW. **The "fix" in my above post does not work properly** as it was picking up the unmodified code, which is why I was experiencing a partial fix. Once I made it use the right code, I experienced compatibility issues, which brings me nicely to the like here, which runs through the whole story.

I am wondering if, given the nature of the fix, whether the code needed refactoring, butbI decided to use it as I had invested a bunch of  time modifying it.