esphome/issues #853: esohomeflasher boot loop not fixex on mac version 1.20

HA-TB303, 2019-11-16:
This issue: https://github.com/esphome/issues/issues/598

1.20 Fixed it for windows, but the MAC os version 1.20 still has the same issue. Took me a 2 hours to figure that one out. ;)

OttoWinter, 2019-11-16:
I don't think that can be true.

The code is shared between macOS and Windows version - and the fix for it would certainly not be platform-specific.

Please check you're actually using the correct esphome-flasher version.

HA-TB303, 2019-11-16:
I have downloaded both today v1.20. On my iMac I can flash, but the esp32 get's in bootloop.
On the same MAC I run the windows 64bit version in parallels and I can flash it and it works.
I can repeat this action multiple times with the same result.

OttoWinter, 2019-11-17:
You keep saying 1.20 - but there's only 1.2.0, please confirm you mean 1.2.0

HA-TB303, 2019-11-18:
Sorry, 1.2.0

closed 2020-08-07 by stale[bot]