esphome/issues #854: Compiling error - Xiaomi HHCCJCY01 BLE Sensor

MaciejOlszewski, 2019-11-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io running on Proxmox Server
ESPHome addon version is 1.12.2

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32
**Affected component:**

Xiaomi HHCCJCY01 BLE Sensor

**Description of problem:**
Compiling fails with "Unable to find component" error.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "testtest"
  password: "testtest"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'test'

ota:
  password: 'test'

esp32_ble_tracker:

sensor:
  - platform: xiaomi_hhccjcy01
    mac_address: '94:2B:FF:5C:91:61'
    temperature:
      name: "Xiaomi HHCCJCY01 Temperature"
    moisture:
      name: "Xiaomi HHCCJCY01 Moisture"
    illuminance:
      name: "Xiaomi HHCCJCY01 Illuminance"
    conductivity:
      name: "Xiaomi HHCCJCY01 Soil Conductivity"
    battery_level:
      name: "Xiaomi HHCCJCY01 Battery Level"

```

**Logs (if applicable):**

```
INFO Reading configuration...
ERROR Unable to find component sensor.xiaomi_hhccjcy01
Failed config

sensor.xiaomi_hhccjcy01: [source /config/esphome/test.yaml:22]
  
  Platform not found: 'sensor.xiaomi_hhccjcy01'
  platform: xiaomi_hhccjcy01
  mac_address: 94:2B:FF:5C:91:61
  temperature: 
    name: Xiaomi HHCCJCY01 Temperature
  moisture: 
    name: Xiaomi HHCCJCY01 Moisture
  illuminance: 
    name: Xiaomi HHCCJCY01 Illuminance
  conductivity: 
    name: Xiaomi HHCCJCY01 Soil Conductivity
  battery_level: 
    name: Xiaomi HHCCJCY01 Battery Level

```

**Additional information and things you've tried:**

Config is copied from the documentation so I assume it should be correct. I have tried removing ESPHome addon from Hassio and reinstalling but I didn't help. 

It compiles fine after changing the sensor platform to "xiaomi_miflora"

Alex9779, 2019-11-17:
The Xiaomi sensors got refactored in 1.14, the docs are for 1.14, I have no idea if it is possible to access the old docs as you can with beta and next developing version so...

lwqcz, 2019-11-17:
Do you have recent FW in those Xiaomi's? Maybe related to https://github.com/esphome/issues/issues/260#issuecomment-487684720

OttoWinter, 2019-11-17:
> ESPHome addon version is 1.12.2

The latest is 1.14.2, please use that.

closed 2019-11-17 by OttoWinter