esphome/issues #855: stack overflow and boot loop in atm90e32 after uploading config file Core 1 Panic'ed

emudryj, 2019-11-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker/hassio in Ubuntu X86

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU ESP32
 expressif ESP32-WROOM-32D

Tried with ESPhome 1.14.2, 1.14.3 and 1.15dev getting similar results with all of them
I also tried OTA and serial communication flashing, again, with similar results

**Affected component:**
not sure
I think you refer to the Circuit Setup 6 channel energy monitor board

**Description of problem:**
When uploading config YAML file with config for more than four power monitor boards the ESP32 enters a constant boot loop 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
substitutions:
# Change the disp_name to something you want  
  disp_name: south_panel
# Interval of how often the power is updated
  update_time: 6s
# 30A/1V SCT-013-030: 8650
# 50A/1V SCT-013-050: 15420
# 100A/50ma SCT-013-000: 32498
  current_cal: '36000'
#9VAC Transformer for versions > meter v1.2
  voltage_cal: '7305'
esphome:
  name: south_panel
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "WiFi"
  password: "password"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "South Panel Fallback Hotspot"
    password: "password"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "password"

ota:
  password: "password"

web_server:
  port: 80

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23
sensor:
  - platform: wifi_signal
    name: ${disp_name} WiFi
    update_interval: 60s
[... 857 lines cut ...]
[23:15:00]Backtrace: 0x4008c8b0:0x3ffaf780 0x4008cae1:0x3ffaf7a0 0x4008caf8:0x3ffaf7c0 0x40089bc4:0x3ffaf7e0 0x4008b828:0x3ffaf800 0x4008b7de:0xfefefefe
[23:15:00]
[23:15:00]Rebooting...
[23:15:00]ets Jun  8 2016 00:22:57
[23:15:00]
[23:15:00]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[23:15:00]configsip: 0, SPIWP:0xee
[23:15:00]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[23:15:00]mode:DIO, clock div:2
[23:15:00]load:0x3fff0018,len:4
[23:15:00]load:0x3fff001c,len:952
[23:15:00]load:0x40078000,len:6084
[23:15:00]load:0x40080000,len:7944
[23:15:00]entry 0x40080310
[23:15:00]
Serial port closed!

```

**Additional information and things you've tried:**
Since everything seemed to work fine with the basic connectivity config, I tried to upload the YAML file little by little. 
Everything works fine until I get over line 360 (sometimes 400, but most of the time line360).
If I try to upload anything over that, the ESP32 enters in a boot loop, with the console error detailed above and all connectivity is lost, until flashed again through serial.   

Please, do not hesitate to contact me if you have ANY question.

Diramu, 2019-11-18:
I did some debugging with my setup.  I have 2 power_boards setup that works fine, until I add power_factor sensor, so I have left it off for now.  

1. I added yaml for a 3rd board, and it crashes similar as above
2. I delete  the yaml for the 2nd board (so a 2 board setup, but with the new board).  It worked.
3. Added back in, half the second board. with failed
4. reduce it to a single phase (of half a board). It failed, but got deeper into the state up.
5. delete add 'template' sensors.  It worked
6. added whole half board back in. It work
7. added there of the board (now all 3 boards without template sensors). I failed
8. when back to original 2 boards, added a single power_factor sensor. failed
9. reduced it to 1 board, a single power_factor, it work!
10. 1 board all 6 power_factor sensors. it worded

My conclusion.  It binary size dependant.  
Some of the failure above did get further into the set setup before crashing (and rebooting)
I can get 2 and half board to work without the extra 'template sensors'.
I can get 1 board to work with power_factor sensors.

My esp32 is reported as ESP32D0WDQ6 (revision 1).
All these test were done with the esp32 not pluged into the power monitor board.

CircuitSetup, 2019-11-18:
It seems like with this many sensors the ESP32 is running out of stack. A similar issue was posted here: https://github.com/esphome/issues/issues/773

After removing extra sensors and id's, I still got a meditation error with the following stack trace: 

```
PC: 0x4012a9ff
EXCVADDR: 0x00000000

Decoding stack results
0x40123887: nvs::Page::findItem(unsigned char, nvs::ItemType, char const*, unsigned int&, nvs::Item&, unsigned char, nvs::VerOffset) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_page.cpp line 745
0x400d77f5: wifi_loop() at C:\Users\John\AppData\Local\Temp\arduino_build_454298\sketch\wifi.cpp line 359
0x400d7a15: wifi_loop() at C:\Users\John\AppData\Local\Temp\arduino_build_454298\sketch\wifi.cpp line 472
0x400de161: checkDigestAuthentication(char const*, char const*, char const*, char const*, char const*, bool, char const*, char const*, char const*) at C:\Users\John\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp line 179
0x400de18e: checkDigestAuthentication(char const*, char const*, char const*, char const*, char const*, bool, char const*, char const*, char const*) at C:\Users\John\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp line 180
0x400d79bd: wifi_loop() at C:\Users\John\AppData\Local\Temp\arduino_build_454298\sketch\wifi.cpp line 451
0x400df17d: AsyncStaticWebHandler::_fileExists(AsyncWebServerRequest*, String const&) at C:\Users\John\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebHandlers.cpp line 145
0x400eb007: ultoa at C:\Users\John\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\stdlib_noniso.c line 85
0x40088ff9: pxPortInitialiseStack at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 227
```

It seems like the web services (captive portal, web server) are putting it over the edge. 

After removing all templates and extra id's I got it to work will all boards. But then you'd have to do the totals for amps & watts somewhere else, along with kwh. Note that I also removed the web server, restart switch, and captive portal. This is the config I used:

```
substitutions:
# Change the disp_name to something you want  
  disp_name: south_panel
# Interval of how often the power is updated
  update_time: 6s
# 30A/1V SCT-013-030: 8650
# 50A/1V SCT-013-050: 15420
# 100A/50ma SCT-013-000: 32498
  current_cal: '36000'
#9VAC Transformer for versions > meter v1.2
  voltage_cal: '7305'
esphome:
  name: south_panel
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "WIFI"
  password: "wifi_password"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23
sensor:
#IC1 Main
  - platform: atm90e32
    cs_pin: 5
    phase_a:
      voltage:
        name: ${disp_name} Volts A Main
        accuracy_decimals: 1
      current:
        name: ${disp_name} CT1 Amps
      power:
        name: ${disp_name} CT1 Watts 
      gain_voltage: ${voltage_cal}
[... 306 lines cut ...]
      power:
        name: ${disp_name} CT34 Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_b:
      current:
        name: ${disp_name} CT35 Amps
      power:
        name: ${disp_name} CT35 Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_c:
      current:
        name: ${disp_name} CT36 Amps
      power:
        name: ${disp_name} CT36 Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    line_frequency: 60Hz
    gain_pga: 1X
    update_interval: ${update_time}
time:
  - platform: sntp
    id: sntp_time
```

CircuitSetup, 2019-11-18:
@OttoWinter is there a switch somewhere to free up some stack? Maybe something to turn off unused global variables?

brandond, 2019-11-18:
I don't believe there's any way to do that. There are a few core things (wifi, logger, api, mqtt) that create globals when enabled. Because they need to be available to other components, they're present whenever enabled; the only way to get rid of them is to not include the component.

From what I'm seeing, platformio configures a 4k stack for the main task. I don't know if there's a good way to change this without rebuilding the entire SDK, though.

CircuitSetup, 2019-11-19:
Thanks @brandond. It sounds like allocating more stack for the main task could work. That setting is [here.](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/config/sdkconfig.h#L140) Changing that manually isn't that simple though, and it would probably revert every time an update was made.

A better long term solution may be to move some functions to a separate task though. I see that Otto moved the BLE stack into the main task previously because of race conditions, [as indicated here.](https://github.com/esphome/issues/issues/773#issuecomment-546633675)

OttoWinter, 2019-11-20:
@CircuitSetup 
> is there a switch somewhere to free up some stack? Maybe something to turn off unused global variables?

No, there is not.

The problem is all variables are used need to be stored somewhere (you can find them at the top of the generated `main.cpp`). Some versions ago, they were all local in `setup()`, but then custom code can't easily use them, which is not good. So now they're all global.

I don't see a good solution of moving the global variables somewhere else. Moving them back into `setup()` isn't going to fix the issue, as the highest stack depth use is probably in `App.setup()` anyway.

One option would be to move everything in a struct that is allocated on the heap, and only store a reference to it globally. Then, `id()` would need to be turned into a macro and some major changes in the codegen would need to be made. And it's quite possible we then run out of heap space.

I'd be interested in the heap usage - could you enable the debug component and print what it says? I'd be interested in knowing how much heap is free on these devices to judge how effective a fix could be.

> A better long term solution may be to move some functions to a separate task though.

That is for #773 and is just for BLE - the BLE feature uses a bunch of stack space. Putting it to a separate task would solve the problem there. This issue here is a different one - here it's globals that cause the majority of the stack use.

CircuitSetup, 2019-11-20:
@OttoWinter thanks for explaining. That all makes a lot of sense. I'll get some debug info your way. 

If there's enough heap, could more stack be allocated without editing sdkconfig.h?

CircuitSetup, 2019-11-20:
Here is the debug output with the above config:

```
[11:45:55][D][debug:023]: ESPHome version 1.15.0-dev
[11:45:55][D][debug:025]: Free Heap Size: 206288 bytes
[11:45:55][D][debug:053]: Flash Chip: Size=4096kB Speed=40MHz Mode=DIO
[11:45:55][D][debug:086]: Chip: Model=ESP32, Features=WIFI_BGN,BLE,BT, Cores=2, Revision=1
[11:45:55][D][debug:088]: ESP-IDF Version: v3.2.3-14-gd3e562907
[11:45:55][D][debug:091]: EFuse MAC: 0000644675AE114C
[11:45:55][D][debug:143]: Reset Reason: Software Reset CPU
[11:45:55][D][debug:186]: Wakeup Reason: Unknown
```

If I add one more sensor then I get a panic.

CircuitSetup, 2019-11-20:
@emudryj I also got it to work with the following. This is ideal if you really only need to get total power and kwh from that, although I left in current for the main board:

```
substitutions:
# Change the disp_name to something you want  
  disp_name: south_panel
# Interval of how often the power is updated
  update_time: 6s
# 30A/1V SCT-013-030: 8650
# 50A/1V SCT-013-050: 15420
# 100A/50ma SCT-013-000: 32498
  current_cal: '36000'
#9VAC Transformer for versions > meter v1.2
  voltage_cal: '7305'
esphome:
  name: south_panel
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "WIFI"
  password: "wifi_password"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23
sensor:
#IC1 Main
  - platform: atm90e32
    cs_pin: 5
    phase_a:
      voltage:
        name: ${disp_name} Volts A Main
        id: ic1Volts
        accuracy_decimals: 1
      current:
        name: ${disp_name} CT1 Amps
        id: ct1Amps
      power:
        name: ${disp_name} CT1 Watts 
        id: ct1Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_b:
      current:
        name: ${disp_name} CT2 Amps 
        id: ct2Amps
      power:
        name: ${disp_name} CT2 Watts 
        id: ct2Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
    phase_c:
      current:
        name: ${disp_name} CT3 Amps
        id: ct3Amps
      power:
        name: ${disp_name} CT3 Watts
        id: ct3Watts
      gain_voltage: ${voltage_cal}
      gain_ct: ${current_cal}
[... 328 lines cut ...]
    lambda: return id(ct31Watts).state + id(ct32Watts).state + id(ct33Watts).state + id(ct34Watts).state + id(ct35Watts).state + id(ct36Watts).state ;
    accuracy_decimals: 1
    unit_of_measurement: W
    icon: "mdi:flash-circle"
    update_interval: ${update_time}
#Total Watts
  - platform: template
    name: ${disp_name} Total Watts
    id: totalWatts
    lambda: return id(totalWattsMain).state + id(totalWattsAddOn1).state + id(totalWattsAddOn2).state + id(totalWattsAddOn3).state + id(totalWattsAddOn4).state + id(totalWattsAddOn5).state ;
    accuracy_decimals: 1
    unit_of_measurement: W
    icon: "mdi:flash-circle"
    update_interval: ${update_time}
#kWh
  - platform: total_daily_energy
    name: ${disp_name} Total kWh
    power_id: totalWatts
    filters:
      - multiply: 0.001
    unit_of_measurement: kWh
time:
  - platform: sntp
    id: sntp_time
```

Vxider, 2020-07-22:
Same problems when adding lots of switches, it is possible if add another esp32 as a slave and communicates by i2c?

seantrue, 2020-10-31:
I have the same issue on a configuration with hundreds of remote_receiver definitions. A way to increase (optionally) the stack size seems a sensible thing.

psgcooldog, 2020-12-09:
Any action on this?  I'm running into the same problem after adding several more items to my power monitoring setup.

BrentVerhaegen, 2020-12-30:
I'm running into the same/a similar issue on my power monitoring board.
Just 1 atm90e32 with 2 phases, but panics when surfing to the web server on an ESP32s.

CircuitSetup, 2020-12-30:
> I'm running into the same/a similar issue on my power monitoring board.
> Just 1 atm90e32 with 2 phases, but panics when surfing to the web server on an ESP32s.

Are there other sensors connected to the esp32? If not you may want to try a different esp32.

nikito7, 2021-03-18:
Check heap

```yaml
  - platform: template
    name: "ESP Free Heap"
    lambda: |-
      int heap = ESP.getFreeHeap();
      return heap / 1024.0;
    unit_of_measurement: "kB"
    update_interval: 5s
    icon: mdi:chip
```

numo68, 2021-04-03:
Any update on this? I am also running into a bootloop with many binary sensors. The heap seems to be fine - ESP.getFreeHeap() shows 118456 before adding the sensor causing the panic which seems reasonable given that I have a display consuming 75 kB for its buffer and relatively many other things.

My problem is definitely the stack. I changed the stack size in `~/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/main.cpp` from 8192 to 16384 and all is fine. The complete project leaves 100 kB heap after this.

@OttoWinter 
> I don't see a good solution of moving the global variables somewhere else. Moving them back into setup() isn't going to fix the issue, as the highest stack depth use is probably in App.setup() anyway.

Hmm, I don't really understand why an added sensor would need more of _stack_ space. I admit that I don't know esp32, platformio nor arduino much so maybe they do things differently but on platforms I am used to a variable outside of any method is allocated in an (initialized) RAM segmet and not on stack. That seems to be confirmed asking the image

```
.../xtensa-esp32-elf-objdump -x firmware.elf | grep automation_2
3ffc14d4 g     O .dram0.bss	00000004 automation_2
```

Most of them are pointers anyway and I guess they are generally setup one after another, not nested in some way, aren't they? Is there any way to pinpoint what exactly is the problem?

Thanks

glmnet, 2021-04-04:
The issue isn't the memory allocated globally, the issue is the stack size, most likely ESPHome not being "very" efficient in handling many components in loop task, which is everything.

glmnet, 2021-04-04:
Id like to know that too. I see the thing crashes quickly into the setup() call as if the stack is already full. I see no reason why it is full already though 

In case you or anyone else don't know, this issue can be fixed by increasing the stack size. If you need more info on how to do that ask me here (I'm on mobile now)

numo68, 2021-04-04:
Does FreeRTOS, ESP, Arduino or esphome have some way of tracking the stack usage, e.g. by pre-filling it and looking at how much was clobbered or something? I've seen things like that in other embedded OS-es.

8 kB is a lot and more likely than not this is just someone doing `char foo[2048]` in some method. My prime suspect would be either the WiFi or the Home Assistant API as I had an instance of a situation where any interaction during the initial phase - but after the setup - caused a crash but it was working fine when everything settled, but of course the reality is often completely different with stack issues :)

And it still does not explain why it seems to be dependent of number of objects, that does not make much sense, unless the stack overflows into the heap or the used BSS storage making mess. That is actually a possibility as I _think_ that I have seen messages `Component already registered` before the crash, not seeing them when all was fine.

ssieb, 2021-06-06:
https://github.com/esphome/esphome/pull/1651 is related and there's a suggestion for how to increase the stack size.

marrobHD, 2021-06-27:
How can I increase the stack size? The platformio.ini gets overwritten.

glmnet, 2021-06-28:
until https://github.com/esphome/esphome/pull/1651 is merged or something better happens, easiest solution might be editing this file...

> My problem is definitely the stack. I changed the stack size in `~/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/main.cpp` from 8192 to 16384 and all is fine. The complete project leaves 100 kB heap after this.

marrobHD, 2021-07-04:
I'm unable to access the .platformio folder.

JeeCee1, 2021-07-23:
If you are using the ESPHome add-on it's located **in** the container.
Easiest way to access the container is to install the Portainer add-on. Then select the ESPHome container, then go to `console`.
`Main.cpp` is located here: `~/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp`.
I couldn't find an editor (I'm not very experienced in Linux nor Docker) so I installed it.

aav7fl, 2021-08-22:
I want to copy my comment from https://github.com/esphome/esphome/pull/1651 to here in case someone missed it.

Increasing the stack size resolved my issue as well.

The https://github.com/esphome/esphome/pull/1651 PR _does_ add build flags to the compile pipeline. However, I don't think `-DARDUINO_LOOP_STACK_SIZE` will work _yet_.

After some digging, I found that we can't modify the `ARDUINO_LOOP_STACK_SIZE` at _compile time_ until https://github.com/espressif/arduino-esp32/pull/5173.

That PR was merged recently and it part of the Arduino [2.0.0-rc1 release](https://github.com/espressif/arduino-esp32/releases/tag/2.0.0-rc1). 

Since ESPHome is still pointing at `1.0.6`, it won't work without some changes.

Possible solutions:
1. Follow [this thread](https://github.com/espressif/arduino-esp32/issues/5436) and you might be able to add some of the packages required to change the `arduino_version` to `2.0.0-rc1`. (`latest` didn't work for me at the time of posting this).
2. (My preferred method) Wait until:
* The `espressif/arduino-esp32` library is fully released to `2.0.0`
* ESPHome updates to support it (since there are breaking changes)
  Only then can you use the build config `-DCONFIG_ARDUINO_LOOP_STACK_SIZE=32768` to increase the loop stack size.

As others have said, in the meantime just update the stack value inside `.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp`. 

from:

```cpp
xTaskCreateUniversal(loopTask, "loopTask", ARDUINO_LOOP_STACK_SIZE, NULL, 1, &loopTaskHandle, CONFIG_ARDUINO_RUNNING_CORE);
```

to: 

```cpp
xTaskCreateUniversal(loopTask, "loopTask", 32768, NULL, 1, &loopTaskHandle, CONFIG_ARDUINO_RUNNING_CORE);
```

It will be a manual process until things are released and updated upstream.

shbatm, 2021-08-29:
@aav7fl Thanks for your comment. I had to make the same manual change on my ESP32 board when adding a 2nd add-on card for the 6-channel Energy Monitor.

julupanter, 2021-11-05:
> 
> 
> The following configuration changes allowed me to switch to using Arduino 2.0.0 and successfully increase the `CONFIG_ARDUINO_LOOP_STACK_SIZE` to 32768
> 
> ```yaml
> esphome:
>   name: power-monitor
>   platformio_options:
>     platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
>     platform_packages:
>       - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0
>     build_flags: 
>       - -DCONFIG_ARDUINO_LOOP_STACK_SIZE=32768
> 
> esp32:
>   board: nodemcu-32s
>   framework:
>     type: arduino
>     version: 2.0.0
> ```

Not working on a raspberry pi 4, <<Error: Could not find the package with 'espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r1' requirements for your system 'linux_aarch64'>>

I have install a complete backup in a widows virtual machine to get it working.

Thanks!

sebPomme, 2021-12-30:
Thanks you guyz for this hard debug. @noderat the fix works for me.

mountainbiker54, 2022-03-16:
> ```yaml
>     build_flags: 
>       - -DCONFIG_ARDUINO_LOOP_STACK_SIZE=32768
> ```

How do I actually edit this file? I tried to access it from the CLI in my home assistant but normal linux commands do not work.

mountainbiker54, 2022-03-16:
"My problem is definitely the stack. I changed the stack size in ~/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/main.cpp from 8192 to 16384 and all is fine. The complete project leaves 100 kB heap after this."

How do I actually edit this file? I tried to access it from the CLI in my home assistant but normal linux commands do not work.

aav7fl, 2022-03-16:
> "My problem is definitely the stack. I changed the stack size in ~/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/main.cpp from 8192 to 16384 and all is fine. The [...]

I've been using noderat's solution posted above for the last few months. That way I don't need to remote into the container to make changes.

mountainbiker54, 2022-03-16:
@aav7fl Regrettably I don't know how to do that either. I tried putting it in the ESPHome configuration tab in HA but it had no effect when I recompiled.

aav7fl, 2022-03-16:
> @aav7fl Regrettably I don't know how to do that either. I tried putting it in the ESPHome configuration tab in HA but it had no effect when I recompiled.

How certain are you that it's uploading the compiled firmware to the hardware you're trying to flash?

Does it error out when compiling or display any warnings?

mountainbiker54, 2022-03-16:
I'm doing a manual download, then I take the .bin file and flash it with ESPHome Flasher. This is the error I get. 

`Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[12:06:04]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
[12:06:04]Debug exception reason: Stack canary watchpoint triggered (loopTask) 
[12:06:04]Core 1 register dump:
[12:06:05]PC      : 0x40111f84  PS      : 0x00060736  A0      : 0x801177b1  A1      : 0x3ffafdc0  
[12:06:05]A2      : 0x3ffb81bc  A3      : 0x3ffb00d0  A4      : 0x3f401980  A5      : 0x3ffb0200  
[12:06:05]A6      : 0x3ffb01d0  A7      : 0x00000008  A8      : 0x3ffb00d0  A9      : 0x3ffafdb0  
[12:06:05]A10     : 0x00000001  A11     : 0x3ff9c510  A12     : 0x000000ff  A13     : 0x0000ff00  
[12:06:05]A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001f  EXCCAUSE: 0x00000001  
[12:06:05]EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
[12:06:06]
[12:06:06]ELF file SHA256: 0000000000000000
[12:06:06]
[12:06:06]Backtrace: 0x40111f84:0x3ffafdc0 0x401177ae:0x3ffb00d0 0x401177ea:0x3ffb0160 0x400d8a75:0x3ffb01a0 0x400d8aa9:0x3ffb01d0 0x400d8add:0x3ffb0220 0x400d8c9d:0x3ffb0240 0x400e14ed:0x3ffb0260 0x400e151a:0x3ffb0290 0x400d8c4d:0x3ffb02f0 0x400e2582:0x3ffb0320 0x400f5e5e:0x3ffb1fb0 0x40089c1e:0x3ffb1fd0
[12:06:06]
[12:06:06]Rebooting...
[12:06:06]ets Jun  8 2016 00:22:57
[12:06:06]
[12:06:06]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[12:06:06]configsip: 0, SPIWP:0xee
[12:06:07]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:06:07]mode:DIO, clock div:2
[12:06:07]load:0x3fff0018,len:4
[12:06:07]load:0x3fff001c,len:1044
[12:06:07]load:0x40078000,len:8896
[12:06:07]load:0x40080400,len:5828
[12:06:07]entry 0x400806ac
[12:06:07]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
[12:06:07]Debug exception reason: Stack canary watchpoint triggered (loopTask) 
[12:06:07]Core 1 register dump:
[12:06:07]PC      : 0x40111f84  PS      : 0x00060736  A0      : 0x801177b1  A1      : 0x3ffafdc0  
[12:06:08]A2      : 0x3ffb8100  A3      : 0x3ffb00d0  A4      : 0x3f401980  A5      : 0x3ffb0200  
[12:06:08]A6      : 0x3ffb01d0  A7      : 0x00000008  A8      : 0x3ffb00d0  A9      : 0x3ffafdb0  
[12:06:08]A10     : 0x00000001  A11     : 0x3ff9c510  A12     : 0x000000ff  A13     : 0x0000ff00  
[12:06:08]A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001f  EXCCAUSE: 0x00000001  
[12:06:08]EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
[12:06:08]
[12:06:08]ELF file SHA256: 0000000000000000
[12:06:08]
[12:06:09]Backtrace: 0x40111f84:0x3ffafdc0 0x401177ae:0x3ffb00d0 0x401177ea:0x3ffb0160 0x400d8a75:0x3ffb01a0 0x400d8aa9:0x3ffb01d0 0x400d8add:0x3ffb0220 0x400d8c9d:0x3ffb0240 0x400e14ed:0x3ffb0260 0x400e151a:0x3ffb0290 0x400d8c4d:0x3ffb02f0 0x400e2582:0x3ffb0320 0x400f5e5e:0x3ffb1fb0 0x40089c1e:0x3ffb1fd0
[12:06:09]
[12:06:09]Rebooting...`

mountainbiker54, 2022-03-20:
That doesn't work for me either :(

CircuitSetup, 2022-03-22:
> That doesn't work for me either :(

You're getting a guru meditation error with the above in the ESPHome config? Before you build are you doing a "Clean Build Files" in the ESPHome interface?

mountainbiker54, 2022-03-22:
1. I put the code in the ESP Home Config Page: [config](https://mike.generalllc.net/projects/powermon/Image%201427.png)
2.  I restart ESP Home
3.  I enter my code into the ESP Home module (don't know what this is actually called): [ Code](https://mike.generalllc.net/projects/powermon/Image%201428.png)
4. I click Install
5. Select Manual Download [download](https://mike.generalllc.net/projects/powermon/Image%201429.png)
6. The code then compiles and I have a .bin file. [compile](https://mike.generalllc.net/projects/powermon/Image%201430.png)
7. I flash the file using ESP home to the esp board [flash](https://mike.generalllc.net/projects/powermon/Image%201431.png)

If I drop down to <19 sensor the flash works, and the esp check in with Home Assistant and I can browse to the page. I'm assuming I put the ESP home config in the wrong place, or maybe the compiler is not seeing the new settings?

I'm running ESP Home 2022.2.6 [version](https://mike.generalllc.net/projects/powermon/Image%201433.png)

CircuitSetup, 2022-03-22:
> 1. I put the code in the ESP Home Config Page: [config](https://mike.generalllc.net/projects/powermon/Image%201427.png)

This should be in the individual esp32 config, not the plugin config.

mountainbiker54, 2022-03-22:
At the top of my code like this? If so that isn't compiling for me. 

`# Example config for when jp8-jp11 are all bridged - this connects all the voltage channels and allows for power to be calculated by the meter directly.
# Boards >= v1.4 jp8-jp11 are removed and have all voltage channels connected
# 1 add-on board
esphome:
  name: power-monitor
  platformio_options:
    build_flags:
      - '-DCONFIG_ARDUINO_LOOP_STACK_SIZE=32768'
esp32:
  board: nodemcu-32s
  variant: esp32
  framework:
    type: arduino
    version: 2.0.2
    source: https://github.com/espressif/arduino-esp32.git#2.0.2
    platform_version: >-
      https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream

substitutions:
# Change the disp_name to something you want  
  disp_name: powermon
# Interval of how often the power is updated
  update_time: 10s
# Current Transformers:
#  20A/25mA SCT-006: 11143
#  30A/1V SCT-013-030: 8650
#  50A/1V SCT-013-050: 15420
#  80A/26.6mA SCT-010: 41660
#  100A/50ma SCT-013-000: 27518
#  120A/40mA: SCT-016: 41787
#  200A/100mA SCT-024: 27518
  current_cal: '27518'
# Jameco 9VAC Transformer: 
#  For meter versions: 
#  >= v1.3: 7305
#  <= v1.2: 42620
  voltage_cal: '7305'

esphome:
  name: powermon
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
    
# mqtt:
#  broker: !secret mqtt_broker
#  username: !secret mqtt_user
#  password: !secret mqtt_pass

# Enable logging
logger:

# Enable Home Assistant API
api:`

CircuitSetup, 2022-03-22:
Yes, just take out the second esphome

mountainbiker54, 2022-03-25:
That did the trick. Thank you!

lux4rd0, 2022-03-28:
Just ran into this myself as I'm adding about 100 Home Assistant sensors to be displayed on an e-ink display. Working backward from a continuous reboot cycle when I added the "n+1" sensor wasn't easy - but this thread did the trick. Thanks!!

mr-sneezy, 2022-04-22:
Guys I've applied the wisdom found in this thread to my ESPhome code (I used @noderat suggested code block), but I've hit a new problem that has created. When starting compilation with ESPhome I get a fatal error related to an unfound package. 
Anyone got a tip on how to sort that out ?

`INFO Generating C++ source...
INFO Compiling app...
Processing irrigation-controller (board: nodemcu-32s; framework: arduino; platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream)
--------------------------------------------------------------------------------
Configuring from remote
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2
Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2
Error: Could not find the package with 'espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2' requirements for your system 'linux_aarch64'`

I'm running ESPhome on an Raspberry Pi 4B and HASSOS...

jpeletier, 2022-06-18:
@mountainbiker54 @CircuitSetup @lux4rd0 @emudryj could you check if [#3577](https://github.com/esphome/esphome/pull/3577) fixes your issue by reducing the stack usage?

aav7fl, 2022-06-18:
@jpeletier Is there a prebuilt docker container for your PR to test with? Or will I need to build it myself to run/compile/test?

jpeletier, 2022-06-18:
@aav7fl there you go: `docker pull epiclabs/esphome:fix-885`

aav7fl, 2022-06-18:
@jpeletier Thanks! It was a success for me. 🥳 

I left a comment over on [#3577. ](https://github.com/esphome/esphome/pull/3577).

jpeletier, 2022-06-18:
Easy way to test this thanks to @lux4rd0 is:

1. remove the changes you added to increase the stack size
2. Add the below to the root of your configuration YAML:

```yaml
external_components:
  - source: github://pr#3577
    components: wifi
    refresh: 0s
```

arthurkomatsu, 2022-08-04:
+1 for this issue, I am affected by this, in my case I needed to use both @jpeletier and @noderat for my project to work. Also, it is generating a lot of warnings of type:

```~/.platformio/packages/framework-arduinoespressif32@src-afacabbe58d012c05a6ee24950dc6ba1/tools/sdk/esp32/include/config/sdkconfig.h:68: warning: "CONFIG_ARDUINO_LOOP_STACK_SIZE" redefined
 #define CONFIG_ARDUINO_LOOP_STACK_SIZE 8192
```

closed 2022-10-04 by jesserockz