esphome/issues #856: Sonoff T1 v.1.0 (3 gang, 2 gang, 1 gang) ghost switching

MariusLaurinaitis, 2019-11-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**
Sonoff T1 v1.0

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```
esphome:
  name: 11_jungiklis
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "........."
  password: "........"
  fast_connect: ON
  
# Enable logging
logger:

# Enable Home Assistant API
api:
  password: .........

ota:
  password: ..........

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
    name: "Vaiku_darbo"
    filters:
      - invert:
      - delayed_on: 50ms
      - delayed_off: 50ms
    on_press:
      - switch.toggle: relay_1
  - platform: gpio
    pin:
      number: GPIO9
      mode: INPUT_PULLUP
    name: "Vaiku_miegamas_1"
    filters:
      - invert:
      - delayed_on: 50ms
      - delayed_off: 50ms
    on_press:
      - switch.toggle: relay_2
  - platform: status
    name: "11_jungiklis_statusas"

switch:
  - platform: gpio
    name: "Vaiku_darbo"
    pin: GPIO12
    id: relay_1
  - platform: gpio
    name: "Vaiku_miegamas_1"
    pin: GPIO5
    id: relay_2

output:
  # Register the blue LED as a dimmable output ....
  - platform: esp8266_pwm
    id: blue_led
    pin: GPIO13
    inverted: True

light:
  # ... and then make a light out of it.
  - platform: monochromatic
    name: "11_jungiklis_led"
    output: blue_led

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
ESPHome Dashboardmore_vert
OTA (Over-The-Air)
memory
0_jungiklismore_vert
 . Full path: /config/esphome/0_jungiklis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
10_jungiklismore_vert
 . Full path: /config/esphome/10_jungiklis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
11_jungiklismore_vert
 . Full path: /config/esphome/11_jungiklis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
12_jungiklismore_vert
 . Full path: /config/esphome/12_jungiklis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
13_jungiklismore_vert
 . Full path: /config/esphome/13_jungiklis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
14_jungiklismore_vert
 . Full path: /config/esphome/14_jungiklis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
15_jungiklismore_vert
 . Full path: /config/esphome/15_jungiklis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
16_jungiklismore_vert
 . Full path: /config/esphome/16_jungiklis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
17_jungiklismore_vert
 . Full path: /config/esphome/17_jungiklis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
18_jungiklismore_vert
 . Full path: /config/esphome/18_jungiklis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
19_jungiklismore_vert
 . Full path: /config/esphome/19_jungiklis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
1_h801more_vert
 . Full path: /config/esphome/1_h801.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
1_jungiklismore_vert
 . Full path: /config/esphome/1_jungiklis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
22_jungiklismore_vert
 . Full path: /config/esphome/22_jungiklis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
2_h801more_vert
 . Full path: /config/esphome/2_h801.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
3_h801more_vert
 . Full path: /config/esphome/3_h801.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
4_jungiklismore_vert
 . Full path: /config/esphome/4_jungiklis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
5_jungiklismore_vert
 . Full path: /config/esphome/5_jungiklis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
6_jungiklismore_vert
 . Full path: /config/esphome/6_jungiklis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
7_jungiklismore_vert
 . Full path: /config/esphome/7_jungiklis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
8_jungiklismore_vert
 . Full path: /config/esphome/8_jungiklis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
9_jungiklismore_vert
 . Full path: /config/esphome/9_jungiklis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
daviklis_1more_vert
 . Full path: /config/esphome/daviklis_1.yaml

system_update Update Available! 1.12.2 ➡️1.14.2

UPLOAD EDIT SHOW LOGS VALIDATE
memory
gartraukis_rekuperacijamore_vert
 . Full path: /config/esphome/gartraukis_rekuperacija.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
gartraukis_rekuperacija_kontaktoriusmore_vert
 . Full path: /config/esphome/gartraukis_rekuperacija_kontaktorius.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
kamera_1more_vert
 . Full path: /config/esphome/kamera_1.yaml

system_update Update Available! 1.13.6 ➡️1.14.2

UPLOAD EDIT SHOW LOGS VALIDATE
memory
lauko_sviesamore_vert
 . Full path: /config/esphome/lauko_sviesa.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
rozeciu_reles_1_1more_vert
 . Full path: /config/esphome/rozeciu-reles-1_1.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
rozeciu_reles_1_2more_vert
 . Full path: /config/esphome/rozeciu_reles_1_2.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
rozeciu_reles_2_1more_vert
 . Full path: /config/esphome/rozeciu_reles_2_1.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
rozeciu_reles_2_2more_vert
 . Full path: /config/esphome/rozeciu_reles_2_2.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
temp_daviklis_1more_vert
 . Full path: /config/esphome/temp_daviklis_1.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
torseras_rgbmore_vert
 . Full path: /config/esphome/torseras-rgb.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
valgomasismore_vert
 . Full path: /config/esphome/valgomasis.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
memory
vonios_kolonelemore_vert
 . Full path: /config/esphome/vonios_kolonele.yaml

UPLOAD EDIT SHOW LOGS VALIDATE
Show Logs 11_jungiklis.yaml
INFO Reading configuration /config/esphome/11_jungiklis.yaml...
WARNING ESP8266: Pin 9 (6-11) might already be used by the flash interface. Be warned.
INFO Starting log output from 11_jungiklis.local using esphome API
INFO Connecting to 11_jungiklis.local:6053 (192.168.1.119)
INFO Successfully connected to 11_jungiklis.local
[12:08:26][I][app:100]: ESPHome version 1.14.2 compiled on Nov 18 2019, 07:20:31
[12:08:26][C][wifi:409]: WiFi:
[12:08:26][C][wifi:277]:   SSID: [redacted]
[12:08:26][C][wifi:278]:   IP Address: 192.168.1.119
[12:08:26][C][wifi:280]:   BSSID: [redacted]
[12:08:26][C][wifi:281]:   Hostname: '11_jungiklis'
[12:08:26][C][wifi:285]:   Signal strength: -59 dB ▂▄▆█
[12:08:26][C][wifi:289]:   Channel: 8
[12:08:26][C][wifi:290]:   Subnet: 255.255.255.0
[12:08:26][C][wifi:291]:   Gateway: 192.168.1.254
[12:08:26][C][wifi:292]:   DNS1: 192.168.1.254
[12:08:26][C][wifi:293]:   DNS2: (IP unset)
[12:08:27][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Vaiku_darbo'
[12:08:27][C][gpio.binary_sensor:016]:   Pin: GPIO0 (Mode: INPUT_PULLUP)
[12:08:27][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Vaiku_miegamas_1'
[12:08:27][C][gpio.binary_sensor:016]:   Pin: GPIO9 (Mode: INPUT_PULLUP)
[12:08:27][C][switch.gpio:042]: GPIO Switch 'Vaiku_darbo'
[12:08:27][C][switch.gpio:043]:   Pin: GPIO12 (Mode: OUTPUT)
[12:08:27][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[12:08:27][C][switch.gpio:042]: GPIO Switch 'Vaiku_miegamas_1'
[12:08:27][C][switch.gpio:043]:   Pin: GPIO5 (Mode: OUTPUT)
[12:08:27][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[12:08:27][C][esp8266_pwm:022]: ESP8266 PWM:
[12:08:27][C][esp8266_pwm:023]:   Pin: GPIO13 (Mode: OUTPUT)
[12:08:27][C][esp8266_pwm:024]:   Frequency: 1000.0 Hz
[12:08:27][C][esp8266_pwm:025]:   Inverted: YES
[12:08:27][C][logger:175]: Logger:
[12:08:27][C][logger:176]:   Level: DEBUG
[12:08:27][C][logger:177]:   Log Baud Rate: 115200
[12:08:27][C][logger:178]:   Hardware UART: UART0
[12:08:27][C][light:177]: Light '11_jungiklis_led'
[12:08:27][C][light:179]:   Default Transition Length: 1.0s
[12:08:27][C][light:180]:   Gamma Correct: 2.80
[12:08:27][C][status:034]: Status Binary Sensor '11_jungiklis_statusas'
[12:08:27][C][status:034]:   Device Class: 'connectivity'
[12:08:27][C][ota:029]: Over-The-Air Updates:
[12:08:27][C][ota:030]:   Address: 11_jungiklis.local:8266
[12:08:27][C][ota:032]:   Using Password.
[12:08:27][C][api:095]: API Server:
[12:08:27][C][api:096]:   Address: 11_jungiklis.local:6053
© 2019 Copyright ESPHome, Made with MaterializeESPHome 1.14.2 Documentation
```

**Additional information and things you've tried:**

 I used filters but after ESPhome update it started again, mostly ghost switch at night, log shows nothing. (Sonoff T1, v1.0. 3 gang, 2 gang, 1 gang).

brandond, 2019-11-18:
What do you mean by ghost switching? Can you turn the log level up to verbose and leave the esphome command running overnight collecting output?

MariusLaurinaitis, 2019-11-18:
The switch activates itself at different times. I did not notice any causal link. Sometimes two switches turn on, sometimes one. I have 23 switches (Sonoff T1), different gang numbers. Sometimes a 1 gang, sometimes a 3 gang switch turns itself on. Of the 3 gangs, only one gang is triggered.

I did
logger:
   level: VERBOSE
Didn't understand "and leave the esphome command running", need to watch SHOW LOGS?

brandond, 2019-11-18:
Rebuild and reflash with level: VERBOSE.

The device does not store logs, so if you want to collect logs overnight you will need to leave a command up with the `esphome NODENAME.yaml logs` command running, or leave the hass.io add-on logs window open to collects logs in a browser if you're doing everything via the add-on.

MariusLaurinaitis, 2019-11-18:
Thanks,
ok i will try. It will be difficult because there are many switches and they switch in different ways. But I'll start with the most common switch.

OttoWinter, 2019-11-20:
Sounds kinda like an electrical issue to me - you're already using binary sensor filters, I'd recommend maybe upping the filter values a bit and using `delayed_on_off` instead of on+off filters.

MariusLaurinaitis, 2019-11-20:
My Sonoff T1 is v1.0, my neighbor also has more than 20 Sonoff T1 but its v2.0. He has no problem. Zoomed in to 50ms and it seems like two nights are quiet :)

minsuke, 2019-11-29:
> My Sonoff T1 is v1.0, my neighbor also has more than 20 Sonoff T1 but its v2.0. He has no problem. I did 50ms and it seems like two nights are quiet :)

I have same problem, ghost switch on GPIO9 

it warns when it check log, and sometimes GPIO9 connected switch is ON itself. 

It saids not just warn, I think it affects something.

sergiorademacher, 2019-12-27:
Hi.

This might be related to a hardware problem with some Sonoff T1 batches.  Sometimes you hear three beeps: beep, pause, then two beeps and then it turns on.   See https://github.com/arendst/Tasmota/issues/5449.  

Basically the Sonoff thinks that there is a long press, hence the beeps.

The fix with Tasmota, was to implement a option to ignore a long press (SetOption40).  What would be the equivalente with ESPHome?

sergiorademacher, 2019-12-29:
Hi.

Just for closure of this, I fixed the ghost switching this way:

`- platform: gpio
    pin:
      number: GPIO9
      mode: INPUT_PULLUP
      inverted: True
    id: button_2
    on_click:
    - min_length: 5ms
      max_length: 999ms
      then:
        - light.toggle: light_2
    - min_length: 1000ms
      max_length: 10000ms
      then:
        - logger.log: "Ghost Switch button_2"
`

MariusLaurinaitis, 2020-01-21:
For me its working good!

`binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
    name: "Laiptine_2"
    filters:
      - invert:
      - delayed_on_off: 50ms
    on_press:
      - switch.toggle: relay_1`

ryrzy, 2020-01-22:
> For me its working good!
> 
> ```
> binary_sensor:
>   - platform: gpio
>     pin:
>       number: GPIO0
>       mode: INPUT_PULLUP
>     name: "Laiptine_2"
>     filters:
>       - invert:
>       - delayed_on_off: 50ms
>     on_press:
>       - switch.toggle: relay_1
> ```

will try this... because config with min_length/max_length dosen't work for me :( still lights switch off/on by it self ;/

stkyoung, 2020-01-22:
I have a Sonoff T2 EU 2 Gang switch and had the same ghosting problems with Tasmota and ESPHome. Right now I run ESPHome without the ghosting issue. I left the switch powered off for a while. I then connected the wires to the switch without the front cover. After that, I flipped the circuit breaker to provide power. The capacitive switches are so sensitive, that they can register a touch while my finger hovers above the surface of the sensor. After that, I added the front cover. This action is registered as a long press, because the switch was beeping. After a couple touches, the sensors started to detect touch of my finger. My switch worked reliable after that. I tried this procedure a couple of times. I don't use any special things in the yaml code to fix the ghosting problem.
Could there be an automatic calibration process after power-up? I am curious if this is really the case or that this procedure works for me by good luck.

felipenanoclick, 2020-01-25:
> Hi.
> 
> Just for closure of this, I fixed the ghost switching this way:
> 
> ```yaml
> - platform: gpio
> 
>     pin:
>       number: GPIO9
>       mode: INPUT_PULLUP
>       inverted: True
>     id: button_2
>     on_click:
>     - min_length: 5ms
>       max_length: 999ms
>       then:
>         - light.toggle: light_2
>     - min_length: 1000ms
>       max_length: 10000ms
>       then:
>         - logger.log: "Ghost Switch button_2"
> ```

> > Hi.
> > Just for closure of this, I fixed the ghost switching this way:
> > ```yaml
> > - platform: gpio
> > 
> >     pin:
> >       number: GPIO9
> >       mode: INPUT_PULLUP
> >       inverted: True
> >     id: button_2
> >     on_click:
> >     - min_length: 5ms
> >       max_length: 999ms
> >       then:
> >         - light.toggle: light_2
> >     - min_length: 1000ms
> >       max_length: 10000ms
> >       then:
> >         - logger.log: "Ghost Switch button_2"
> > ```
> 
> hello, did this multiclick config fixed your problem with ghost switching on GPIO9 ? i have this problem to on Touch Switch (not sonoff but that same pinouts ;) ).
> Tonight i had light on xD

> Hi.
> 
> Just for closure of this, I fixed the ghost switching this way:
> 
> ```yaml
> - platform: gpio
> 
>     pin:
>       number: GPIO9
>       mode: INPUT_PULLUP
>       inverted: True
>     id: button_2
>     on_click:
>     - min_length: 5ms
>       max_length: 999ms
>       then:
>         - light.toggle: light_2
>     - min_length: 1000ms
>       max_length: 10000ms
>       then:
>         - logger.log: "Ghost Switch button_2"
> ```

not work to me

felipenanoclick, 2020-01-25:
> For me its working good!
> 
> ```
> binary_sensor:
>   - platform: gpio
>     pin:
>       number: GPIO0
>       mode: INPUT_PULLUP
>     name: "Laiptine_2"
>     filters:
>       - invert:
>       - delayed_on_off: 50ms
>     on_press:
>       - switch.toggle: relay_1
> ```

not work to me

sergiorademacher, 2020-01-25:
@felipenanoclick are you seeing the message "Ghost Switch button_2” on Home Assistant’s logbook when the ghost switch happens?

The ghost switch is caused by a hardware issue that makes the switch believe that there's a long physical press (several seconds ).  So I program the switch to actually switch the light if the press is below one second and to log it to HA if it's over one second.

You could try also try increasing the first min_length to avoid electrical bouncing.

ryrzy, 2020-01-26:
@sergiorademacher for me it didn't worked :( but i have make something like this:

`    filters:
      - delayed_on_off: 50ms
    on_click:
    - min_length: 5ms
      max_length: 59ms
      then:
        - logger.log: GHOST short ${name} Relay R
        
    - min_length: 60ms
      max_length: 999ms
      then:
        - switch.toggle: relay_R

    - min_length: 1000ms
      max_length: 10000ms
      then:
        - logger.log: GHOST long ${name} Relay R`

and for now it's good :) my ghost switching was not always related with long press gpio9 (with beep for RF coding), sometimes i had short switches. if this will be still a problem i think i will change my wall switches to zigbee or wifi but wihout touch button like this:

https://www.aliexpress.com/item/-/4000289352430.html
or
https://www.aliexpress.com/item/-/4000298926256.html

felipenanoclick, 2020-01-28:
> @ryrzy sim, corrigiu a mudança de fantasma para mim.
> 
> @stkyoung não acha que seja um problema de calibração, mas remover a tampa realmente corrige o problema, mas infelizmente não é seguro.
> 
> @felipenanoclick você está vendo a mensagem "Botão Ghost Switch_2" no diário de bordo do Assistente de residência quando a mudança fantasma acontece?
> 
> O comutador fantasma é causado por um problema de hardware que faz com que o comutador acredite que há uma pressão física longa (vários segundos). Então, programo o interruptor para realmente mudar a luz se a impressora estiver abaixo de um segundo e registrá-la no HA se houver mais de um segundo.
> 
> Você também pode tentar aumentar o primeiro min_length para evitar quedas elétricas.

[image]

[image]

[image]

[image]

Above is the ghost log.

Atheists of ESPHOME I used tasmota and the problem is identical.

felipenanoclick, 2020-03-17:
whith tasmota 8.11 firmware, bye bye ghost!

Fusseldieb, 2020-06-15:
> whith tasmota 8.11 firmware, bye bye ghost!

But I want to use ESPHome... All 30 T1's have ESPHome and I don't want to switch, because ESPHome is much safer, as the whole config is flashed and can not easily be corrupted.

Fusseldieb, 2020-06-15:
> For me its working good!
> 
> ```
> binary_sensor:
>   - platform: gpio
>     pin:
>       number: GPIO0
>       mode: INPUT_PULLUP
>     name: "Laiptine_2"
>     filters:
>       - invert:
>       - delayed_on_off: 50ms
>     on_press:
>       - switch.toggle: relay_1
> ```

Doesn't work for me. I've tried using this and min_length (50) and max_length (350) **combined** to no avail. They beep and some light gets turned on.

Later I disassembled a Sonoff and ripped off the RF receiver, since I don't even use it and I thought he was part of the problem. They still beep.

After that, I ripped off all resistors and capacitors belonging to that chip, basically un-populating the whole bottom part of the board. Still beeps.

Ultimatively I removed the transistor responsible for the beeper from the board. As expected, no more beeping, but instead lights just randomly come on and off.

I don't know anymore.

sergiorademacher, 2020-06-16:
Hi @Fusseldieb 

In my experience, the only way of getting rid of the beeping was to remove the buzzer from the bottom part of the T1.  Regarding fixing the ghost switching, it largely depends on the unit.  In some cases it is just a matter of using the min/max setting, as the unit thinks it is a long press, after all, it is a hardware issue.  With more rebellious units, the only solution is to swap it out.  It is supposed to be fixed with tasmota 8.1, but personally I rather stick with esphome.

Maybe @OttoWinter has a solution for it.

Fusseldieb, 2020-06-16:
> Hi @Fusseldieb
>
> In my experience, the only way of getting rid of the beeping was to remove the buzzer from the bottom part of the T1. Regarding fixing the ghost switching, it largely depends on the [...]

What boggles my mind is that they were fine for 1 year and then begun doing such crazy things, and every flash it gets worse...

Yea, I know that it's supposed to be fixed in tasmota 8.1, but **how**?

I hav some soldering skills, so I was considering of swapping out the "sensitivy" capacitor with a smaller one, but if it's miraculously fixed in some firmware, I want to know **how** before I modify all my 30 T1s, since it's a lot of work that **maybe** isn't even necessary.

sergiorademacher, 2020-06-16:
Looks like the issue resurfaced in later versions after 8.1, so it isn't magically fixed either.  I haven't been able to completely isolate the issue, as with some switches the frequency of ghosting varies with the code length and arduino version.  Just like you I have a lot of bad T1s and I ended up removing the buzzer in all of them.  😞

sergiorademacher, 2020-07-03:
Hi @Fusseldieb ,

Did you find a more permanent solution?  I'm still getting random ghost switching in some units from time to time.

flexy2dd, 2020-09-25:
Hi,
FYI I have the same issue on SONOFF Basic V1 :(
i have a LEDs garden spots and when they are turned off they flash (1s)
this seems to be related to the home-assistant ping (every 15s)
I will try your solutions
`
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Jardin Piscine status"

switch:
  - platform: gpio
    pin: GPIO12
    name: "Jardin Piscine relay"
    id: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes

sensor:
  - platform: wifi_signal
    name: "Jardin Piscine WiFi Signal"
    update_interval: 60s
`

closed 2021-02-06 by stale[bot]

Fusseldieb, 2021-02-08:
It is a hardware issue, unfortunately...

**Don't buy Sonoff T1!**

sergiorademacher, 2021-02-09:
> It is a hardware issue, unfortunately...
> 
> **Don't buy Sonoff T1!**

Yep, indeed it is a hardware issue. But is interesting that I switched to tasmota for my T1 ( I didn't want to though) and the problem went away. No particular configuration to mitigate the issue, just plain vanilla tasmota.  🤨

flexy2dd, 2021-02-09:
I agréé, I have buy à zigbee version and is thé same issue 😕