esphome/issues #857: OTA update failing over esphome command line

pszafer, 2019-11-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Arch / Esphome installed via pip

**ESP (ESP32/ESP8266, Board/Sonoff):**
```
platform: ESP32
  board: mhetesp32minikit
```

**Affected component:**
None.

**Description of problem:**
Hello.
I'm not sure if it is esphome bug, but I just want to report that something is wrong...
It began with version 1.14.2. (I had 1.13 before).
If I run `esphome something.yaml run` it just reset connection and fails. ESP32 is restarting with old firmware.
Upgrading via web interface is working without a problem.
I upgraded python on my pc to version 3.8 as well recently.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: quinled
  platform: ESP32
  board: mhetesp32minikit

wifi:
  ssid: !secret wifi_ap
  password: !secret wifi_pass
  domain: !secret domain

logger:
  baud_rate: 0
  # level: VERY_VERBOSE
  hardware_uart: UART1

api:

ota:

web_server:
  port: 80

light:
  - platform: fastled_clockless
    chipset: ws2812b
    pin: GPIO1
    num_leds: 78
    rgb_order: GRB
    name: "quinled_ekspres"
    effects:
      - addressable_rainbow:
      - addressable_rainbow:
          name: Rainbow Effect With Custom Values
          speed: 10
          width: 512
      - addressable_rainbow:
          name: Rainbow mega fast
          speed: 20
          width: 150
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_fireworks:
  - platform: fastled_clockless
    chipset: ws2812b
    pin: GPIO16
    num_leds: 120
    rgb_order: GRB
    name: "quinled_plyta"
    effects:
      - addressable_rainbow:
      - addressable_rainbow:
          name: Rainbow Effect With Custom Values
          speed: 10
          width: 512
      - addressable_rainbow:
          name: Rainbow mega fast
          speed: 20
          width: 150
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_fireworks:

```

**Logs (if applicable):**
Logs from ESP32
```
[12:11:15][V][ota:097]: OTA features is 0x00                                                              
DEBUG Got message: <class 'api_pb2.SubscribeLogsResponse'>:                                               
  level: VERY_VERBOSE                                                                                     
  message: "\033[0;37m[V][ota:171]: OTA size is 941584 bytes\033[0m"                                      
[12:11:15][V][ota:171]: OTA size is 941584 bytes                                                          
DEBUG Got message: <class 'api_pb2.SubscribeLogsResponse'>:                                               
  level: VERY_VERBOSE                                                                                     
  message: "\033[0;37m[V][ota:219]: Update: Binary MD5 is 208d247f39f68fa77dfc688432b3b331\033[0m"        
[12:11:15][V][ota:219]: Update: Binary MD5 is 208d247f39f68fa77dfc688432b3b331                            
DEBUG Got message: <class 'api_pb2.SubscribeLogsResponse'>:                                               
  level: VERBOSE                                                                                          
  message: "\033[0;36m[D][ota:243]: OTA in progress: 0.1%\033[0m"                                         
[12:11:15][D][ota:243]: OTA in progress: 0.1%                                                             
DEBUG Got message: <class 'api_pb2.SubscribeLogsResponse'>:                                               
  level: VERBOSE                                                                                          
  message: "\033[0;36m[D][ota:243]: OTA in progress: 8.7%\033[0m"                                         
[12:11:17][D][ota:243]: OTA in progress: 8.7%                                                             
DEBUG Got message: <class 'api_pb2.SubscribeLogsResponse'>:                                               
  level: VERBOSE                                                                                          
  message: "\033[0;36m[D][ota:243]: OTA in progress: 17.4%\033[0m"                                        
[12:11:18][D][ota:243]: OTA in progress: 17.4%                                                            
DEBUG Sending <class 'api_pb2.PingRequest'>:                                                              
                                                                                                          
DEBUG Got message: <class 'api_pb2.SubscribeLogsResponse'>:                                               
  level: VERBOSE                                                                                          
  message: "\033[0;36m[D][ota:243]: OTA in progress: 26.2%\033[0m"                                        
[12:11:19][D][ota:243]: OTA in progress: 26.2%                                                            
DEBUG Got message: <class 'api_pb2.SubscribeLogsResponse'>:                                               
  level: VERBOSE                                                                                          
  message: "\033[0;36m[D][ota:243]: OTA in progress: 34.9%\033[0m"                                        
[12:11:20][D][ota:243]: OTA in progress: 34.9%                                                            
DEBUG Got message: <class 'api_pb2.SubscribeLogsResponse'>:                                               
  level: VERBOSE                                                                                          
  message: "\033[0;36m[D][ota:243]: OTA in progress: 43.6%\033[0m"                                        
[12:11:21][D][ota:243]: OTA in progress: 43.6%                                                            
DEBUG Got message: <class 'api_pb2.SubscribeLogsResponse'>:                                               
  level: VERBOSE                                                                                          
  message: "\033[0;36m[D][ota:243]: OTA in progress: 52.2%\033[0m"                                        
[12:11:22][D][ota:243]: OTA in progress: 52.2%                                                            
DEBUG Got message: <class 'api_pb2.SubscribeLogsResponse'>:                                               
  level: VERBOSE                                                                                          
  message: "\033[0;36m[D][ota:243]: OTA in progress: 60.9%\033[0m"                                        
[12:11:23][D][ota:243]: OTA in progress: 60.9%                                                            
WARNING Disconnected from API: Timeout while waiting for message response!                                
INFO Connecting to quinled.velomelon.com:6053 (192.168.4.66)                                              
DEBUG Sending <class 'api_pb2.HelloRequest'>:                                                             
  client_info: "ESPHome v1.14.2"                                                                          
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 1 seconds
INFO Connecting to quinled.velomelon.com:6053 (192.168.4.66)                                              
DEBUG Sending <class 'api_pb2.HelloRequest'>:                                                             
  client_info: "ESPHome v1.14.2"                                                                          
DEBUG Got message: <class 'api_pb2.HelloResponse'>:                                                       
  api_version_major: 1                                                                                    
  api_version_minor: 3                                                                                    
  server_info: "quinled (esphome v1.14.2)"                                                                
DEBUG Successfully connected to quinled.velomelon.com ('quinled (esphome v1.14.2)' API=1.3)
```

Logs from command line:
```
INFO Uploading quinled/.pioenvs/quinled/firmware.bin (941584 bytes)                   
Uploading: [===============================================             ] 79%                             
                                                     
ERROR Error sending data: [Errno 104] Connection reset by peer
```
Doesn't matter how much `-vvv` I put in run command it is all error I have.

**Additional information and things you've tried:**

Should I try downgrading esphome and check if 1.13 is working without a problem?

pauln, 2019-12-31:
I've just experienced a very similar OTA issue with an ESP32, which I think may shed some more light on what happened here.  I had the device open, with wires attached for programming... so I watched the logs over serial, and got the following:
```
[D][ota:072]: Starting OTA Update from 192.168.0.161...
[D][ota:243]: OTA in progress: 0.1%
[D][ota:243]: OTA in progress: 6.9%
[D][ota:243]: OTA in progress: 13.2%
[D][ota:243]: OTA in progress: 20.0%
[D][ota:243]: OTA in progress: 26.8%
E (21743) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (21743) task_wdt:  - IDLE0 (CPU 0)
E (21743) task_wdt: Tasks currently running:
E (21743) task_wdt: CPU 0: loopTask
E (21743) task_wdt: Aborting.
abort() was called at PC 0x400e4f8b on core 0
```

So it looks as though the WDT isn't being fed during the OTA upload process - meaning that if your update takes too long to upload, a WDT reset will happen in the middle of it, causing the update to fail.

pauln, 2019-12-31:
@pszafer Out of interest, is the ESP32 you're using a "normal" dual-core one, or a single-core variant such as is found in some Xiaomi/Mijia smart home devices?  I experienced it (see previous comment) on a single-core one in a doorbell.

I'd be surprised if this is an issue on dual-core ESP32s, as surely it'd be a much more widespread issue if so.  If you're also using a single-core ESP32, it could be that normally the WDT would be fed by the core which isn't handling the OTA upload...

pszafer, 2019-12-31:
As my problem haven't had any attention in that time I didn't update it.
But I have problems on 1.14.x with both ESP8266 and ESP32. Wemos D1 Mini and Wemos D32 Mini.
Problem occurs when ESP (either 8266 or 32) is powered up by board Quinled-Dig-Uno thorugh VCC pin (12V -> DC-DC converter to 5V).

Anyway OTA success if I try several times, for ESP8266 2 times is enough. For ESP32 it more like 6 times, then 6th is successful.

I powered up Wemos D32 Mini through USB from PC, but for 2 days I couldn't reproduce the issue...

closed 2020-03-12 by brandond