esphome/issues #858: RMT method causing device to become unresponsive.

icefire555, 2019-11-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker/Unraid
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 NODEMCU
**Affected component:**

https://esphome.io/components/light/neopixelbus.html

**Description of problem:**
I am unable to use the ESP32_RMT_0 method. Earlier I posted https://github.com/esphome/issues/issues/839 but it appears the issue is still around. 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: gameroomlights
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "******"
  password: "******"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Gameroomlights Fallback Hotspot"
    password: "DLKrbA0xbyUT"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

light:
 - platform: neopixelbus
   type: BRG
   name: "GameRoom Lights"
   id: ledstrip
   num_leds: 295
   pin: GPIO2
   method: ESP32_RMT_7
   effects:
    - random:
    - strobe:
    - flicker:
    - addressable_rainbow:
    - addressable_color_wipe:
    - addressable_scan:
    - addressable_twinkle:
    - addressable_random_twinkle:
    - addressable_fireworks:
    - addressable_flicker:
```

**Logs (if applicable):**

```
INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from gameroomlights.local using esphome API
WARNING Error resolving IP address of gameroomlights.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -2] Name or service not known). Re-Trying in 1 seconds
INFO Connecting to gameroomlights.local:6053 (192.168.1.237)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.237: [Errno 111] Connection refused). Re-Trying in 1 seconds
INFO Connecting to gameroomlights.local:6053 (192.168.1.237)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.237: [Errno 111] Connection refused). Re-Trying in 2 seconds
INFO Connecting to gameroomlights.local:6053 (192.168.1.237)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.237: [Errno 111] Connection refused). Re-Trying in 3 seconds
INFO Connecting to gameroomlights.local:6053 (192.168.1.237)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.237: [Errno 111] Connection refused). Re-Trying in 5 seconds
INFO Connecting to gameroomlights.local:6053 (192.168.1.237)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.237: [Errno 111] Connection refused). Re-Trying in 7 seconds
INFO Connecting to gameroomlights.local:6053 (192.168.1.237)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.237: [Errno 111] Connection refused). Re-Trying in 11 seconds
INFO Connecting to gameroomlights.local:6053 (192.168.1.237)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.237: [Errno 111] Connection refused). Re-Trying in 17 seconds
INFO Connecting to gameroomlights.local:6053 (192.168.1.237)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.237: [Errno 111] Connection refused). Re-Trying in 25 seconds
INFO Connecting to gameroomlights.local:6053 (192.168.1.237)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.237: [Errno 111] Connection refused). Re-Trying in 30 seconds
INFO Connecting to gameroomlights.local:6053 (192.168.1.237)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.237: [Errno 111] Connection refused). Re-Trying in 30 seconds
INFO Connecting to gameroomlights.local:6053 (192.168.1.237)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.237: [Errno 111] Connection refused). Re-Trying in 30 seconds
INFO Connecting to gameroomlights.local:6053 (192.168.1.237)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.237: [Errno 111] Connection refused). Re-Trying in 30 seconds
INFO Connecting to gameroomlights.local:6053 (192.168.1.237)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.237: [Errno 111] Connection refused). Re-Trying in 30 seconds
INFO Connecting to gameroomlights.local:6053 (192.168.1.237)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.237: [Errno 111] Connection refused). Re-Trying in 30 seconds
INFO Connecting to gameroomlights.local:6053 (192.168.1.237)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.237: [Errno 111] Connection refused). Re-Trying in 30 seconds
WARNING Error resolving IP address of gameroomlights.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -2] Name or service not known). Re-Trying in 30 seconds
INFO Connecting to gameroomlights.local:6053 (192.168.1.237)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.237: [Errno 111] Connection refused). Re-Trying in 30 seconds
```

**Additional information and things you've tried:**

The ESP32 locks up. However, if I remove the method, or try bitbang/I2S it uploads without an issue and becomes responsive again. 
There are no useful logs when I click the log tab.

icefire555, 2019-11-19:
I also tried ESP32_RMT_0, ESP32_RMT_1, and 'ESP32_RMT_0'

brandond, 2019-11-19:
You're probably going to need to collect logs via USB.

icefire555, 2019-11-20:
Cool! I went ahead and swapped from pin 2 to pin 4. Just to see if it helped. And it didn't.
This was the entire log Via the com ports on one reboot.

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
[0;32m[I][logger:156]: Log initialized[0m
[0;35m[C][ota:364]: There have been 0 suspected unsuccessful boot attempts.[0m
[0;32m[I][app:028]: Running through setup()...[0m
[0;36m[D][esp-idf:000]: E (40) rmt: rmt_driver_install(725): RMT driver already installed for channel[0m
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4008da14
file: ".piolibdeps/gameroomlights/NeoPixelBus-esphome_ID6592/src/internal/NeoEsp32RmtMethod.h" line 292
func: void NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::Initialize() [with T_SPEED = NeoEsp32RmtSpeed800Kbps; T_CHANNEL = NeoEsp32RmtChannel0]
expression: rmt_driver_install(T_CHANNEL::RmtChannelNumber, 0, 0)

Backtrace: 0x4008d518:0x3ffb1af0 0x4008da17:0x3ffb1b10 0x400e0372:0x3ffb1b30 0x400e03e1:0x3ffb1b80 0x400e03f6:0x3ffb1bc0 0x40162a91:0x3ffb1be0 0x400d7131:0x3ffb1c00 0x40162eed:0x3ffb1c20 0x400ddd85:0x3ffb1c40 0x400e10ce:0x3ffb1c90 0x400eb87b:0x3ffb1fb0 0x40089c61:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
[0;32m[I][logger:156]: Log initialized[0m
[0;35m[C][ota:364]: There have been 1 suspected unsuccessful boot attempts.[0m
[0;32m[I][app:028]: Running through setup()...[0m
[0;36m[D][esp-idf:000]: E (40) rmt: rmt_driver_install(725): RMT driver already installed for channel[0m
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4008da14
file: ".piolibdeps/gameroomlights/NeoPixelBus-esphome_ID6592/src/internal/NeoEsp32RmtMethod.h" line 292
func: void NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::Initialize() [with T_SPEED = NeoEsp32RmtSpeed800Kbps; T_CHANNEL = NeoEsp32RmtChannel0]
expression: rmt_driver_install(T_CHANNEL::RmtChannelNumber, 0, 0)

Backtrace: 0x4008d518:0x3ffb1af0 0x4008da17:0x3ffb1b10 0x400e0372:0x3ffb1b30 0x400e03e1:0x3ffb1b80 0x400e03f6:0x3ffb1bc0 0x40162a91:0x3ffb1be0 0x400d7131:0x3ffb1c00 0x40162eed:0x3ffb1c20 0x400ddd85:0x3ffb1c40 0x400e10ce:0x3ffb1c90 0x400eb87b:0x3ffb1fb0 0x40089c61:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
[0;32m[I][logger:156]: Log initialized[0m
[0;35m[C][ota:364]: There have been 2 suspected unsuccessful boot attempts.[0m
[0;32m[I][app:028]: Running through setup()...[0m
[0;36m[D][esp-idf:000]: E (40) rmt: rmt_driver_install(725): RMT driver already installed for channel[0m
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4008da14
file: ".piolibdeps/gameroomlights/NeoPixelBus-esphome_ID6592/src/internal/NeoEsp32RmtMethod.h" line 292
func: void NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::Initialize() [with T_SPEED = NeoEsp32RmtSpeed800Kbps; T_CHANNEL = NeoEsp32RmtChannel0]
expression: rmt_driver_install(T_CHANNEL::RmtChannelNumber, 0, 0)

Backtrace: 0x4008d518:0x3ffb1af0 0x4008da17:0x3ffb1b10 0x400e0372:0x3ffb1b30 0x400e03e1:0x3ffb1b80 0x400e03f6:0x3ffb1bc0 0x40162a91:0x3ffb1be0 0x400d7131:0x3ffb1c00 0x40162eed:0x3ffb1c20 0x400ddd85:0x3ffb1c40 0x400e10ce:0x3ffb1c90 0x400eb87b:0x3ffb1fb0 0x40089c61:0x3ffb1fd0
[... 217 lines cut ...]
[0;35m[C][ota:030]:   Address: gameroomlights.local:3232[0m
[0;33m[W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts[0m
[0;32m[I][app:058]: setup() finished successfully![0m
[0;32m[I][ota:379]: Waiting for OTA attempt.[0m
[0;32m[I][app:100]: ESPHome version 1.14.3 compiled on Nov 20 2019, 04:47:16[0m
[0;35m[C][wifi:415]: WiFi:[0m
[0;35m[C][wifi:283]:   SSID: [5m'Home Wifi 2.4Ghz'[6m[0m
[0;35m[C][wifi:284]:   IP Address: 192.168.1.237[0m
[0;35m[C][wifi:286]:   BSSID: [5m78:8A:20:5D:26:3B[6m[0m
[0;35m[C][wifi:287]:   Hostname: 'gameroomlights'[0m
[0;35m[C][wifi:291]:   Signal strength: -43 dB [0;32m▂▄▆█[0m[0m
[0;35m[C][wifi:295]:   Channel: 11[0m
[0;35m[C][wifi:296]:   Subnet: 255.255.255.0[0m
[0;35m[C][wifi:297]:   Gateway: 192.168.1.1[0m
[0;35m[C][wifi:298]:   DNS1: 8.8.4.4[0m
[0;35m[C][wifi:299]:   DNS2: 209.222.18.222[0m
[0;35m[C][logger:175]: Logger:[0m
[0;35m[C][logger:176]:   Level: DEBUG[0m
[0;35m[C][logger:177]:   Log Baud Rate: 115200[0m
[0;35m[C][logger:178]:   Hardware UART: UART0[0m
[0;35m[C][captive_portal:169]: Captive Portal:[0m
[0;35m[C][ota:029]: Over-The-Air Updates:[0m
[0;35m[C][ota:030]:   Address: gameroomlights.local:3232[0m
[0;33m[W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts[0m
`

icefire555, 2019-11-20:
`�[0;36m[D][esp-idf:000]: E (40) rmt: rmt_driver_install(725): RMT driver already installed for channel�[0m
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4008da14
file: ".piolibdeps/gameroomlights/NeoPixelBus-esphome_ID6592/src/internal/NeoEsp32RmtMethod.h" line 292
func: void NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::Initialize() [with T_SPEED = NeoEsp32RmtSpeed800Kbps; T_CHANNEL = NeoEsp32RmtChannel0]
expression: rmt_driver_install(T_CHANNEL::RmtChannelNumber, 0, 0)

Backtrace: 0x4008d518:0x3ffb1af0 0x4008da17:0x3ffb1b10 0x400e0372:0x3ffb1b30 0x400e03e1:0x3ffb1b80 0x400e03f6:0x3ffb1bc0 0x40162a91:0x3ffb1be0 0x400d7131:0x3ffb1c00 0x40162eed:0x3ffb1c20 0x400ddd85:0x3ffb1c40 0x400e10ce:0x3ffb1c90 0x400eb87b:0x3ffb1fb0 0x40089c61:0x3ffb1fd0

Rebooting...`

OttoWinter, 2019-11-20:
That looks like a bug in neopixelbus. Offending line is here for reference: https://github.com/Makuna/NeoPixelBus/blob/b6a6c4162810fbe143d91e178f564d66e463af51/src/internal/NeoEsp32RmtMethod.h#L292

@icefire555 Just checking, is that the complete config you posted? If you have multiple neopixel lights with the same RMT channel I would expect that error. Also check if using different RMT channels helps. 

Also is that the config you got the stacktrace with? Because the RMT channel from the stacktrace is 0 whereas you put 7 in the config.

icefire555, 2019-11-20:
To clarify. The method works on the library without esphome. I'll post my code to that when I get home.
And changing the RMT number didn't help. I only am running 1 strip of lights. But it's a long one. And the yaml I posted is the same code that threw the error.

icefire555, 2019-11-21:
Ignore the horribly commented and organized code. I only wrote it so I could make my lights into a rainbow while I figured out how to put the ESP32 onto home assistant.

icefire555, 2019-11-21:
`#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>
#include <Update.h>

const char* host = "GameRoom Lights";
const char* ssid = "Home Wifi 2.4Ghz";
const char* password = "******";

WebServer server(80);

/*
   Login page
*/

const char* loginIndex =
  "<form name='loginForm'>"
  "<table width='20%' bgcolor='A09F9F' align='center'>"
  "<tr>"
  "<td colspan=2>"
  "<center><font size=4><b>ESP32 Login Page</b></font></center>"
  "<br>"
  "</td>"
  "<br>"
  "<br>"
  "</tr>"
  "<td>Username:</td>"
  "<td><input type='text' size=25 name='userid'><br></td>"
  "</tr>"
  "<br>"
  "<br>"
  "<tr>"
  "<td>Password:</td>"
  "<td><input type='Password' size=25 name='pwd'><br></td>"
  "<br>"
  "<br>"
  "</tr>"
  "<tr>"
  "<td><input type='submit' onclick='check(this.form)' value='Login'></td>"
  "</tr>"
  "</table>"
  "</form>"
  "<script>"
  "function check(form)"
  "{"
  "if(form.userid.value=='' && form.pwd.value=='')"
  "{"
  "window.open('/serverIndex')"
  "}"
  "else"
  "{"
  " alert('Error Password or Username')/*displays error message*/"
  "}"
  "}"
  "</script>";

/*
   Server Index Page
*/

const char* serverIndex =
  "<script src='https://ajax.googleapis.com/ajax/libs/jquery/3.2.1/jquery.min.js'></script>"
  "<form method='POST' action='#' enctype='multipart/form-data' id='upload_form'>"
  "<input type='file' name='update'>"
  "<input type='submit' value='Update'>"
  "</form>"
  "<div id='prg'>progress: 0%</div>"
  "<script>"
  "$('form').submit(function(e){"
[... 190 lines cut ...]
{
  // wait for this animation to complete,
  // we are using it as a timer of sorts
  if (param.state == AnimationState_Completed)
  {
    // done, time to restart this position tracking animation/timer
    animations.RestartAnimation(param.index);

    // rotate the complete strip one pixel to the right on every update
    strip.RotateRight(1);
  }
}

void DrawTailPixels()
{
  // using Hsl as it makes it easy to pick from similiar saturated colors
  float hue = random(360) / 360.0f;
  for (uint16_t index = 0; index < strip.PixelCount() && index <= TailLength; index++)
  {
    float lightness = index * MaxLightness / TailLength;
    RgbColor color = HslColor(hue, 1.0f, lightness);

    strip.SetPixelColor(index, colorGamma.Correct(color));
  }
}`

icefire555, 2019-11-21:
But with the code above. I was able to use neo pixel bus to turn each lights into a rainbow. However It was not set up to animate. Only do it once. Because I have signal strength issues with the wiring and 3.3v logic that causes flickering.

icefire555, 2019-11-21:
Lol never mind. That was an older one. I have newer code that makes a rainbow. 
pretty much just this code.

`void Rainbow()
{
  int i = 0;
  int ColorR = 255;
  int ColorG = 0;
  int ColorB = 0;
  while (i < PixelCount)
  {
    //strip.SetPixelColor(i, RgbColor(255, 0, 255));
    //RgbColor red(colorSaturation, 0, 0);

    if (ColorR == 255 && ColorB == 0) //Red -> Yellow
    {
      strip.SetPixelColor(i, RgbColor(ColorR, ColorG, ColorB));
      ColorG += ColorChange;
    }
    if (ColorG == 255 && ColorB == 0) //Yellow -> Green
    {
      strip.SetPixelColor(i, RgbColor(ColorR, ColorG, ColorB));
      ColorR -= ColorChange;
    }
    if (ColorG == 255 && ColorR == 0) //Green -> Teal
    {
      strip.SetPixelColor(i, RgbColor(ColorR, ColorG, ColorB));
      ColorB += ColorChange;
    }
    if (ColorB == 255 && ColorR == 0) //Teal -> Blue
    {
      strip.SetPixelColor(i, RgbColor(ColorR, ColorG, ColorB));
      ColorG -= ColorChange;
    }
    if (ColorB == 255 && ColorG == 0) //Blue -> Purple
    {
      strip.SetPixelColor(i, RgbColor(ColorR, ColorG, ColorB));
      ColorR += ColorChange;
    }
    if (ColorR == 255 && ColorG == 0) //Purple -> red
    {
      strip.SetPixelColor(i, RgbColor(ColorR, ColorG, ColorB));
      ColorB -= ColorChange;
    }
    i++;
  }
  isUpdateDone = 1;
}`

legomind, 2020-01-21:
Any update on this? I am having the same issue.

gibranZawahra, 2020-01-26:
yup same here all rmt channels kill the esp32, i can still ota but it doesn't connect and doesn't display anything on the screen
thanks !

icefire555, 2020-01-26:
I'm glad I'm not the only one. This issue literally stops me from using esp home. When I get some free time I'll probably look for alternatives.

gibranZawahra, 2020-01-26:
i think it's everyone, i think it's just not working at all on rmt, why are you using rmt instead of i2s?

icefire555, 2020-01-26:
Doesn't i2c need a clock pin?
Also the ESP32 has rmt on it for this purpose, and they work in the original library, just not in ESPhome.

gibranZawahra, 2020-01-26:
no I2S in method: ESP32_I2S_0 OR method: ESP32_I2S_1 (used by def)

icefire555, 2020-01-26:
Those methods seem to flicker a lot for me.

icefire555, 2020-01-26:
But I'm also driving 300 "LEDs" 3 LEDs per led controller.

gibranZawahra, 2020-01-26:
is the flicker maybe coming from somewhere else ? like i2c screen or the power supply ? do you have a resistor on the data line of the leds ? mine is not flickering at all for me with i2s

icefire555, 2020-01-26:
I have nothing wired up to the ESP32 besides the LEDs, what value do you recommend for the resister? Last time I couldn't make it work without using a very low value. The powersupply is it's own dedicated 12v PSU.

gibranZawahra, 2020-01-27:
im using a 330 ohm for my neoPixels what leds are you using that run off 12v ?

icefire555, 2020-01-27:
I'm currently using these LEDs
https://www.amazon.com/gp/product/B01CNL6MNM/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

brandond, 2020-01-27:
I'm driving 300 of the same as @icefire555 using clockless with no flickering. Im not trying to run anything else at the same time though.

poelzi, 2020-02-16:
I have different strips on different controllers. I sell flickering with or without resistor on:
esp32-poe (pin does not matter, nor ESP32_I2S_0/ESP32_I2S_1, The RMT bug applies as well).
Same strip works flawless with ESP8266 without resistor.
I have another strip on a esp32-poe which very seledome flickers, like one pixel very 5 minutes. This one is connected by ethernet.

fastled seem not to work at all

I'm using SK6812 strips. https://www.aliexpress.com/item/32476317187.html?spm=2114.12010612.8148356.2.128f6031HPOFL4

gibranZawahra, 2020-02-16:
I managed to fix my flickering problem both on neopixel and fast led by increasing i2c speed because I noticed it only flickers when drawing the display, the resistor is there to protect the leds long term wise you might never have a problem without but I always had the first led start misbehaving and then dying without it on long term use. I'd love for the emt to start working since I need i2s for connecting a mic...

poelzi, 2020-02-17:
@gibranZawahra how did you set the frequency on the bus ?

icefire555, 2020-02-17:
Using a resistor between 280-100ohm leads to no signal coming out.

gibranZawahra, 2020-02-17:
Using rmt or the other methods? The resistor needs to be closest to the leds.. And make sure it's the correct pin and that you're not assigning Teo different strips to the same i2s bus

poelzi, 2020-02-17:
The resistor is only there for limiting the inrush current on level change, should not make any difference where it is placed.
Since rmt is not working, I can only use the i2c method. When I try to change the i2c frequency, it flickers even worse, at least all frequencies I tried.

poelzi, 2020-02-17:
I just discovered that the flickering starts exactly as soon as the wifi starts to be connected, which would explain why I see the flickering only seldom on the other controller which only uses the ethernet port. I think the I2C method is affected by some interrupt issue.

poelzi, 2020-03-02:
I tested this simple program and it works like a charm.

poelzi, 2020-03-10:
I spend some more time digging into this problem. unfortunately, I don't have a proper debugger for the esp32, so progress is slow.
The RMT fails because rmt_driver_install checks if the channel is already in use through a bitmask on [s_rmt_driver_channels](https://github.com/espressif/esp-idf/blob/63cd2b06137ddfab9a3df0a9905698b526f4e87d/components/driver/rmt.c#L55).
For some reason, this is not 0 after starting the esphome firmware, while using a the providet .ino from above works. I have not found any rmt usage in a minimal firmware, so my current guess is, that for some reason, the static counter variable is not initialized with 0 like a normal firmware.

Any ideas ? @OttoWinter @brandond

poelzi, 2020-03-10:
It could also be an alginment problem of the variable. I had problems as well, where a uint8_t caused strange behavior while a uint32_t made the code work as expected. Is there some compile or linker flag involved for alignment ?

stefanuc111, 2020-03-24:
I've the same problem and I'm trying to run 750 leds, with i2s doesn't work it update only the first ~80 leds and the other update rarerly, with RMT I get this same error discussed here, there are any news?

poelzi, 2020-03-26:
I think somebody with a hardware debugger could more likely debug the problem, or more experience with esp platform.
When I developed the gradient patch, the cpu ran into an exception when I used a uint8_t but with uint32_t it worked fine. The problematic counter is also uint8_t, so I think it might be alignment problems when accessing such variables, but thats just an guess.

legomind, 2020-11-21:
Go away stale be!

Makuna, 2021-02-09:
Just adding a pointer here for a work around; but I don't see the above mentioned problem in a normal build so it is something messing with that const channel number within the context of this project.

With the latest build, you can now dynamically select the channel for RMT along with the compile time it already supports.
```
NeoPixelBus<NeoGrbFeature, NeoEsp32RmtNWs2812xMethod>  MyNeoPixelBus(PixelCount, PixelPin, NeoBusChannel_0);
```
You might try switching to the new RmtN method.

angelnu, 2021-04-05:
I think I found the issue in esphome `neopixelbus_light.h` -> Begin is called twice:
- in add_leds
- in setup

https://github.com/esphome/esphome/blob/dev/esphome/components/neopixelbus/neopixelbus_light.h#L71-L81

I manually removed the begin call in setup and the esp32 booted up.

@OttoWinter @Makuna - where is correct to call Begin? At setup or at add_leds? Or should the library be more tolerant with already initialized RMT HW?

Makuna, 2021-04-05:
Begin should only ever be called once and anytime during or after Setup() is called.   Note that RMT can be used by other systems than just NeoPixelBus; so it is left to the sketch author to manage the channels.

NOTE:  I2s method is working again; there is an active "feature" where it will throw an exception on a bus of only one pixel as the i2s DMA buffer doesn't like so little data (it also starts sending faster!).  I just merged in a fix for the assert so now if you tell it one pixel it just sets it to two pixels anyhow.

angelnu, 2021-04-05:
Thanks @Makuna for fixing i2s - I saw people complaining about it not working - I believe I still have to go with RMT to be able to use multiple lights with one ESP32 since only one i2s is available. Would you recommend using i2s for the first light and RMT for the others or just RMT for all?

I did a PR to remove the duplicated begin call.

Makuna, 2021-04-05:
There are actually TWO I2s on the ESP32.  If you are using large strips and short on memory, the RMT methods have smaller  buffers; but it does rely on a very short ISR to translate and fill the RMT DMA buffers (which are rather small and shared across all the RMT channels); so if you have a poorly written ISR that does too much work in it; it may effect the RMT methods.  Normally this will never be a problem.  Note that most of the WiFi will not cause a problem unless you cause some RTOS blocking issues.

angelnu, 2021-04-05:
I tried with 3 lights and 300 leds in total (this is my expected setup) and I am not seeing any errors and Wifi is stable. The esphome is only going to control these - no other sensors. I do not have the lights yet so I cannot tell until I get them in a few days.

Hopefully PR gets into an esphome fix pack soon but I would build a docker container with the fix until then (I run my esphome dev env in Kubernetes)

closed 2021-04-06 by OttoWinter