esphome/issues #860: Template cover - position_action and position_lambda missing

KristopherMackowiak, 2019-11-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HASS.IO

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Affected component:**

Template cover

**Description of problem:**
The parameters for position_action and position_lambda are missing (the same as in tilt_action and tilt_lambda) to easily control position from the home assistant.

**Problem-relevant YAML-configuration entries:**
```yaml
not applicable

```

**Logs (if applicable):**

```
not applicable

```

**Additional information and things you've tried:**

KristopherMackowiak, 2019-11-19:
I think you could call open_action when changing position as below
https://github.com/esphome/esphome/pull/866/commits/60af98ea47f87479ea9da30a9075ae44ee99159b

OttoWinter, 2019-11-20:
This is a feature request, not an issue - but glad you're already working on a PR! If you need help creating the patch, please ask here.

closed 2019-11-20 by OttoWinter

KristopherMackowiak, 2019-11-20:
#875
I added and tested on my blinds but PR return failed. Can you check? https://github.com/esphome/esphome/pull/875

KristopherMackowiak, 2019-11-21:
I tried again commit and PK #877. Code is ok but Travis CI still return error.