esphome/issues #861: No Logs with esphome CLI

ellavas, 2019-11-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Mac OS 10.14.6
Python 3.7
Esphome 1.14.3

**ESP (ESP32/ESP8266, Board/Sonoff):**

SonOff Pow R2
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash

**Affected component:**

https://esphome.io/components/logger.html?highlight=logger

**Description of problem:**
When deploying and running a very simple project, there is no log ouput, even when the debug level is set to debug.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: pow
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: ''
  password: ''

debug:

# Enable logging
logger:
  level: debug

# Enable Home Assistant API
api:

ota:

switch:
  - platform: gpio
    pin: GPIO12
    name: "Living Room Torsher"
    id: relay
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
INFO Successfully compiled program.
Found multiple options, please choose one:
  [1] /dev/cu.SLAB_USBtoUART (CP2102 USB to UART Bridge Controller)
  [2] Over The Air (pow.local)
(number): 1
INFO Running:  esptool.py --before default_reset --after hard_reset --baud 460800 --chip esp8266 --port /dev/cu.SLAB_USBtoUART write_flash 0x0 pow/.pioenvs/pow/firmware.bin
esptool.py v2.7
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 80:7d:3a:33:50:93
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0340
Compressed 373024 bytes to 258077...
Wrote 373024 bytes (258077 compressed) at 0x00000000 in 6.4 seconds (effective 465.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/cu.SLAB_USBtoUART with baud rate 115200
```

**Additional information and things you've tried:**

brandond, 2019-11-19:
Is ESPHome actually running? The upload tool will try to reset the device to start running your code, but this is not guaranteed to work. If you've connected a USB TTL adapter to your device, the RTS pin is probably not actually connected to the reset pin.

Disconnect your USB adapter and power on the device. Does it come up on the network?

ellavas, 2019-11-19:
Yes I'm using a usb ttl adapter.
I've tried also to manually reset the device and then to read logs without success.

OttoWinter, 2019-11-20:
@ellavas 
1. You should see *something* when you go manually reset the device. Otherwise the UART connection can't be working correctly.
2. Make sure you don't have GPIO0 attached - putting it to a certain logic level (can't remember which) puts it in flash mode, in which case no code runs.

ellavas, 2019-11-21:
1. Nothing happens after rebooting the device. :( I also tried to change the baud rate. In that case, an error occurs and the connection is closed.

2. GPIO0 is not attached as you can see in my yaml config. I have a very limited setup.

brandond, 2019-11-21:
Otto means GPIO0 physically connected. Connecting GPIO0 to ground is how you put it into flash mode. Can you provide a picture of how you connected your USB adapter to the Sonoff, and how you put the Sonoff into flash mode?

ellavas, 2019-11-22:
Connection nn the SonOff side:
[image]

Connection on the usb TTL side:
[image]

I put the Sonoff into flash mode by pressing the button, then plugging the usb cable, and finally I released the button. The upload works well.

OttoWinter, 2019-11-29:
@ellavas Did you reconnect the module after having uploaded it? When the flasher exits, the device is still in flash mode and doesn't run any user code. Please reconnect the device (this time without holding the button down)

closed 2019-12-13 by no-response[bot]