esphome/issues #862: PZEM 004T V1.0 wiring problem

MariusLaurinaitis, 2019-11-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

sendorm, 2019-11-19:
Can you try with stop_bits: 1?

MariusLaurinaitis, 2019-11-20:
And what about 5 or 3 V? Do I need resistor?

sendorm, 2019-11-20:
You should use 5V's for the PZEM's. You can use the VU pin on your 8266. 
I am using the PZEM's without a 1K resistor. And they work. 
For safety, try with a resistor first. If it works, fine. If not try without the resistor.

MariusLaurinaitis, 2019-11-20:
Thanks, I will try.

MariusLaurinaitis, 2019-11-21:
Thank, after ESPhome update to 14.3 it's started to work.

uart:
  id: uartpz
  tx_pin: D6
  rx_pin: D5
  baud_rate: 9600
  stop_bits: 1

sensor:
  - platform: pzem004t
    uart_id: uartpz
    current:
      name: "PZEM-004T-1 Current"
    voltage:
      name: "PZEM-004T-1 Voltage"
    power:
      name: "PZEM-004T-1 Power"
    update_interval: 5s

closed 2019-11-29 by OttoWinter