esphome/issues #863: esphome.io is blocked in Russia

Anonym-tsk, 2019-11-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:** [web_server](https://esphome.io/components/web_server.html)

**Description of problem:**
In Russia, we have very strange laws.
The government can block any website.
And people who execute with these laws are not very smart.

So, esphome.io is not available in Russia.
The IP addresses 104.28.17.115 and 104.28.16.115 are blocked because another site with illegal information was located on them.

Because of this blocking, we cannot access the esphome.io and the web_server component does not work, because scripts and styles are loaded on the same domain.

@OttoWinter, I think it’s not difficult to change the IP addresses on the hosting provider. Please do it. Users from Russia will be very grateful.

**Problem-relevant YAML-configuration entries:**
```yaml
```

**Logs (if applicable):**

```
```

**Additional information and things you've tried:**

OttoWinter, 2019-11-20:
Huh that's no good.

Has this been the case for a longer time? I don't see any sudden drop-off in analytics from russian users.

[This page](https://isitblockedinrussia.com/?host=esphome.io) shows that it is indeed blocked (sometimes, other times it says it's not blocked, don't know why).

ESPHome is hosted on netlify.com - their instant build service is very good and the reason we use it.

In front of the netlify CDN there's cloudflare (to which the IP address you mentioned belongs). The main reason is that Cloudflare allows to set good caching values - previously (without cloudflare) the main page loaded much slower because each asset had to be re-validated on each load. Another reason is that netlify's pricing is just not competitive. I'd like to support them, but spending 40-60$ per month (20$ per 100GB bandwidth) on something cloudflare can do essentially for free is a no-brainer).

I'll contact cloudflare to see if they can move the domain to another IP.

As a workaround, you should be able to use https://esphome.netlify.com - that goes straight to netlify without cloudflare in-between (but with the mentioned downsides).

OttoWinter, 2019-11-20:
Also, it's strange to me a country would do IP-based blocking - I mean today with CDNs like cloudflare IP-based blocking affects a lot of innocent people as well.

I always thought these whole-country more sophisticated firewalls would use SNI-based blocking (and the less sophisticated ones just block via DNS).

Anonym-tsk, 2019-11-20:
> Also, it's strange to me a country would do IP-based blocking - I mean today with CDNs like cloudflare IP-based blocking affects a lot of innocent people as well.

Yes. And we have same problems with a lot of innocent websites.
Our government tried to block Telegram but unsuccessful. They blocked IPs, networks (/16 subnets!), data-centers.
We used to use proxy and vpn. But we fight stupid laws as best we can.

Anonym-tsk, 2019-11-20:
esphome.io is already blocked for about a month.

OttoWinter, 2019-11-20:
Does https://esphome.netlify.com work?

Anonym-tsk, 2019-11-20:
> Does https://esphome.netlify.com work?

Yes, it works.

Tuckie, 2020-05-26:
Maybe a host your own option for assets could work?

closed 2020-12-05 by stale[bot]