esphome/issues #865: Problem with reverting to 1.13.6 from 1.14.3 on ESP32

fribse, 2019-11-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**
I have serious stability problems with 1.14.3 on my ESP8266's, they work for a day or so, and then they go offline. So I had to roll back to 1.13.6.
After doing that, my ESP32-cam (AI Thinker) from Geekcreit fails to boot properly.
Looking in the console I see these messages:
```
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40160f74: 0300df21 d0091064 d04cd03b
```

**Problem-relevant YAML-configuration entries:**
```
substitutions:
  # Modify variables based on your settings
  hostname: 'camera_kitchen'

esphome:
  name: $hostname
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "SSID"
  password: "PW"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Camera Kitchen Fallback Hotspot"
    password: "PW"

#captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "PW"

# ESP32-CAM
esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32

  name: kitchen
  # ... Frame Settings

# Flashlight
output:
  - platform: gpio
    pin: GPIO4
    id: gpio_4
light:
  - platform: binary
    output: gpio_4
    name: $hostname light

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

```

OttoWinter, 2019-11-29:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40160f74: 0300df21 d0091064 d04cd03b
```

That looks like the error is in the bootloader (before ESPHome code runs). Please post more information about the device. Please post what ESPHome-Flasher says about the device while it's uploading.

fribse, 2019-11-30:
Ok, this is the upload, it's an ESP32CAM module by Geekcreit:

```
INFO Reading configuration...
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 03/29/20 02:00:00 to 10/25/20 03:00:00
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/camera_hallway
Processing camera_hallway (platform: espressif32@1.6.0; framework: arduino; board: esp32dev)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP> 1.0.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Retrieving maximum program size /data/camera_hallway/.pioenvs/camera_hallway/firmware.elf
Checking size /data/camera_hallway/.pioenvs/camera_hallway/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=         ]  12.5% (used 41020 bytes from 327680 bytes)
PROGRAM: [======    ]  57.8% (used 946470 bytes from 1638400 bytes)
========================= [SUCCESS] Took 6.72 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 10.11.12.74
INFO Uploading /data/camera_hallway/.pioenvs/camera_hallway/firmware.bin (946592 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 10.11.12.74 using esphome API
INFO Connecting to 10.11.12.74:6053 (10.11.12.74)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 10.11.12.74: timed out). Re-Trying in 1 seconds
INFO Connecting to 10.11.12.74:6053 (10.11.12.74)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 10.11.12.74: timed out). Re-Trying in 2 seconds
INFO Connecting to 10.11.12.74:6053 (10.11.12.74)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 10.11.12.74: timed out). Re-Trying in 4 seconds
INFO Connecting to 10.11.12.74:6053 (10.11.12.74)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 10.11.12.74: [Errno 113] No route to host). Re-Trying in 8 seconds
INFO Connecting to 10.11.12.74:6053 (10.11.12.74)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 10.11.12.74: [Errno 113] No route to host). Re-Trying in 16 seconds
```

OttoWinter, 2019-11-30:
Those are not logs from ESPHome-Flasher - those are logs from an OTA upload.

Also, you say this is when reverting to 1.13.6 from 1.14.3. Does this only happen OTA or over USB as well?

If it's only OTA, that's expected. The bootloaders espressif publishes are not backwards-compatible and the esp-idf core has been upgraded in 1.14

fribse, 2019-11-30:
Ok, this is the output from esphome flasher for windows compiled with 1.14.3 :

```
Using 'COM6' as serial port.
Connecting....
Detecting chip type... ESP32
Connecting...

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 4C:11:AE:75:AF:2C
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
...
Using 'COM6' as serial port.
Writing at 0x0008c000... (100 %)Wrote 970688 bytes (524168 compressed) at 0x00010000 in 12.8 seconds (effective 608.3 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!
```

This is esphome flasher with binary compiled with 1.13.6:

```
Using 'COM6' as serial port.
Connecting......
Detecting chip type... ESP32
Connecting...

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 4C:11:AE:75:AF:2C
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
...
Using 'COM6' as serial port.
Writing at 0x0008c000... (100 %)Wrote 946592 bytes (509530 compressed) at 0x00010000 in 12.4 seconds (effective 608.9 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
```

I hope this helps in some way?

fribse, 2019-11-30:
And it's the same if I use ESPHome flasher to go from 1.14.3 to 1.13.6, the logs shows:
```
Using 'COM6' as serial port.
Showing logs:
[12:41:52]ets Jun  8 2016 00:22:57
[12:41:52]
[12:41:52]rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[12:41:52]configsip: 0, SPIWP:0xee
[12:41:52]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:41:52]mode:DIO, clock div:2
[12:41:52]load:0x3fff0018,len:4
[12:41:52]load:0x3fff001c,len:1044
[12:41:52]load:0x40078000,len:8896
[12:41:52]load:0x40080400,len:5828
[12:41:52]entry 0x400806ac
[12:41:53]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[12:41:53]Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[12:41:53]Memory dump at 0x40162e50: 030000df d0091064 d04cd03b
[12:41:53]Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[12:41:53]Memory dump at 0x40162e50: 030000df d0091064 d04cd03b
[12:41:53]Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[12:41:53]Memory dump at 0x40162e50: 030000df d0091064 d04cd03b
[12:41:53]Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[12:41:53]Memory dump at 0x40162e50: 030000df d0091064 d04cd03b
[12:41:53]Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[12:41:53]Memory dump at 0x40162e50: 030000df d0091064 d04cd03b
[12:41:53]Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[12:41:53]Memory dump at 0x40162e50: 030000df d0091064 d04cd03b
[12:41:53]Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[12:41:53]Memory dump at 0x40162e50: 030000df d0091064 d04cd03b
[12:41:53]Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[12:41:53]Memory dump at 0x40162e50: 030000df d0091064 d04cd03b
[12:41:53]Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[12:41:53]Memory dump at 0x40162e50: 030000df d0091064 d04cd03b
[12:41:53]Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[12:41:53]Memory dump at 0x40162e50: 030000df d0091064 d04cd03b
```

OttoWinter, 2019-12-03:
Ah ok I see. Using esphome-flasher with ESPHome v1.13.6 won't work for certain ESP32s because of the bootloader version.

There's not much that can be done here. ESPHome-Flasher can't auto-detect from the fw blob which bootloader to use.

You can use an older ESPHome-Flasher version, but in general I would say ESPHome-Flasher only supports the most recent ESPHome version.

fribse, 2019-12-04:
Ok, that explains it, thankyou for clearing that up. I will keep it at 1.14.3, and hope that my esp8266 and esp32 will converge into one version at some point (I have to restore 1.13.6/upgrade to 1.14.3 all the time at the moment :-D)

OttoWinter, 2019-12-05:
Ok, as this problem is not really fixable I will close the issue.

The roots of the issues with 1.14.3 you're seeing with your ESP32 should be discussed in another thread.

closed 2019-12-05 by OttoWinter