esphome/issues #867: DF-Player USB working

redskinhu, 2019-11-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Linux x86_64, Docker, Hass.io 0.101.3

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266 / ESP-12F

**Affected component:**
https://esphome.io/components/dfplayer.html

**Description of problem:**
Requested in the docs. (If you connect a USB stick and found it works please create an issue at ESPHome GitHub.) 1-32GB USB drive connected to the DF-Player module (+5V, USB-, USB+, GND).

**Problem-relevant YAML-configuration entries:**

I modified the docs:
```yaml
  - service: dfplayer_set_device_tf
    then:
      - dfplayer.set_device:
          device: TF_CARD

  - service: dfplayer_set_device_usb
    then:
      - dfplayer.set_device:
          device: USB
```
But it could be good to use this way:
```yaml
  - service: dfplayer_set_device
    variables:
      device: string
    then:
      - dfplayer.set_device:
          device: !lambda 'return device;'
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2019-11-21:
Thanks I’ll update docs. 

Setting device directly from the string won’t work. But you can do some in-line if there and return the enum value, e.g. 

`!lambda: 'return (device == "USB")? dfplayer::USB : dfplayer::TF_CARD;'`

glmnet, 2019-11-22:
Ahhh this won’t work as device is not Templatable

Use a service per type 
Use esphome `if:`
Use set_device api

redskinhu, 2019-11-22:
Yes, I realized that when I tried template.

closed 2019-11-22 by glmnet