esphome/issues #868: DF-Player EQ issue

redskinhu, 2019-11-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Linux x86_64, Docker, Hass.io 0.101.3

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266 / ESP-12F

**Affected component:**
https://esphome.io/components/dfplayer.html

**Description of problem:**
The EQ is not working as suggested. It can't accept the ( NORMAL, POP, ROCK, JAZZ, CLASSIC and BASS) values if I call the service from the HASS with this service data `{ "preset": "ROCK" }` but it accept numbers `{ "preset": 1 }`. I dug out the number / value pairs from the source:
```
enum EqPreset {
  NORMAL = 0,
  POP = 1,
  ROCK = 2,
  JAZZ = 3,
  CLASSIC = 4,
  BASS = 5,
};
```
**Problem-relevant YAML-configuration entries:**
```yaml
  - service: dfplayer_set_eq
    variables:
      preset: int
    then:
      - dfplayer.set_eq: !lambda 'return static_cast<dfplayer::EqPreset>(preset);'
```

**Logs (if applicable):**
Home Assistant error if I use `{ "preset": "ROCK" }` service data.
```
Failed to call service esphome/dfplayer_esp12f_dfplayer_set_eq. expected int for dictionary value @ data['preset']
```

**Additional information and things you've tried:**

brandond, 2019-11-21:
Yes, when calling it like that you need to use the actual integer values. The presets are not strings, they're enum constants. The python code takes care of converting the string to enum value during code generation if you call is as an action, but in lambda C code you will have to manage the lookup yourself, as you noted.

glmnet, 2019-11-22:
Exactly as @brandond said, if you really want to accept strings comming from HA you can define the service like this:

````

    - service: dfplayer_set_eq
      variables:
        preset: string
      then:
        - if:
            condition:
              lambda: |-
                return (preset == "NORMAL");
            then:
              - dfplayer.set_eq: NORMAL
        - if:
            condition:
              lambda: |-
                return (preset == "POP");
            then:
              - dfplayer.set_eq: POP
        - if:
            condition:
              lambda: |-
                return (preset == "ROCK");
            then:
              - dfplayer.set_eq: ROCK
        - if:
            condition:
              lambda: |-
                return (preset == "JAZZ");
            then:
              - dfplayer.set_eq: JAZZ
        - if:
            condition:
              lambda: |-
                return (preset == "CLASSIC");
            then:
              - dfplayer.set_eq: CLASSIC
        - if:
            condition:
              lambda: |-
                return (preset == "BASS");
            then:
              - dfplayer.set_eq: BASS
````

closed 2019-11-22 by glmnet