esphome/issues #869: BLE tracker, OTA and not so good Wifi connections

Alex9779, 2019-11-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io and pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 modules and dev boards

**Affected component:**

https://esphome.io/components/esp32_ble_tracker.html
https://esphome.io/components/ota.html

**Description of problem:**
Playing with my Xiaomi devices and the BLE tracker scan parameters I stumble over another issue.
Depending on the scan parameters and the quality of the wifi connection I had serious problems to use OTA to get new firmwares to my devices.
Small windows and intervals or a ration greater 0.5 of window/interval seem to block the device for continuous reliable wifi communication at least for long transmissions which OTA is.
Devices at locations with a bad wifi connection are affected more than others.
Sometimes OTA does not start or the communication stops during transmission of the new interface. I also see a high increased upload time.
During my last test I was not able to push new firmware to two devices, I pinged the devices then constantly and saw replies from 5ms up to 5000ms during normal operation. Pinging helped to shift the focus of the device to wifi it seems, in the end I was able to push new firmwares during the ping operation after a few tries.
I searched for a solution and am now using a hacked dev version where I disable the BLE scanner when OTA starts. With this hack I am able to push a firmware to all devices no matter what the BLE scan params are.
My question and the purpose of this issue is, is there a way to access modules from other modules? SO that in case of an aborted OTA the scan can be restarted?
Stopping it is easy just using the esp function which has no parameters and works globally, but starting it again is not possible without the settings of the module.
Or I would just restart the device after an OTA try, if it is successful it gets restarted anyway why not if failed?
I read another issue with the ESP32 cam which also seems to bind the main process too much so a successful OTA is not often possible.
Maybe this could be a general improvement of deactivating critical processes when an OTA starts?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

Luc3as, 2019-11-22:
I am experiencing  the same issue here.

Luc3as, 2019-11-22:
I found another thread where is this described https://github.com/esphome/issues/issues/824#issuecomment-553884382 can you point me to how did you disabled BLE scan when OTA starts ?

Alex9779, 2019-11-22:
ota_component.cpp around line 10ff:
```
#ifdef ARDUINO_ARCH_ESP32
#include <Update.h>
#include <esp_gap_ble_api.h>
#endif
```
and right before line 69:
```
#ifdef ARDUINO_ARCH_ESP32
    esp_ble_gap_stop_scanning();
  #endif
```

This is just a dirty hack as the ble gap is system wide, I can't restart the scan if OTA fails and if you used too short durations for the scan then the watchdog for that scanner will step in and restart the scan. I am working with the default duration of 300s so with around 40s for an OTA update this is ok for now.
Basically I would like to be able to control the scanner module, extend it so that I can stop the scan and the watchdog and restart it...

dedors, 2020-01-01:
I experience the same issue. Got my first two ESP32 (mhetesp32minikit and mhetesp32devkit) and as soon as I added ble_tracker for my (many) Xiaomi sensors, OTA stops working (I get 0, rarely 1-2%). Happened on both. As soon as I update via USB and remove ble_tracker, the OTA works again.

Alex9779, 2020-01-02:
My experience the last few weeks was that with the BLETracker default values for interval and window OTA works normal for me, but for some devices, mostly Xiaomi stuff I have to increase the window, 120ms works so far and also allows OTA to run most of the times normal. But if you increase the windows to about 2/3 or greater of the interval then OTA is gambling...

Misiu, 2020-01-07:
@Alex9779 I really like the idea to disable BLETracker when OTA starts.
I saw multiple projects that do a restart when the OTA failed.
@OttoWinter what do you think about disabling BLETracker? Any ideas about how to restart it?

tismondo, 2020-06-15:
I'm also experiencing OTA problems. I cannot complete an OTA at all and have been playing with interval/window settings in the hopes of finding something that works.  Disabling BLE immediately prior to an OTA seems a reasonable (partial) solution.

buxtronix, 2020-06-17:
OTA is just another component, so perhaps the ble_tracker could expose a switch and OTA could turn it off prior to update (and re-enable if ota failed) via some lambda-style automation.

JoeSydney, 2020-09-15:
I have had issues with OTA when BLE Tracker is enabled... I just upgraded to 1.15.0 and 3 out of 3 times, OTAs worked fine! Would be interested to see if this is working for others as well, or if I've just been lucky.

marc-gist, 2020-10-30:
is there a way to setup two devices? I.e. have an esp32 dedicated to BLE and talk via serial to an ESP8266/32 that would handle WiFi?

Alex9779, 2020-10-30:
Sure this is possible, the upcoming Ruuvi Gateway does this but not with two ESPs. They use an ESP for Wifi and a Nordic NRF for BT...
You just need two firmwares and some kind of protocol, multiple possibilities here...

Alex9779, 2020-11-04:
As I don't have so many issues with recent builds I close this for now...

closed 2020-11-04 by Alex9779