esphome/issues #870: Unavailable sensor from HA is not handled in ESPHome (just sets value to 0)

fribse, 2019-11-22:
Home Assistant 0.101.3, ESPHome 1.13.6 / 1.14.0 / 1.14.2 / 1.14.3

ESP8266 D1 Mini with Waveshare ePaper 2.9" display

Sensor shows 0 instead of unavailable

I want to display a sensor state on a small display, so I fetch the binary sensor from Home Assistant.
If the device is offline, it shows unavailable in HA, but ESPHome cannot handle this.

                                                                                   ^

So it throws a warning in the log:
`[W][homeassistant.sensor:014]: Can't convert 'unavailable' to number!`
And the sensor is then set to 0.

It could be me just doing it wrong, as I see this in the compilation log:

```
src/main.cpp: In lambda function:
src/main.cpp:562:83: warning: format '%f' expects argument of type 'double', but argument 7 has type 'esphome::homeassistant::HomeassistantSensor*' [-Wformat=]
         it.printf(268, 71, bigtext, TextAlign::BASELINE_RIGHT, "%.1f°C", pooltemp);
```

**Problem-relevant YAML-configuration entries:**
```yaml
binary_sensor:
  - platform: homeassistant
    id: pumprunning
    entity_id: binary_sensor.pumprunning
    internal: true
```

**Logs (if applicable):**

```
`[W][homeassistant.sensor:014]: Can't convert 'unavailable' to number!`
```

**Additional information and things you've tried:**

I've tried different ways to see if I could catch the 'unavailable', but so far without any such luck.

glmnet, 2019-11-22:
I guess you need to add `.state` to `id(pooltemp)`

That should clear the warning.

Is it working as is? I mean, do you get temperature readings when it is online?

glmnet, 2019-11-22:
On the other hand, that warning the logger wrote there means you will get a `NAN` in `state`, so `has_state()` won't help you here but you can change that check by 

```` 
!isnan(id(pooltemp).state)
````

fribse, 2019-11-22:
Hllo @glmnet You were absolutely right on both points.
If I add !isnan as a test, it absolutely shows offline when it's not there, and the sensor when it's there.
And I also needed to add the .state to it, otherwise it just showed a big fat 0 ;-) And adding the .state also removed the warning!
Thankyou, now I learned a new test in esphome, very cool :-)
So I guess it's not an ESP problem, but my problem ;-D

closed 2019-11-22 by fribse