esphome/issues #874: Does not save when 'Save' is clicked

PierreScerri, 2019-11-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio 102/ESPHome add-on
**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

N/A

**Description of problem:**
Edits are not saved when 'Save' is clicked.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2019-11-29:
More information please:

 - What browser is this with? Does it happen with different browsers as well?
 - When you open up the debug console of the browser, does it show anything? (Right click anywhere on the page, click inspect element and go to tab "Console")
 - What is the Hass.io setup? Do you use ingress or not?

PierreScerri, 2019-11-29:
Browser: Safari.  Does not happen on Chrome

Debug:
[Error] Failed to load resource: The network connection was lost. (edit, line 0)
[Error] Unhandled Promise Rejection: TypeError: The network connection was lost.
	(anonymous function)
	promiseReactionJob
Setup: Hassio /Hassos on RPi3 Ingress in use.

Was working some time ago. Can't tell what changed.

kerimt-dek, 2019-12-24:
same here, with safari and virtual box hassio vm

jblankendaal, 2020-02-18:
Same here, latest hassio version 0.105.5, on a rpi4, Safari Version 13.0.5 (15608.5.11).
Could be something with keep-alive server setting according to this stackoverflow issue
[https://stackoverflow.com/questions/33895463/safari-ajax-request-failed-to-load-resource-the-network-connection-was-lost/46469548](url)

Sometimes after pressing save a few times it does save and then displays the Saved popup message.

It happens on line 656 of esphome.js

[image]
[image]

closed 2020-04-29 by glmnet