esphome/issues #876: first time run hang at archiving libFrameworkArduino.a

austrisv, 2019-11-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
pip on windows commmandline

**ESP (ESP32/ESP8266, Board/Sonoff):**

esp32 with ttgo-t display

**Affected component:**

first upload

**Description of problem:**
have just finished `esphome livingroom.yaml wizard`
Then, when running `esphome livingroom.yaml run` it hangs at `Archiving .pioenvs\livingroom2\libFrameworkArduino.a`

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: newnode
  platform: ESP32
  board: ttgo-t1

wifi:
  ssid: "aaaa"
  password: "12345678"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Newnode Fallback Hotspot"
    password: "UFGSsp0Swbyn"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2019-11-26:
What computer is this? The archiving step can take quite a bit of time on lower end machines.

The compiling is handled by platformio - so the archiving/compiling would be an issue in the toolchain distributed through them. But I think it's more of a resource exhausition issue with your machine.

austrisv, 2019-11-27:
It is Lenovo X1
 i7-8550U CPU @ 1.80GHz, 1992 Mhz, 4 Core(s), 8 Logical Processor(s)
16Gb RAM
don't see much spike in CPU load or Mem usage, none of them maxing out:(

OttoWinter, 2019-11-29:
Hmm, so the machine certainly sounds beefy enough.

But that's pretty much all I can help - the issue would be somewhere in the linker (in the archiving step), which is published in this case by Espressif (and distributed through platformio).

austrisv, 2019-11-30:
Thank you for looking into it anyways. What else can I do if I want to give esphome a try? Try from docker config? Maybe compiling on a spare rpi? Anything else? TA!

OttoWinter, 2019-12-03:
I'm not sure if docker will behave the same way (don't know if docker on windows is using a VM or something else). But certainly worth a try.

A separate RPi should also work - see Hass.io for easy setup with the esphome addon.

austrisv, 2019-12-11:
with docker was able to get over the last Archiving and can see firmware.bin file created. Shall be good enough for my initial tests. Thank you for patience and your efforts supporting my weird config:)

closed 2019-12-11 by austrisv