esphome/issues #877: Home Assitant sensors alternates quickly from available to unavailable

nbergont, 2019-11-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

- Esphome 1.14.3
- Home Assistant 0.101.3 or 0.102.0
- Compilation from Debian (pip install)

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU (ESP8266)

**Affected component:**

[api.connection](https://esphome.io/components/api.html)

**Description of problem:**

Since esphome 1.14 my esphome configuration have connections problem with Home Assitant :
- Wifi connection seem stable (log & ota is OK)
- Home Assitant sensors alternates quickly from available to unavailable (some are always unavailable)
- After some investigations, I have setup verbose logs & disable uart. 
- It seem that api.connection is out of space with TCP buffer space (see log) and API disconnect
- I have try to reduce uart buffer and remote_receiver buffer : no change
- I have try a other NodeMCU (with nothing connected) : no change
- Remove log component : no change

I have found a strange work around to fix this bug : 
- Setup verbose logs (other levels don't fix the bug)
- Active uart log

And every thing works very well... It seem that uart logs writes slowdown something and solve api connections (probably related with ESPAsyncTCP ?).

**Problem-relevant YAML-configuration entries:**

test2.yaml
```yaml
substitutions: !include substitutions.yaml

esphome:
  name: test2
  platform: ESP8266
  board: esp12e
  #esp8266_restore_from_flash: True

wifi:
  ssid: 'xxxxxxxxx'
  password: 'xxxxxxxxxx'
  reboot_timeout: 0s

# Enable logging
logger:
  level: VERBOSE
  baud_rate: 0 #Remove this line seem correct the api connection problem

api:
  reboot_timeout: 0s

# Enable OTA
ota:

status_led:
  pin: 
    number: GPIO0
    inverted: True

remote_transmitter:
  pin: GPIO4
  carrier_duty_percent: 100%
  
remote_receiver:
  pin: GPIO5
[... 412 lines cut ...]

JAMMING_PING:  '010101010101010101010101'
```

**Logs (if applicable):**

```
[21:58:50][V][api.connection:567]: Hello from client: 'Home Assistant 0.102.0 (192.168.1.22)'
[21:58:50][D][api.connection:583]: Client 'Home Assistant 0.102.0 (192.168.1.22)' connected successfully!
[21:58:50][V][api.connection:651]: Cannot send message because of TCP buffer space
[21:58:51][D][api:067]: Disconnecting Home Assistant 0.102.0 (192.168.1.22)
[21:58:51][V][api.connection:567]: Hello from client: 'Home Assistant 0.102.0 (192.168.1.22)'
[21:58:51][D][api.connection:583]: Client 'Home Assistant 0.102.0 (192.168.1.22)' connected successfully!
[21:58:51][V][api.connection:651]: Cannot send message because of TCP buffer space
[21:58:51][D][api:067]: Disconnecting Home Assistant 0.102.0 (192.168.1.22)
[21:58:56][V][api.connection:567]: Hello from client: 'Home Assistant 0.102.0 (192.168.1.22)'
[21:58:57][D][api.connection:583]: Client 'Home Assistant 0.102.0 (192.168.1.22)' connected successfully!
[21:58:57][V][api.connection:651]: Cannot send message because of TCP buffer space
[21:58:57][D][api:067]: Disconnecting Home Assistant 0.102.0 (192.168.1.22)
[21:58:58][V][api.connection:567]: Hello from client: 'Home Assistant 0.102.0 (192.168.1.22)'
[21:58:58][D][api.connection:583]: Client 'Home Assistant 0.102.0 (192.168.1.22)' connected successfully!
[21:58:58][V][api.connection:651]: Cannot send message because of TCP buffer space
```

**Additional information and things you've tried:**

spattinson, 2019-11-23:
I was getting a similar problem too a few days ago on an esp8266 that used remote_receiver, well it may have been the same issue I didn't have the api.connection messages because my log level was at default. It seemed to go away after a while if you just left it. I have just reflashed it with log level set to VERBOSE but I can't reproduce it anymore I am using version 1.14.3. This is my remote_receiver config:
```
remote_receiver:
  pin:
    number: 2
  dump: rc_switch
  tolerance: 50%
  filter: 50us
  buffer_size: 2kb
```

pove, 2019-11-24:
This also happens to me (HA 0.101.2, esphome 1.14.3), and also works fine when I enable logger and uart (commented on the code to make it fail):

```
esphome:
  name: wemos_14
  platform: ESP8266
  board: d1_mini

wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
  - ssid: !secret wifi_ssid_2
    password: !secret wifi_password_2

api:
  password: !secret api_password

ota:
  password: !secret ota_password

#logger:
# level: VERBOSE

#uart:
#  tx_pin: GPIO1
#  rx_pin: GPIO3
#  baud_rate: 9600

remote_receiver:
  pin: 
    number: GPIO14  # GPIO14 --> D5
    inverted: True
  dump: all

remote_transmitter:
  pin: D6
  carrier_duty_percent: 50%

switch:
  - platform: template
    name: "Samsung TV Power"
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E040BF
  - platform: template
    name: "Samsung TV Mute"
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E0F00F
  - platform: template
    name: "Samsung TV Volume Up"
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E0E01F
  - platform: template
    name: "Samsung TV Volume Down"
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E0D02F
  - platform: template
    name: "Samsung TV Home"
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E09E61
  - platform: template
    name: "Samsung TV Menu"
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E058A7
  - platform: template
    name: "Samsung TV Enter"
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E016E9
  - platform: template
    name: "Samsung TV Return"
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E01AE5
  - platform: template
    name: "Samsung TV Exit"
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E0B44B
  - platform: template
    name: "Samsung TV Source"
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E0807F
  - platform: template
    name: "Samsung TV Play"
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E0E21D
  - platform: template
    name: "Samsung TV Pause"
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E052AD
  - platform: template
    name: "Samsung TV Stop"
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E0629D
  - platform: template
    name: "Samsung TV RW Prev"
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E0A25D
  - platform: template
    name: "Samsung TV FF Next"
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E012ED
  - platform: template
    name: "Samsung TV Up"
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E006F9
  - platform: template
    name: "Samsung TV Down"
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E08679
  - platform: template
    name: "Samsung TV Right"
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E046B9
  - platform: template
    name: "Samsung TV Left"
    turn_on_action:
      remote_transmitter.transmit_samsung:
        data: 0xE0E0A659
  - platform: template
    name: "Ceilin Fan Power"
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          protocol: 5
          code: '011111111111011'
          repeat: 
            times: 5
      - remote_transmitter.transmit_raw:
          code: [603]
  - platform: template
    name: "Ceilin Fan Light"
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          protocol: 5
          code: '011111111110111'
          repeat: 
            times: 5
      - remote_transmitter.transmit_raw:
          code: [601]
```

nbergont, 2019-11-24:
Yes, it seem happen when using remote_receiver component.
I have made more investigations :
- use esphome 1.15.0-dev
- modify source log in api_connection.cpp with
```
ESP_LOGV(TAG, "Cannot send message because of TCP buffer space %d %d %d", needed_space, this->client_->space(), message_type);
``` 
- New log output is
```
[12:04:46][V][api.connection:670]: Cannot send message because of TCP buffer space 73 70 17
```
needed_space = 73  
client_->space = 70  
message_type = 17  (use by send_list_entities_switch_response in api_pb2_service.cpp:128)

Something go wrong when sending switch state...

- Try with VERY_VERBOSE level is not very usefull

```
[12:30:45][D][api.connection:602]: Client 'Home Assistant 0.101.3 (192.168.1.36)' connected successfully!
[12:30:45][VV][api.service:017]: send_connect_response: ConnectResponse {
  invalid_password: NO
}
[12:30:45][VV][api.service:266]: on_device_info_request: DeviceInfoRequest {}
[12:30:45][VV][api.service:042]: send_device_info_response: DeviceInfoResponse {
  uses_password: NO
  name: 'test2'
  mac_address: '5C:CF:7F:F7:AA:48'
  esphome_version: '1.15.0-dev'
  compilation_time: 'Nov 24 2019, 12:27:00'
  model: 'PLATFORMIO_ESP12E'
  has_deep_sleep: NO
}
[12:30:45][VV][api.service:273]: on_list_entities_request: ListEntitiesRequest {}
[12:30:45][VV][api.service:053]: send_list_entities_binary_sensor_response: ListEntitiesBinarySensorResponse {
  object_id: 'pir_1'
  key: 1036708158
  name: 'PIR 1'
  unique_id: 'test2binary_sensorpir_1'
  device_class: 'motion'
  is_status_binary_sensor: NO
}
[12:30:45][VV][api.service:053]: send_list_entities_binary_sensor_response: ListEntitiesBinarySensorResponse {
  object_id: 'pir_2'
  key: 1036708157
  name: 'PIR 2'
  unique_id: 'test2binary_sensorpir_2'
  device_class: 'motion'
  is_status_binary_sensor: NO
}
[12:30:45][VV][api.service:053]: send_list_entities_binary_sensor_response: ListEntitiesBinarySensorResponse {
  object_id: 'pir_3'
  key: 1036708156
  name: 'PIR 3'
  unique_id: 'test2binary_sensorpir_3'
  device_class: 'motion'
  is_status_binary_sensor: NO
}
[12:30:45](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:45](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:45](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:45](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:45](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:45](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:45](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:45](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:45](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:45](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:45](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[... 107 lines cut ...]
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[12:30:46][D][api:067]: Disconnecting Home Assistant 0.101.3 (192.168.1.36)
```

nbergont, 2019-11-24:
From fresh install of Home Assistant and only buggy esphome, I got this error from Home assistant log : 
```
components/esphome/__init__.py (WARNING) 
Error getting initial data: Timeout while waiting for API response!
```

And also this error : 
```
components/esphome/__init__.py (ERROR)
Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 674, in _async_remove_listener
    self._listeners[event_type].remove(listener)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 539, in __eq__
    if self.unique_id is None or other.unique_id is None:
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 575, in unique_id
    if not self._static_info.unique_id:
  File "/usr/src/homeassistant/homeassistant/components/esphome/binary_sensor.py", line 32, in _static_info
    return super()._static_info
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 538, in _static_info
    info = self._entry_data.info[self._component_key].get(self._key)
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 533, in _entry_data
    return self.hass.data[DATA_KEY][self._entry_id]
KeyError: 'd716770d54b74477b3223e2b43ea07e9'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 432, in async_remove_entity
    await self.entities[entity_id].async_remove()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 472, in async_remove
    self._on_remove.pop()()
  File "/usr/src/homeassistant/homeassistant/core.py", line 617, in remove_listener
    self._async_remove_listener(event_type, listener)
  File "/usr/src/homeassistant/homeassistant/core.py", line 682, in _async_remove_listener
    _LOGGER.warning("Unable to remove unknown listener %s", listener)
  File "/usr/local/lib/python3.7/logging/__init__.py", line 1390, in warning
    self._log(WARNING, msg, args, **kwargs)
  File "/usr/local/lib/python3.7/logging/__init__.py", line 1514, in _log
    self.handle(record)
  File "/usr/local/lib/python3.7/logging/__init__.py", line 1524, in handle
    self.callHandlers(record)
  File "/usr/local/lib/python3.7/logging/__init__.py", line 1586, in callHandlers
    hdlr.handle(record)
  File "/usr/local/lib/python3.7/logging/__init__.py", line 894, in handle
    self.emit(record)
  File "/usr/src/homeassistant/homeassistant/components/system_log/__init__.py", line 176, in emit
    record, stack, _figure_out_source(record, stack, self.hass)
  File "/usr/src/homeassistant/homeassistant/components/system_log/__init__.py", line 102, in __init__
    self.message = record.getMessage()
  File "/usr/local/lib/python3.7/logging/__init__.py", line 369, in getMessage
    msg = msg % self.args
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 554, in __repr__
    return "<Entity {}: {}>".format(self.name, self.state)
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 589, in name
    return self._static_info.name
  File "/usr/src/homeassistant/homeassistant/components/esphome/binary_sensor.py", line 32, in _static_info
    return super()._static_info
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 538, in _static_info
    info = self._entry_data.info[self._component_key].get(self._key)
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 533, in _entry_data
    return self.hass.data[DATA_KEY][self._entry_id]
KeyError: 'd716770d54b74477b3223e2b43ea07e9'
```

BertrumUK, 2019-11-25:
I am also seeing this. It seems to occur only when I try to add an IR transmitter to the config 

```
  - platform: template
    name: Remote Button 0
    turn_on_action:
      - remote_transmitter.transmit_raw:
          code: !secret humax0
          transmitter_id: IR
          carrier_frequency: 38kHz

It seems stable with a 433Mhz transmitter and receiver only but as soon as I add 1 single IR transmitter or more it goes unstable.

```

OttoWinter, 2019-11-26:
@nbergont 
> Cannot send message because of TCP buffer space

That should not be an issue - all important messages are re-sent when the buffer has run out of space (but not for logs, as storing those is not possible on the limited memory - hence the message this is only cosmetic).

Also your device is *really* low on TCP buffer space. Could you enable the debug component and print how much heap memory is free?

> Error getting initial data: Timeout while waiting for API response!

That looks like the TCP buffer is taking a *long* time to clear - so the messages are sent out very slowly. After something like 15s of no data, HA just closes the connection an retries - which is correct.

> During handling of the above exception, another exception occurred:

That is a different issue, and appears when deleting a config entry - but it doesn't affect the HA install (and should be partially fixed in next version too).

> I am also seeing this. It seems to occur only when I try to add an IR transmitter to the config

ESPHome does not distinguish between IR and RF - I don't see how enabling two of them could cause any difference - may be related to interrupt disable though.

BertrumUK, 2019-11-27:
I have just added 1 IR transmitter switch back and the issues appears straight away. Logs show this 

```
[12:40:19][D][api.connection:583]: Client 'Home Assistant 0.102.2 (192.168.0.60)' connected successfully!
[12:40:19][D][debug:204]: Free Heap Size: 8552 bytes
[12:40:20][V][api.connection:567]: Hello from client: 'Home Assistant 0.102.2 (192.168.0.81)'
[12:40:20][D][api.connection:583]: Client 'Home Assistant 0.102.2 (192.168.0.81)' connected successfully!
[12:40:25][D][dallas.sensor:148]: 'Pump Temperature': Got Temperature=49.4°C
[12:40:25][V][sensor:013]: 'Pump Temperature': Received new state 49.437500
[12:40:25][D][sensor:092]: 'Pump Temperature': Sending state 49.43750 °C with 1 decimals of accuracy
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.0.130:6053 (192.168.0.130)
WARNING Couldn't connect to API (Error connecting to 192.168.0.130: timed out). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.0.130:6053 (192.168.0.130)
INFO Successfully connected to 192.168.0.130
[12:41:28][V][api.connection:567]: Hello from client: 'Home Assistant 0.102.2 (192.168.0.60)'
[12:41:28][D][api.connection:583]: Client 'Home Assistant 0.102.2 (192.168.0.60)' connected successfully!
```

A DS18B20 temperature sensor was added this morning and there were no issues - it only seems happens if I have both the 433 and IR switches in the config. I have tried on a NodeMCU and Wemos D1 Mini and both do the same.

nbergont, 2019-11-27:
@OttoWinter thank for answer so quickly !

I have made a minimal configuration that reproduce the bug :
 
```yaml
esphome:
  name: test2
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: 'xxxxxxx'
  password: 'xxxxxx'
  reboot_timeout: 0s

# Enable logging
logger:
  level: VERBOSE
  baud_rate: 0

#debug:

api:
  reboot_timeout: 0s

# Enable OTA
ota:

status_led:
  pin: 
    number: GPIO0
    inverted: True

remote_transmitter:
  pin: GPIO4
  carrier_duty_percent: 100%
  
remote_receiver:
  pin: GPIO5
  #dump: raw

binary_sensor:
  - platform: remote_receiver
    name: "PIR 1"
    device_class: motion
    rc_switch_raw:
      code: '010101010101010101010101'
      protocol:
        pulse_length: 385
    filters:
       - delayed_off: 2s
       
  - platform: remote_receiver
    name: "PIR 2"
    device_class: motion
    rc_switch_raw:
      code: '010101010101010101010101'
      protocol:
        pulse_length: 385
    filters:
       - delayed_off: 2s

  - platform: remote_receiver
    name: "PIR 3"
    device_class: motion
    rc_switch_raw:
      code: '010101010101010101010101'
      protocol:
        pulse_length: 385
    filters:
       - delayed_off: 2s
       
  - platform: remote_receiver
    name: "PIR 4"
    device_class: motion
    rc_switch_raw:
      code: '010101010101010101010101'
      protocol:
        pulse_length: 385
    filters:
       - delayed_off: 2s
switch:

  - platform: template
    name: "Transmit 1"
    turn_on_action:
      remote_transmitter.transmit_rc_switch_raw:
        code: '010101010101010101010101'
        protocol: 1
        repeat: 
          times: 10
          wait_time: 0s
          
  - platform: template
    name: "Transmit 2"
    turn_on_action:
      remote_transmitter.transmit_rc_switch_raw:
        code: '010101010101010101010101'
        protocol: 1
        repeat: 
          times: 10
          wait_time: 0s
 
  - platform: template
    name: "Transmit 3"
    turn_on_action:
      remote_transmitter.transmit_rc_switch_raw:
        code: '010101010101010101010101'
        protocol: 1
        repeat: 
          times: 10
          wait_time: 0s
          
  - platform: template
    name: "Transmit 4"
    turn_on_action:
      remote_transmitter.transmit_rc_switch_raw:
        code: '010101010101010101010101'
        protocol: 1
        repeat: 
          times: 10
          wait_time: 0s

  - platform: template
    name: "Transmit 5"
    turn_on_action:
      remote_transmitter.transmit_rc_switch_raw:
        code: '010101010101010101010101'
        protocol: 1
        repeat: 
          times: 10
          wait_time: 0s
```
**Observations** : 
- If you remove one or two transmitter : no more disconnections
- Last switch Transmit 4 & 5 stay offline
- With this yaml configuration no TCP buffer space error
- If you active debug component : no more disconnections (Free Heap Size: 30216 bytes)
- Disconnecting append every 20s exactly (see log)
- After waiting ~25min connections seem stable (see capture)

> That looks like the TCP buffer is taking a _long_ time to clear - so the messages are sent out very slowly. After something like 15s of no data, HA just closes the connection an retries - which is correct.

You have right, this happens randomly if YAML become bigger.
Next step : analyze TCP communication to find out what is missing.

Log : 
```
[19:05:34][D][api:067]: Disconnecting Home Assistant 0.101.3 (192.168.1.36)
[19:05:40][V][api.connection:567]: Hello from client: 'Home Assistant 0.101.3 (192.168.1.36)'
[19:05:40][D][api.connection:583]: Client 'Home Assistant 0.101.3 (192.168.1.36)' connected successfully!
[19:06:00][D][api:067]: Disconnecting Home Assistant 0.101.3 (192.168.1.36)
[19:06:06][V][api.connection:567]: Hello from client: 'Home Assistant 0.101.3 (192.168.1.36)'
[19:06:06][D][api.connection:583]: Client 'Home Assistant 0.101.3 (192.168.1.36)' connected successfully!
[19:06:26][D][api:067]: Disconnecting Home Assistant 0.101.3 (192.168.1.36)
```
Home assistant "Transmit 2" become stable suddenly : 
[image]

bram-rongen, 2019-12-07:
It seems like I'm running into the same issue here. I've built a custom component which defines 11 sensors. When setting the state of all these sensors at once the home assistant api gets disconnected.

code for the component:
```
#include "esphome.h"

class DsmrP1CustomSensor : public PollingComponent
{
public:
    Sensor *consumption_low_tarif_sensor = new Sensor();
    Sensor *consumption_high_tarif_sensor = new Sensor();
    Sensor *actual_consumption_sensor = new Sensor();
    Sensor *instant_power_current_sensor = new Sensor();
    Sensor *instant_power_usage_sensor = new Sensor();
    Sensor *gas_meter_m3_sensor = new Sensor();
    Sensor *actual_tarif_sensor = new Sensor();
    Sensor *short_power_outages_sensor = new Sensor();
    Sensor *long_power_outages_sensor = new Sensor();
    Sensor *short_power_drops_sensor = new Sensor();
    Sensor *short_power_peaks_sensor = new Sensor();

    DsmrP1CustomSensor() : PollingComponent(1000) {}

    void update() override
    {
        consumption_low_tarif_sensor->publish_state(123321);
        consumption_high_tarif_sensor->publish_state(312312);
        actual_consumption_sensor->publish_state(220);
        instant_power_current_sensor->publish_state(123123);
        instant_power_usage_sensor->publish_state(123);
        gas_meter_m3_sensor->publish_state(1221);
        actual_tarif_sensor->publish_state(2);
        short_power_outages_sensor->publish_state(0);
        long_power_outages_sensor->publish_state(0);
        short_power_drops_sensor->publish_state(0);
        short_power_peaks_sensor->publish_state(0);
    }
};
```
and the yaml:
```
esphome:
  name: breakerbox
  platform: ESP8266
  board: d1_mini
  includes:
    - dummy_sensor.h

wifi:
  ssid: !secret esphome_wifi_ssid
  password: !secret esphome_wifi_pass

debug:

logger:

api:
  password: !secret esphome_api_pass

ota:
  password: !secret esphome_ota_pass

sensor:
  - platform: uptime
    name: Breakerbox uptime
  - platform: custom
    lambda: |-
      auto dsmr_p1_sensor = new DsmrP1CustomSensor();
      App.register_component(dsmr_p1_sensor);
      return {dsmr_p1_sensor->consumption_low_tarif_sensor, dsmr_p1_sensor->consumption_high_tarif_sensor, dsmr_p1_sensor->actual_consumption_sensor, dsmr_p1_sensor->instant_power_current_sensor, dsmr_p1_sensor->instant_power_usage_sensor, dsmr_p1_sensor->gas_meter_m3_sensor, dsmr_p1_sensor->actual_tarif_sensor, dsmr_p1_sensor->short_power_outages_sensor, dsmr_p1_sensor->long_power_outages_sensor, dsmr_p1_sensor->short_power_drops_sensor, dsmr_p1_sensor->short_power_peaks_sensor};
    sensors:
      - name: "Consumption Low Tarif Sensor"
        unit_of_measurement: kWh
        accuracy_decimals: 4
        filters:
          - multiply: 0.001
      - name: "Consumption High Tarif Sensor"
        unit_of_measurement: kWh
        accuracy_decimals: 3
        filters:
          - multiply: 0.001
      - name: "Actual Consumption Sensor"
        unit_of_measurement: kW
        accuracy_decimals: 3
        filters:
          - multiply: 0.001
      - name: "Instant Power Current Sensor"
        unit_of_measurement: A
        accuracy_decimals: 3
        filters:
          - multiply: 0.001
      - name: "Instant Power Usage Sensor"
        unit_of_measurement: kW
        accuracy_decimals: 3
        filters:
          - multiply: 0.001
      - name: "Gas Meter M3 Sensor"
        unit_of_measurement: m3
        accuracy_decimals: 3
        filters:
          - multiply: 0.001
      - name: "Actual Tarif Sensor"
      - name: "Short Power Outages Sensor"
        accuracy_decimals: 0
      - name: "Long Power Outages Sensor"
        accuracy_decimals: 0
      - name: "Short Power Drops Sensor"
        accuracy_decimals: 0
      - name: "Short Power Peaks Sensor"
        accuracy_decimals: 0
``` 

When I decrease the amount of sensors setting the state to 6 the api disconnects stop. When I set the log level to very verbose the api disconnects stop. When I introduce a delay of a second between the first 6 and the last 5 sensors the api disconnects stop. 

The log doesn't provide any useful messages.

Not sure how I can contribute to a solution but at least this provides an easy way to reproduce the issue..

spattinson, 2019-12-07:
It happened to my  esp a few days ago. I increased logging to try to get info but the problem went away. I did notice that asynctcp updated when i flashed. I dont know if that has anything to do with it.

bram-rongen, 2019-12-07:
Even the very verbose log level doesn't fix the issue anymore now..

Adding a delay of 100ms between the status updates does seem to work reliably.
```
      consumption_low_tarif_sensor->publish_state(CONSUMPTION_LOW_TARIF);
      delay(100);
      consumption_high_tarif_sensor->publish_state(CONSUMPTION_HIGH_TARIF);
      delay(100);
      return_low_tarif_sensor->publish_state(RETURN_LOW_TARIF);
      delay(100);
      .......
```

peterforeman, 2019-12-15:
See also: #923. Adding delays indeed makes the problem "go away".

S-Przybylski, 2019-12-16:
Could also be related to 
Home Assistant disconnects after switching a switch group #794 
which is not solved by now.

brandond, 2019-12-18:
Yeah definitely related. Something to do with overrunning the buffer and truncating the protobuf stream, if I had to guess.

marcelodavanzo, 2020-01-10:
I can confirm I've got the same issue with an ESP01S connected to my doorbell circuit on ESPHome v1.14.3 and Home Assistant 0.103.6.

I did not have these issues with the same code running on 1.13.x. It started from 1.14.0 onwards. For some odd reason however, it went away on HA 0.103.5 but it's back now on 0.103.6.

[image]

[image]

```
[12:57:54][V][api.connection:672]: Error: Disconnecting Home Assistant 0.103.6 (192.168.xx.xxx)
[12:57:54][D][api:067]: Disconnecting Home Assistant 0.103.6 (192.168.xx.xxx)
[12:57:55][V][api.connection:567]: Hello from client: 'Home Assistant 0.103.6 (192.168.xx.xxx)'
[12:57:55][D][api.connection:583]: Client 'Home Assistant 0.103.6 (192.168.xx.xxx)' connected successfully!
```

```
esphome:
  name: doorbell
  platform: ESP8266
  board: esp01_1m
  arduino_version: dev

wifi:
  ssid: !secret ssid
  password: !secret ssid_pass
  manual_ip:
    static_ip: 192.168.xx.xxx
    gateway: 192.168.xx.xxx
    subnet: 255.255.255.0

logger:
  level: VERBOSE

api:

ota:

web_server:
  port: 80

time:
  - platform: homeassistant
    id: homeassistant_time

text_sensor:
  - platform: version
    name: Doorbell ESPHome Version

sensor:
  - platform: uptime
    name: Doorbell Uptime

  - platform: wifi_signal
    name: Doorbell WiFi Signal
    update_interval: 60s

globals:
  - id: chime
    type: bool
    restore_value: true
    initial_value: 'true'

switch:
  - platform: restart
    name: Doorbell Restart

  - platform: gpio
    id: relay
    inverted: true
    name: Doorbell Chime
    pin: GPIO0

  - platform: template
    name: Doorbell Chime Active
    id: chime_active
    restore_state: false
    turn_on_action:
      - globals.set:
          id: chime
          value: 'true'
    turn_off_action:
      - globals.set:
          id: chime
          value: 'false'
    lambda: |-
      return id(chime);
      
binary_sensor:
  - platform: gpio
    id: button
    name: Doorbell Button
    pin:
      number: GPIO2
      mode: INPUT_PULLUP
      inverted: true
    filters:
      - delayed_on: 40ms
      - delayed_off: 40ms
    on_press:
      then:
        if:
          condition:
            - switch.is_on: chime_active
          then:
            - switch.turn_on: relay
    on_release:
      - switch.turn_off: relay
```

ripvega, 2020-01-15:
I too experience this issue.

I've detailed [here on home Assistant](https://community.home-assistant.io/t/esphome-devices-lose-their-connection/160062/8?u=ashscott) forum.

I’ve been struggling with devices not being available to HA.

The device log (Sonoff RF Bridge) shows it continually disconnecting and reconnecting to the HA API:

```
[18:16:11][D][time:029]: Synchronized time: Mon Jan  6 18:16:11 2020
[18:16:11][D][api:067]: Disconnecting Home Assistant 0.103.2 (10.0.1.235)
[18:16:15][D][sensor:092]: 'Sonoff RF Bridge Wifi Signal': Sending state -67.00000 dB with 0 decimals of accuracy
[18:16:17][D][api.connection:583]: Client 'Home Assistant 0.103.2 (10.0.1.235)' connected successfully!
[18:16:17][D][time:029]: Synchronized time: Mon Jan  6 18:16:17 2020
[18:16:17][D][api:067]: Disconnecting Home Assistant 0.103.2 (10.0.1.235)
[18:16:22][D][api.connection:583]: Client 'Home Assistant 0.103.2 (10.0.1.235)' connected successfully!
[18:16:22][D][time:029]: Synchronized time: Mon Jan  6 18:16:22 2020
[18:16:23][D][api:067]: Disconnecting Home Assistant 0.103.2 (10.0.1.235)
[18:16:25][D][sensor:092]: 'Sonoff RF Bridge Wifi Signal': Sending state -67.00000 dB with 0 decimals of accuracy
[18:16:28][D][api.connection:583]: Client 'Home Assistant 0.103.2 (10.0.1.235)' connected successfully!
[18:16:28][D][time:029]: Synchronized time: Mon Jan  6 18:16:28 2020
[18:16:28][D][api:067]: Disconnecting Home Assistant 0.103.2 (10.0.1.235)
[18:16:33][D][api.connection:583]: Client 'Home Assistant 0.103.2 (10.0.1.235)' connected successfully!
```
I have another device (Node MCU) running ESPHome and it does not have this issue.

I have, by trial and error, determined that the problem occurs when the number of ‘remote_receiver’ devices exceeds 12.

If I drop below that number I don’t get the disconnection issue.

Here’s the yaml from the RF Bridge:

```
esphome:
  name: rf_bridge
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_SSID
  password: !secret wifi_password
#  manual_ip:
#    static_ip: 10.0.1.99
#    gateway: 10.0.1.1
#    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

switch:
  - platform: restart
    name: "RF Bridge Restart"

time:
  - platform: homeassistant
    id: homeassistant_time

web_server:
  port: 80
  
sensor:
  - platform: wifi_signal
    name: Sonoff RF Bridge Wifi Signal
    update_interval: 10s
  - platform: uptime
    name: Sonoff RF Bridge Uptime

binary_sensor:
#  - platform: status
#    name: Sonoff RF Bridge Status

  - platform: remote_receiver
    name: Test Tamper
    device_class: motion
    rc_switch_raw:
      code: '110101110001010110011011'
      protocol: 1
    filters:
      delayed_off: 5s

### Motion Sensors ###
  - platform: remote_receiver
    name: Garage Motion
    device_class: motion
    rc_switch_raw:
      code: '110100110010001001101110'
      protocol: 1
    filters:
      delayed_off: 5s

  - platform: remote_receiver
    name: Telegraph Pole Garage
    device_class: motion
    rc_switch_raw:
      code: '010100011111011001011010'
      protocol: 1
    filters:
      delayed_off: 5s

  - platform: remote_receiver
    name: Gym Motion
    device_class: motion
    rc_switch_raw:
      code: '011100010101111111111111'
      protocol: 1
    filters:
      delayed_off: 5s

  - platform: remote_receiver
    name: Garage Motion 2
    device_class: motion
    rc_switch_raw:
      code: '100111110001010100001010'
      protocol: 1
    filters:
      delayed_off: 5s

  - platform: remote_receiver
    name: Letterbox
    device_class: motion
    rc_switch_raw:
      code: '101001101000101100100011'
      protocol: 1
    filters:
      delayed_off: 5s

  - platform: remote_receiver
    name: Stable Motion
    device_class: motion
    rc_switch_raw:
      code: '000101000010000110000110'
      protocol: 1
    filters:
      delayed_off: 5s

  - platform: remote_receiver
    name: North East PIR
    device_class: motion
    rc_switch_raw:
      code: '010101001111011001011010'
      protocol: 1
    filters:
      delayed_off: 5s

  - platform: remote_receiver
    name: North Central PIR
    device_class: motion
    rc_switch_raw:
      code: '010001011111011001001010'
      protocol: 1
    filters:
      delayed_off: 5s

  - platform: remote_receiver
    name: North West PIR
    device_class: motion
    rc_switch_raw:
      code: '010011011111011001001010'
      protocol: 1
    filters:
      delayed_off: 5s

  - platform: remote_receiver
    name: Garage North PIR
    device_class: motion
    rc_switch_raw:
      code: '010010011111011001001010'
      protocol: 1
    filters:
      delayed_off: 5s

  - platform: remote_receiver
    name: Pergola PIR
    device_class: motion
    rc_switch_raw:
      code: '010110011111011001011010'
      protocol: 1
    filters:
      delayed_off: 5s

  - platform: remote_receiver
    name: Courtyard PIR
    device_class: motion
    rc_switch_raw:
      code: '011011011111011001101010'
      protocol: 1
    filters:
      delayed_off: 5s

remote_receiver:
  pin: 4
  dump: rc_switch
  tolerance: 50
  filter: 4us
  idle: 4ms

remote_transmitter:
  pin: 5
  carrier_duty_percent: 100%

status_led:
  pin:
    number: GPIO13
    inverted: yes
```
There’s nothing too onerous in there.

sstefanowski, 2020-01-21:
Hi All,

I also experienced this issue.

The HW is Sonoff RF Bridge with "Hardware hack" - bypass EFM8BB1 microcontroller (https://www.youtube.com/watch?v=pXm1HIPmVuA) after "hardware hack" its just ESP8266.

ESPHome version 1.14.3
Home Assistant version: 0.103.0

I defined 24 sensors and issue started. I removed most of them (to like 4 sensors) and it was working fine. 

It's definitely painful issue as timing and responsiveness of "RF Bridge with ESPHome" to HA is way better than with other firmware (important if you have RF remotes/wall panels for light switching). Unfortunately, this issue is blocking from migration to ESP Home.

BartBourgeois, 2020-01-22:
Same problem here.
ESPHome version 1.14.3
Home Assistant version: 0.104.3

None of the suggestions in this thread seem to mitigate but problem.

But, what does seem to work for reasons yet unknown is opening the "Show Logs" on the device via the ESPHome Dashboard. As long as the logger is connected to the webclient, the device does not disconnect at all, not even once. As soon as the "show log" is closed, the disconnects start appearing again.

```yaml
esphome:
  name: nodemcu
  platform: ESP8266
  board: nodemcuv2
  includes:
    - my_custom_sensor.h
  libraries:
    - "Adafruit ADS1X15"

wifi:
  ssid: "xxx"
  password: "xxx"
  use_address: "nodemcu.xxx"
  power_save_mode: none

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

sensor:
- platform: custom
  lambda: |-
    auto my_sensor = new SCT013Sensor();
    App.register_component(my_sensor);
    return { my_sensor->volt_sensor,my_sensor->current_sensor[0],my_sensor->current_sensor[1],my_sensor->current_sensor[2],my_sensor->current_sensor[3],my_sensor->kwh_sensor[0],my_sensor->kwh_sensor[1],my_sensor->kwh_sensor[2],my_sensor->kwh_sensor[3] };    
  sensors:
  - name: "SCT013-Volt"
    unit_of_measurement: V
    accuracy_decimals: 2
  - name: "SCT013-01"
    unit_of_measurement: A
    accuracy_decimals: 2
  - name: "SCT013-02"
    unit_of_measurement: A
    accuracy_decimals: 2
  - name: "SCT013-03"
    unit_of_measurement: A
    accuracy_decimals: 2
  - name: "SCT013-04"
    unit_of_measurement: A
    accuracy_decimals: 2            
  - name: "SCT013-01-kWh"
    unit_of_measurement: kWh
    accuracy_decimals: 2
  - name: "SCT013-02-kWh"
    unit_of_measurement: kWh
    accuracy_decimals: 2
  - name: "SCT013-03-kWh"
    unit_of_measurement: kWh
    accuracy_decimals: 2
  - name: "SCT013-04-kWh"
    unit_of_measurement: kWh
    accuracy_decimals: 2
```

OkhammahkO, 2020-01-25:
I am experiencing the same issue(s).

drthanwho, 2020-02-09:
Same here 
ESPHOME version is 1.14.3
But nodes haven't been updated yet so they are all 1.13.6
Home Assistant: 0.105.2

```[18:54:47][D][sensor:092]: 'Koudouni Uptime': Sending state 105942.45312 s with 0 decimals of accuracy
[18:55:09][D][sensor:092]: 'Koudouni WiFi Signal': Sending state -58.00000 dB with 0 decimals of accuracy
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to entrance.local:6053 (192.168.1.20)
INFO Successfully connected to entrance.local
[18:55:54][D][time:030]: Synchronized time: Sun Feb  9 18:55:54 2020
[18:56:09][D][sensor:092]: 'Koudouni WiFi Signal': Sending state -58.00000 dB with 0 decimals of accuracy

[19:28:47][D][sensor:092]: 'Koudouni Uptime': Sending state 107982.45312 s with 0 decimals of accuracy
[19:29:09][D][sensor:092]: 'Koudouni WiFi Signal': Sending state -58.00000 dB with 0 decimals of accuracy
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to entrance.local:6053 (192.168.1.20)
INFO Successfully connected to entrance.local
[19:29:42][D][time:030]: Synchronized time: Sun Feb  9 19:29:42 2020
```

> But, what does seem to work for reasons yet unknown is opening the "Show Logs" on the device via the ESPHome Dashboard.
This isn't happening here though as I had the logs open on HA while it happened 

> I can confirm I've got the same issue with an ESP01S connected to my doorbell circuit on ESPHome v1.14.3 and Home Assistant 0.103.6.
>
> I did not have these issues with the same code running on [...]

To avoid make this too lengthy, my logs are also from the same setup. Frenck's ESP01S doorbell. But I can verify that it happened simultaneously for at least two other devices.

I will put something together to see if this seems to happen to all the nodes at exactly the same time

JonGilmore, 2020-02-16:
i'm also experiencing this same issue. all my devices were rock solid on 1.13.x - anybody know if there's a downgrade path?

marcelodavanzo, 2020-02-16:
In the config for the ESPHome add on you can enter `esphome_version: 1.13.6` and that will override the latest version. I am using `esphome_version: dev` and my defective nodes are now rock solid again so it looks like whatever issue there was has been fixed on the dev version but not yet part of the main release.

fillwe, 2020-02-26:
Have about 25 esphome devices, just having issues with the 6 ones running 1.14.3. Have tried updating all to 1.15 but didn’t solve it for me. Have tried all suggestions referenced in the esphome faq and to disable logger.

Can’t seem to downgrade to 1.13.6 either, getting an error when the esphome dashboard is about to start. Moved a few devices back to tasmota the other day and since then they have been stable without any disconnects. Really hoping that we can narrow down the issue!

drthanwho, 2020-02-27:
I set up some notifications to see if the error is consistent or just random. Seems like it will occasionally work fine for 3 days and then act up again like before

fillwe, 2020-02-29:
Have tried moving from api to mqtt but that didn't really make any difference. Still getting disconnects from wifi. It's just the ESP8266 devices though. My ESP32's running 1.15 is working without any issues

JonGilmore, 2020-03-01:
> I set up some notifications to see if the error is consistent or just random. Seems like it will occasionally work fine for 3 days and then act up again like before

This seems to be different than what I see - basically all of my esp8266 devices disconnect for 7-20 minutes at least a couple times per day.

drthanwho, 2020-03-01:
> This seems to be different than what I see - basically all of my esp8266 devices disconnect for 7-20 minutes at least a couple times per day.

Definitely different case. I've seen it from the history of sensors of the devices but also from the log itself. It looses API connection to HA and then connects again a few seconds later. I've not noticed any king disconnections so far at least.

vitalykorolev, 2020-03-01:
What version of HA are you using? I'm a couple of months behind because of
python upgrade and wondering if there is correlation between HA version and
ESPHome version.

glmnet, 2020-03-02:
Check what’s going on at the ya side. You should enable ha logs for esphome and get something like this https://github.com/esphome/issues/issues/426#issuecomment-500620617

nbergont, 2020-03-02:
I have found a "solution" for disable logging and solve disconnections problem.
Add some delay on every loop (maybe not recommended) :
```
esphome:
  #...
  on_loop:
    then:
      - lambda: delay(5);
```

I dont understand why this solve the problem, but it is working for me.

spattinson, 2020-03-02:
What I see is bursts of all the sensors disappearing which I assume is a TCP disconnect for whatever reason. Other times its only some of them, for example I configured a switch  to reboot the esp, currently that is not available. Sometimes it is only some of the binary sensors which are remote_receiver sensors (433Mhz). The switch for reboot is not mentioned in the logs, like the node does not have it in the configuration! At some point I may transfer all the sensors over to an esp32 instead, I get the feeling it is a resource issue. The 433Mhz is constantly flickering, I have it one the LED pin and its always on, so there's a lot of background noise, the esp is constantly examining all the noise to see if it matches any of my sensors. I have 10 remote receiver sensors, bmp280 temp/humidity, an analog sensor in ADC1, and a PIR on another pin.

peterforeman, 2020-03-02:
@spattinson What's happening is that the TCP buffer becomes full (fills too fast, faster than it sends).  Result is that it cannot append to the buffer anymore, resulting in an incomplete message to HASS which drops the connection because of incorrect message (protocol). 

This is why adding the delay, mentioned by @nbergont works: it delays the filling of the buffer so it can flush the buffer to HASS in time. See also @OttoWinter's previous response (https://github.com/esphome/issues/issues/877#issuecomment-558745873).

I'm wondering if there's any progress on resolving this issue @OttoWinter ? Can we help in any way?

spattinson, 2020-03-02:
I have a correction about my report, the unavailable entities had all been renamed with a _2 suffix probably something to do with the upgrade and entities needing unique IDs now. So a quick update of my lovelace solved that.

I may try putting a delay in, for now I disabled logging and set the update interval for each sensor so they are not all the same. 60s is too short anyway for my usage, I've made them 4, 5 and 7 minutes, maybe that will help with the size of the buffer because they won't all be sending at the same time.

glmnet, 2020-03-02:
May be this is related to the new scheduler https://github.com/esphome/esphome/pull/609

tezgray, 2020-03-10:
losing th esensors over night, show logs gives
ERROR Error while reading incoming messages: Error while receiving data: [Errno 104] Connection reset by peer
WARNING Disconnected from API: Error while receiving data: [Errno 104] Connection reset by peer

Even when connection established and seeing sensor data stays "unavailable" in Home Asistant.

Nodemcu
1 * DHT22
1 * TSL2561
1 * hc-sr505
4 LED strip accessed individually as a temp indicator

Wouldn't have thought that would overload things

Everything running on latest versions

vitalykorolev, 2020-03-12:
This issue is gone in my setup after I upgraded to the latest HA. I also move HA to a different hardware, but that shouldn't matter. In the process of transferring HA data from one host to another, I ended up deleting the DB and log files. It's been 5 days and not a single ESPhome sensor/switch disconnect.

BartBourgeois, 2020-03-12:
Update on my installation, was able to get rid of the problem by removing one specific ESPHome device. 

After debugging and monitoring, it became clear this one module had heavy (time consuming) code in the "loop" part. The loop took longer then period between loops, thus I guess overloading some component of ESPHome. 

Other devices on ESPHome where also experiencing disconnection problems because of this, after removing the one device, everything works again like expected. 

What I'am still not able to figure out, is why the problem DOES NOT occur when I view the Live-Log of this device via ESPHome. For some reason everything also works perfectly then.

brano1990, 2020-03-14:
I can confirm I've got the same issue with an d1_mini  is most busy
. two other work 
connected to  on ESPHome v1.14.3

theosoft-git, 2020-03-16:
I have the same issue with about 70 switches and both remote transmitter and receiver.
I 'fixed' this by remove the remote receiver.

ripvega, 2020-03-16:
So, you're only using your hub to transmit?

nbergont, 2020-03-21:
> 
> 
> I have found a "solution" for disable logging and solve disconnections problem.
> Add some delay on every loop (maybe not recommended) :
> 
> ```
> esphome:
>   #...
>   on_loop:
>     then:
>       - lambda: delay(5);
> ```
> 
> I dont understand why this solve the problem, but it is working for me.

Have you try this work around ? Work very well for me : 20 days uptime with no disconnection or reboot.

spattinson, 2020-04-11:
> I have found a "solution" for disable logging and solve disconnections problem.
> Add some delay on every loop (maybe not recommended) :
> 
> ```
> esphome:
>   #...
>   on_loop:
>     then:
>       - lambda: delay(5);
> ```
> 
> I dont understand why this solve the problem, but it is working for me.

Last night I reflashed all my esphome devices to change the ssid as I now have a dedicated wii for them. I was getting constant available/unavailable, different entities, different times, at any one time some or all would be missing. logbook getting massive, I have just implemented the above and it is stable again

drthanwho, 2020-04-12:
> esphome:
>   #...
>   on_loop:
>     then:
>       - lambda: delay(5);

I tried that but it didn't seem to work for me

onkelbeh, 2020-04-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

    ESPHome 1.14.3
    Home Assistant 0.108.3
    Native Install on Gentoo Linux in a virtual machine

**ESP (ESP32/ESP8266, Board/Sonoff):**

   NodeMCU (ESP8266) with 4 I2C IO Expanders PCF8574, 24 Output lines, 8 INPUT lines
   also tried a D1 Mini

**Affected component:**

api.connection

**Description of problem:**

API disconnects from Home Assistant, Dashboard stays connected, Log still is displayed.
I am not sure if this fits here, because these disconnects happen only on higher network load, mostly when a Scene is loaded and many outputs have to change their state.

Every time a bigger Scene loads (10 or more outputs change state), the following happens:
  1. Home assistant seems to hang shortly (about half a second)
  2. some states are updated in the gui, but not all
  3. all devices become unavailable for about one second
  4. as soon as the decives come back, the rest of the states are updated

Looks like more network traffic is happening than the ESP can handle.

This happens mostly with OUTPUTS configured as light, switch outputs seem to be more 'robust'. If I flip all lines the situations is reproducible, if 10 switches are flipped, it disconnects sometimes, if 10 lights are flipped, is disconnects often. Seems acting a light produces more network traffic.

I tried disabling/enabling logging, changed I2C pins, also the `on_loop delay` does not help. Changed arduino_version & board type, also replaced hardware with a different one. First it was a D1 Mini,now now to a NodeMCU with an 12F.

The problem box has 4 PCF8574 I2C IO Expander connected.

3 other 'Appliances' like that run perfect, but with mostly INPUT (and a very few OUTPUT), the  difference here is that 3 of the 4 PCF's are configured as 24 OUTPUT lines, and only 8 lines are INPUT. 

server.log says:
```
2020-04-12 22:39:43 INFO (MainThread) [aioesphomeapi.connection] 172.16.4.9: Error while reading incoming messages: Invalid preamble
2020-04-12 22:39:43 INFO (MainThread) [homeassistant.components.esphome] Disconnected from ESPHome API for 172.16.4.9
2020-04-12 22:39:44 INFO (MainThread) [homeassistant.components.esphome] Successfully connected to 172.16.4.9
```
Dashboard stays connected, it's log says:
```
[22:39:43][D][api:067]: Disconnecting Home Assistant 0.108.3 (172.16.2.24)
[22:39:44][D][api.connection:583]: Client 'Home Assistant 0.108.3 (172.16.2.24)' connected successfully!
```
Here's the current config:
```
substitutions:
  name: ff1

esphome:
  name: ${name}
  platform: ESP8266
  board: nodemcuv2
  #arduino_version: 2.4.2
  #on_loop:
  #  then:
  #    - lambda: delay(5);

<<: !include _common/global.yaml

sensor:
  - !include _common/sensor_wifi_signal.yaml
  - !include _common/sensor_uptime.yaml

i2c:
  #scl: GPIO5
  #sda: GPIO4
[... 299 lines cut ...]
      inverted: True
  - platform: gpio
    id: '${name}_Relais_22'
    pin:
      pcf8574: pcf8574_h22
      number: 5
      mode: OUTPUT
      inverted: True
  - platform: gpio
    id: '${name}_Relais_23'
    pin:
      pcf8574: pcf8574_h22
      number: 6
      mode: OUTPUT
      inverted: True
  - platform: gpio
    id: '${name}_Relais_24'
    pin:
      pcf8574: pcf8574_h22
      number: 7
      mode: OUTPUT
      inverted: True

```
Tomorrow I will the ESP12F with an ESP32 NodeMCU. If this won't help, I'll try the dev version next.

glmnet, 2020-04-12:
1. Try a full cleanup of the build, that is delete the folder that has the same name as your node name. I dont know for sure your case because you are using substitutions.

Most likely this wont fix it but just to be sure.

2. Increase verbosity level on the logs we need both sides of the logs, set esphome verbosity level to VERY_VERBOSE and HA level to DEBUG too, (you are only seeing INFO on server.log posted)

onkelbeh, 2020-04-13:
Here's a Video of the Lab, while doing the debug, note the strange timing.
https://youtu.be/jp41WVKHf1k

onkelbeh, 2020-04-13:
Thanks for your quick reponse.

1) did a clean build with logger level: very_verbose
2) changed HA's logger levels:
    homeassistant.components.esphome: debug
    aioesphomeapi: debug
3) Made sure all lines of the affected esp6 have been turned off
4) started logging on serial console (console via API is missing a lot)
5) started logging on server.log (grepped for esp6's IP, there are 22 more devices active, so the log is impressive. I you need to see the rest, just let me know.)
6) then I turned all 24 OUTPUTS on at once (via group switch)
7) waited or HA to complete the reconnect and all status is shown again (~ 6 seconds)
8) turned off all 24 lines again
9) stopped logging

level: very_verbose changes behavior a lot. It now takes about 6 seconds to set, normally this takes about 2-3 seconds. The first 8-10 outputs are switched one by one with a 200ms delay, the rest is switched very fast then.

current configs:
global.yaml:
```yaml
wifi:
  ssid: "WLAN4-HOME"
  password: !secret wifi
  power_save_mode: none

logger:
  #level: INFO
  level: very_verbose
  #level: verbose
  #baud_rate: 0
  logs:
    i2c: debug

api:
  password: !secret api

ota:
  password: !secret ota

#mqtt:
#  broker: mosquitto.in.edevau.net

time:
  - platform: sntp
    servers: 172.16.2.6
    timezone: Europe/Berlin
  #- platform: homeassistant
  #  id: homeassistant_time

binary_sensor:
  - !include binary_sensor_status.yaml

sensor:
  - !include sensor_wifi_signal.yaml

switch:
  - !include switch_restart.yaml

text_sensor:
  - !include text_sensor_version.yaml
  - !include text_sensor_wifi_info.yaml

web_server:
  auth:
    username: admin
    password: !secret api

```
esp6_ff1.yaml:
```yaml
substitutions:
[... 324 lines cut ...]
      number: 5
      mode: OUTPUT
      inverted: True
  - platform: gpio
    id: '${name}_Relais_23'
    pin:
      pcf8574: pcf8574_h22
      number: 6
      mode: OUTPUT
      inverted: True
  - platform: gpio
    id: '${name}_Relais_24'
    pin:
      pcf8574: pcf8574_h22
      number: 7
      mode: OUTPUT
      inverted: True

```
Here are the logs:
**Home Assistant:**
[ha_server.log](https://github.com/esphome/issues/files/4468361/ha_server.log)

**Serial console:**
[esp6_ff1_very_verbose.log](https://github.com/esphome/issues/files/4468362/esp6_ff1_very_verbose.log)

onkelbeh, 2020-04-13:
Hi again,
just changed the board to an ESP32 Lolin dev board. No Problems with that, everything working good. Seems to be a buffer/performance/load thing.

glmnet, 2020-04-14:
I must assume you also tried without `web_server`

onkelbeh, 2020-04-14:
Sure, first thing I tried was to turn off all unneccesary stuff, like timesync, web, logging, etc.
All I left on was OTA, API and the uptime & wifi sensors.

In my humble opinion this looks like when the ESP is flooded with requests too much outbound traffic from the acknowledgements is causing massive packet loss by a send buffer overrun.
Further debugging will be very time consuming, think there are not many ways to avoid this, some ideas:
1) limit the outgoing requests from the api to the ESP through a queue and release the next one as soon as the first is acknowleged.
2) accumulate some more switch events for the same target and the resulting acknowledgements in packages on the host, there is simply to much code to run for flipping a single I2C bit

Both ideas will totally break the current API concept. This is a big decision, and (at least here) this is only happening in a extreme setup for this tiny hardware, I will understand when a fix is not possible.

For now I will keep the ESP32, just ordered more. I have this amount of relays only in one big power distribution unit. The only thing I will loose here is the printed DIN-rail case for D1 mini, took this from thingiverse, so no CAD time was wasted ;-)

I the lab setup I was able to add total six PCF8574 (had not more) configured as OUTPUT without problems to the ESP32. This makes a total of 48 output lines with one wifi device, thats enough for this box :-)

And my other two boxes with 24 INPUT lines do not encounter any problems with the ESP12E's.

At this point a very big "Thank you" to all of you. ESPHome is a great thing.

onkelbeh, 2020-04-16:
After 2 days of intensive testing I can tell that the ESP32 migration fixed it. I will put it back in the power distribution this evening :-)

glmnet, 2020-04-18:
Even though using ESP32 is a work-around (not really), the issue still bothers me

I did a regrestion test, this PR causes this https://github.com/esphome/esphome/pull/633 makes sense since it is heavily related to api unfortunately it's a huge commit.

onkelbeh, 2020-04-19:
I will do a test with my (still existing) lab setup as soon as my testbox is up to date.

tb8851, 2020-04-20:
Hi, I've tested the ""fix-api-drop-connection" branch on a modified Sonoff RF Bridge.

On V1.14.3 I got disconnects every few seconds and the majority of sensors were showing as "unavailable" in Home Assistant...after upgrading (no other changes) I've had a stable connection for 1 hour with no disconnects and all sensors showing as available. And everything else seems to be fine too.

It's my first post - if I can provide any further information to assist please let me know!

Thank-you very much @glmnet !

glmnet, 2020-04-20:
well... let's see how this goes on... I just disabled a "feature" introduced in esphome/esphome#633 that is meant somehow to improve the communication speed.

jakowenko, 2020-04-21:
I am also facing a similar issue that seems to have just started with my EPS8266s. I have a pretty basic config. I tried to use the lambda delay, but it doesn't appear to have solved the issue. I get period disconnects and reconnects.

```
esphome:
  name: kitchen
  platform: ESP8266
  board: nodemcuv2
  on_loop:
    then:
      - lambda: delay(5);

wifi:
  ssid: !secret ssid
  password: !secret password
  manual_ip:
    static_ip: !secret ip
    gateway: !secret gateway
    subnet: !secret subnet

sensor:
  - platform: dht
    model: DHT11
    pin: D4
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
    update_interval: 60s
  - platform: uptime
    name: " Uptime"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
```

berthavekes2, 2020-04-23:
> Hi, I've tested the ""fix-api-drop-connection" branch on a modified Sonoff RF Bridge.
>
> On V1.14.3 I got disconnects every few seconds and the majority of sensors were showing as "unavailable" in [...]

I forked the repository, and created a local addon from that. I updated one of my lamps. I will let you all know if it worked.

berthavekes2, 2020-04-23:
> I forked the repository, and created a local addon from that. I updated one of my lamps. I will let you all know if it worked.

It did not work for me.
@tb8851 could you please share what you exactly did?

tb8851, 2020-04-24:
> > I forked the repository, and created a local addon from that. I updated one of my lamps. I will let you all know if it worked.
> 
> It did not work for me.
> @tb8851 could you please share what you exactly did?

Hi @berthavekes2,

I'm running Hass.io.

I added this to the configuration section of ESPHome: -

`esphome_version: fix-api-drop-connection`

Then restarted ESPHome and let it install the new version (keep refreshing the log so you know when the install is complete).

Then I went to the node in the ESPHome WebUI, compiled and uploaded to the device.

I've been running for a few days now without any dropouts I've noticed.

Let me know if I can provide any further info.

berthavekes2, 2020-04-24:
I think there is a small bug in esphome.
It did not help with the first lamp, but the second one is now connected for 12 minutes straight. The weird thing is that their config is the same...

[27 more replies]

closed 2020-05-01 by glmnet