esphome/issues #879: 'Timed out waiting for packet header' on serial uploads

WhistleMaster, 2019-11-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

- ESPHome version 1.14.3
- OS X Mojave 10.14.5 (18F132)
- Python 3.7.3 & pip 19.0.3

**ESP (ESP32/ESP8266, Board/Sonoff):** 

ESP32-PICO-D4 (M5Stick-C)

**Affected component:**

- Upload fails with default baud rate of 460800
- Related to #814 

**Description of problem:**
Using the M5-Stick, the upload fails with the following error message:

`A fatal error occurred: Timed out waiting for packet header`

**Problem-relevant YAML-configuration entries:**
```yaml
n/a
```

**Logs (if applicable):**

```
$ esphome light_gw.yaml upload --upload-port=/dev/cu.usbserial-9952108F45
INFO Reading configuration light_gw.yaml...
WARNING ESP32: Pin 10 (6-11) might already be used by the flash interface. Be warned.
WARNING ESP32: Pin 10 (6-11) might already be used by the flash interface. Be warned.
INFO Running:  platformio run -d light_gw -t upload --upload-port /dev/cu.usbserial-9952108F45
Processing light_gw (board: m5stack-core-esp32; framework: arduino; platform: espressif32@1.11.0)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Retrieving maximum program size .pioenvs/light_gw/firmware.elf
Checking size .pioenvs/light_gw/firmware.elf
DATA:    [=         ]  12.8% (used 41960 bytes from 327680 bytes)
PROGRAM: [=====     ]  51.1% (used 836602 bytes from 1638400 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: /dev/cu.usbserial-9952108F45
Uploading .pioenvs/light_gw/firmware.bin
Serial port /dev/cu.usbserial-9952108F45
Connecting.....
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
MAC: d8:a0:1d:65:f7:ec
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...

A fatal error occurred: Timed out waiting for packet header
*** [upload] Error 2
```

**Additional information and things you've tried:**
If I manually change the setting baud rate to 115200 in file platformio.ini (upload_speed = 115200) of the project, upload succeeds:

```
$ esphome light_gw.yaml upload --upload-port=/dev/cu.usbserial-9952108F45
INFO Reading configuration light_gw.yaml...
WARNING ESP32: Pin 10 (6-11) might already be used by the flash interface. Be warned.
WARNING ESP32: Pin 10 (6-11) might already be used by the flash interface. Be warned.
INFO Running:  platformio run -d light_gw -t upload --upload-port /dev/cu.usbserial-9952108F45
Processing light_gw (board: m5stack-core-esp32; framework: arduino; platform: espressif32@1.11.0)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Retrieving maximum program size .pioenvs/light_gw/firmware.elf
Checking size .pioenvs/light_gw/firmware.elf
DATA:    [=         ]  12.8% (used 41960 bytes from 327680 bytes)
PROGRAM: [=====     ]  51.1% (used 836602 bytes from 1638400 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: /dev/cu.usbserial-9952108F45
Uploading .pioenvs/light_gw/firmware.bin
Serial port /dev/cu.usbserial-9952108F45
Connecting....
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
MAC: d8:a0:1d:65:f7:ec
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 15872 bytes to 10319...

Writing at 0x00001000... (100 %)
Wrote 15872 bytes (10319 compressed) at 0x00001000 in 0.9 seconds (effective 137.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 143...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (143 compressed) at 0x00008000 in 0.0 seconds (effective 1237.2 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 8058.1 kbit/s)...
Hash of data verified.
Compressed 836720 bytes to 467289...

Writing at 0x00010000... (3 %)
Writing at 0x00014000... (6 %)
Writing at 0x00018000... (10 %)
Writing at 0x0001c000... (13 %)
Writing at 0x00020000... (17 %)
Writing at 0x00024000... (20 %)
Writing at 0x00028000... (24 %)
Writing at 0x0002c000... (27 %)
Writing at 0x00030000... (31 %)
Writing at 0x00034000... (34 %)
Writing at 0x00038000... (37 %)
Writing at 0x0003c000... (41 %)
Writing at 0x00040000... (44 %)
Writing at 0x00044000... (48 %)
Writing at 0x00048000... (51 %)
Writing at 0x0004c000... (55 %)
Writing at 0x00050000... (58 %)
Writing at 0x00054000... (62 %)
Writing at 0x00058000... (65 %)
Writing at 0x0005c000... (68 %)
Writing at 0x00060000... (72 %)
Writing at 0x00064000... (75 %)
Writing at 0x00068000... (79 %)
Writing at 0x0006c000... (82 %)
Writing at 0x00070000... (86 %)
Writing at 0x00074000... (89 %)
Writing at 0x00078000... (93 %)
Writing at 0x0007c000... (96 %)
Writing at 0x00080000... (100 %)
Wrote 836720 bytes (467289 compressed) at 0x00010000 in 41.7 seconds (effective 160.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
======================================================================================================== [SUCCESS] Took 47.01 seconds ========================================================================================================
INFO Successfully uploaded program.
```

brandond, 2019-11-25:
Fixed in dev. It will automatically fall back to a lower rate.

OttoWinter, 2019-11-26:
@brandond That was only for ESP8266 uploads - I forgot to do the same for ESP32.

closed 2020-08-07 by stale[bot]