esphome/issues #880: Xiaomi CGG1 BLE Sensor not found

tomlut, 2019-11-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio NUC image
**ESP (ESP32/ESP8266, Board/Sonoff):**

  platform: ESP32
  board: mhetesp32minikit

**Affected component:**

https://esphome.io/components/sensor/xiaomi_cgg1.html

**Description of problem:**
The Bluetooth tracker seems unable to find Xiaomi CGG1 BLE Sensors. It has no trouble with the [LYWSD02](https://esphome.io/components/sensor/xiaomi_lywsd02.html) or [HHCCJCY01](https://esphome.io/components/sensor/xiaomi_hhccjcy01.html) sensors. I have tried two CGG1s. I tried updating the firmware on one CGG1 but it made no difference. I have tried pressing the button on the back of the CGG1 to speed up the process but still get no mac address for these types of device.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: bt_gw_ensuite
  platform: ESP32
  board: mhetesp32minikit

wifi:
  ssid: 'WAPLO'
  password: !secret wifi_pwd
  manual_ip:
    static_ip: 10.1.1.67
    gateway: 10.1.1.1
    subnet: 255.255.255.0

api:
  password: !secret api_password

logger:
  # level: WARN

ota:
  password: !secret esp_pwd

esp32_ble_tracker:

```

**Logs (if applicable):**

```
INFO Successfully connected to 10.1.1.67
[12:38:30][I][app:100]: ESPHome version 1.14.3 compiled on Nov 26 2019, 12:37:46
[12:38:30][C][wifi:415]: WiFi:
[12:38:30][C][wifi:283]:   SSID: 

[redacted]
[12:38:30][C][wifi:284]:   IP Address: 10.1.1.67
[12:38:30][C][wifi:286]:   BSSID: 

[redacted]
[12:38:30][C][wifi:287]:   Hostname: 'bt_gw_ensuite'
[12:38:30][C][wifi:291]:   Signal strength: -37 dB ▂▄▆█
[12:38:30][C][wifi:295]:   Channel: 1
[12:38:30][C][wifi:296]:   Subnet: 255.255.255.0
[12:38:30][C][wifi:297]:   Gateway: 10.1.1.1
[12:38:30][C][wifi:298]:   DNS1: 0.0.0.0
[12:38:30][C][wifi:299]:   DNS2: 0.0.0.0
[12:38:30][C][logger:175]: Logger:
[12:38:30][C][logger:176]:   Level: DEBUG
[12:38:30][C][logger:177]:   Log Baud Rate: 115200
[12:38:30][C][logger:178]:   Hardware UART: UART0
[12:38:30][C][esp32_ble_tracker:442]: BLE Tracker:
[12:38:30][C][esp32_ble_tracker:443]:   Scan Duration: 300 s
[12:38:30][C][esp32_ble_tracker:444]:   Scan Interval: 320.0 ms
[... 111 lines cut ...]
[12:46:35][D][xiaomi_ble:158]: Got Xiaomi LYWSD02 (E7:2E:00:D2:E0:BF):
[12:46:36][D][xiaomi_ble:161]:   Temperature: 20.0°C
[12:46:50][D][xiaomi_ble:158]: Got Xiaomi LYWSD02 (E7:2E:00:D2:E0:BF):
[12:46:50][D][xiaomi_ble:161]:   Temperature: 20.0°C
[12:46:55][D][xiaomi_ble:158]: Got Xiaomi HHCCJCY01 (C4:7C:8D:6B:12:1F):
[12:46:55][D][xiaomi_ble:173]:   Illuminance: 0lx
[12:46:56][D][xiaomi_ble:158]: Got Xiaomi HHCCJCY01 (C4:7C:8D:6B:12:1F):
[12:46:56][D][xiaomi_ble:176]:   Moisture: 45%
[12:46:57][D][xiaomi_ble:158]: Got Xiaomi LYWSD02 (E7:2E:00:D2:E0:BF):
[12:46:57][D][xiaomi_ble:164]:   Humidity: 54.0%
[12:47:36][D][xiaomi_ble:158]: Got Xiaomi HHCCJCY01 (C4:7C:8D:6B:12:1F):
[12:47:36][D][xiaomi_ble:176]:   Moisture: 45%
[12:47:40][D][xiaomi_ble:158]: Got Xiaomi LYWSD02 (E7:2E:00:D2:E0:BF):
[12:47:40][D][xiaomi_ble:161]:   Temperature: 20.1°C
[12:47:54][D][xiaomi_ble:158]: Got Xiaomi LYWSD02 (E7:2E:00:D2:E0:BF):
[12:47:54][D][xiaomi_ble:161]:   Temperature: 20.0°C
[12:47:59][D][xiaomi_ble:158]: Got Xiaomi HHCCJCY01 (C4:7C:8D:6B:12:1F):
[12:47:59][D][xiaomi_ble:161]:   Temperature: 18.1°C
[12:48:01][D][xiaomi_ble:158]: Got Xiaomi LYWSD02 (E7:2E:00:D2:E0:BF):
[12:48:01][D][xiaomi_ble:164]:   Humidity: 54.0%

```

**Additional information and things you've tried:**
I believe the PUBLIC and RANDOM BT devices are my iPhone.

OttoWinter, 2019-11-26:
Related to https://github.com/esphome/issues/issues/735

Try with these ble scanner settings:

```yaml
esp32_ble_tracker:
  scan_parameters:
    window: 200ms
```

tomlut, 2019-11-26:
Thank you Otto. That fixed it.

closed 2019-11-26 by tomlut