esphome/issues #881: ESPHome 1.14.3 shows offline, but can still connect (ESP8266)

fribse, 2019-11-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome 1.14.3

**Entity is shown as offline, but it's not**

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  device_name: refridgerator
  # Higher value gives lower watt readout
  current_res: '0.00290'
  # Lower value gives lower voltage readout
  voltage_div: '940'

esphome:
  name: '${device_name}'
  platform: ESP8266
  board: esp8285
  on_boot:
    then:
      - switch.turn_on: relay
          
wifi:
  ssid: SSID
  password: PW
  manual_ip:
    static_ip: 10.11.12.76
    gateway: 10.11.12.1
    subnet: 255.255.255.0
  use_address: 10.11.12.76
    
logger:
  baud_rate: 0

api:

ota:
  password: "PW"

time:
  - platform: homeassistant
    id: homeassistant_time

text_sensor:
  - platform: version
    name: "${device_name} ESPHome Version"
  - platform: wifi_info
    ip_address:
      name: "${device_name} ip"
    ssid:
      name: "${device_name} ssid"

########################
# Should never turn of #
# button disabled      #
########################

#binary_sensor:
#  - platform: gpio
#    id: button
#    internal: true
#    pin:
#      number: GPIO13
#    on_press:
#      - switch.toggle: relay

output:
  - platform: gpio
    pin: GPIO00
    inverted: true
    id: led

switch:
  - platform: gpio
    pin: GPIO15
    id: relay
    name: '${device_name} relay'
    on_turn_on:
      - output.turn_on: led
    on_turn_off:
      - output.turn_off: led

status_led:
  pin:
    number: GPIO02
    inverted: true

sensor:
  - platform: wifi_signal
    name: '${device_name} wifi signal'
    update_interval: 60s
    accuracy_decimals: 0
  - platform: uptime
    name: '${device_name} uptime'
    unit_of_measurement: days
    update_interval: 300s
    accuracy_decimals: 1
    filters:
      - multiply: 0.000011574
  - platform: hlw8012
    sel_pin:
      number: GPIO12
      inverted: true
    cf_pin: GPIO05
    cf1_pin: GPIO14
    current_resistor: ${current_res}
    voltage_divider: ${voltage_div}
    current:
      name: '${device_name} current'
      unit_of_measurement: 'A'
      accuracy_decimals: 3
      icon: mdi:flash-outline
    voltage:
      name: '${device_name} voltage'
      unit_of_measurement: 'V'
      icon: mdi:flash-outline
    power:
      name: '${device_name} power'
      unit_of_measurement: 'kW'
      id: power
      accuracy_decimals: 5
      filters:
        - multiply: 0.001
      icon: mdi:flash-outline
    change_mode_every: 4
    update_interval: 10s
  - platform: total_daily_energy
    name: '${device_name} daily energy'
    power_id: power
    unit_of_measurement: kWh

```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/refridgerator.yaml...
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 29 March 02:00:00 to 25 October 03:00:00
INFO Starting log output from 10.11.12.76 using esphome API
INFO Connecting to 10.11.12.76:6053 (10.11.12.76)
INFO Successfully connected to 10.11.12.76
[07:02:09][I][app:100]: ESPHome version 1.14.3 compiled on Nov 25 2019, 22:45:24
[07:02:09][C][status_led:019]: Status LED:
[07:02:09][C][status_led:020]:   Pin: GPIO2 (Mode: OUTPUT, INVERTED)
[07:02:09][C][wifi:415]: WiFi:
[07:02:09][C][wifi:283]:   SSID: [redacted]
[07:02:09][C][wifi:284]:   IP Address: 10.11.12.76
[07:02:09][C][wifi:286]:   BSSID: [redacted]
[07:02:09][C][wifi:287]:   Hostname: 'refridgerator'
[07:02:09][C][wifi:291]:   Signal strength: -73 dB ▂▄▆█
[07:02:09][C][wifi:295]:   Channel: 1
[07:02:09][C][wifi:296]:   Subnet: 255.255.255.0
[07:02:09][C][wifi:297]:   Gateway: 10.11.12.1
[07:02:09][C][wifi:298]:   DNS1: (IP unset)
[07:02:09][C][wifi:299]:   DNS2: (IP unset)
[07:02:09][C][switch.gpio:042]: GPIO Switch 'refridgerator relay'
[07:02:09][C][switch.gpio:043]:   Pin: GPIO15 (Mode: OUTPUT)
[07:02:09][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[07:02:09][C][uptime.sensor:030]: Uptime Sensor 'refridgerator uptime'
[07:02:09][C][uptime.sensor:030]:   Unit of Measurement: 'days'
[07:02:09][C][uptime.sensor:030]:   Accuracy Decimals: 1
[07:02:09][C][uptime.sensor:030]:   Icon: 'mdi:timer'
[07:02:09][C][gpio.output:010]: GPIO Binary Output:
[07:02:09][C][gpio.output:011]:   Pin: GPIO0 (Mode: OUTPUT)
[07:02:09][C][gpio.output:012]:   Inverted: YES
[07:02:09][C][logger:175]: Logger:
[07:02:09][C][logger:176]:   Level: DEBUG
[07:02:09][C][logger:177]:   Log Baud Rate: 0
[07:02:09][C][logger:178]:   Hardware UART: UART0
[07:02:09][C][hlw8012:020]: HLW8012:
[07:02:09][C][hlw8012:021]:   SEL Pin: GPIO12 (Mode: OUTPUT, INVERTED)
[07:02:09][C][hlw8012:022]:   CF Pin: GPIO5 (Mode: INPUT_PULLUP)
[07:02:09][C][hlw8012:023]:   CF1 Pin: GPIO14 (Mode: INPUT_PULLUP)
[07:02:09][C][hlw8012:024]:   Change measurement mode every 4
[07:02:09][C][hlw8012:025]:   Current resistor: 2.9 mΩ
[07:02:09][C][hlw8012:026]:   Voltage Divider: 940.0
[07:02:09][C][hlw8012:027]:   Update Interval: 10.0s
[07:02:09][C][hlw8012:028]:   Voltage 'refridgerator voltage'
[07:02:09][C][hlw8012:028]:     Unit of Measurement: 'V'
[07:02:09][C][hlw8012:028]:     Accuracy Decimals: 1
[07:02:09][C][hlw8012:028]:     Icon: 'mdi:flash-outline'
[07:02:09][C][hlw8012:029]:   Current 'refridgerator current'
[07:02:09][C][hlw8012:029]:     Unit of Measurement: 'A'
[07:02:10][C][hlw8012:029]:     Accuracy Decimals: 3
[07:02:10][C][hlw8012:029]:     Icon: 'mdi:flash-outline'
[07:02:10](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[07:02:10](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[07:02:10](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[07:02:10](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[07:02:10][C][total_daily_energy:022]: Total Daily Energy 'refridgerator daily energy'
[07:02:10][C][total_daily_energy:022]:   Unit of Measurement: 'kWh'
[07:02:10][C][total_daily_energy:022]:   Accuracy Decimals: 7
[07:02:10][C][total_daily_energy:022]:   Icon: 'mdi:flash-outline'
[07:02:10][C][homeassistant.time:010]: Home Assistant Time:
[07:02:10][C][homeassistant.time:011]:   Timezone: 'CET-1CEST-2,M3.5.0/2,M10.4.0/3'
[07:02:10][C][version.text_sensor:015]: Version Text Sensor 'refridgerator ESPHome Version'
[07:02:10][C][version.text_sensor:015]:   Icon: 'mdi:new-box'
[07:02:10][C][ota:029]: Over-The-Air Updates:
[07:02:10][C][ota:030]:   Address: 10.11.12.76:8266
[07:02:10][C][ota:032]:   Using Password.
[07:02:10][C][api:095]: API Server:
[07:02:10][C][api:096]:   Address: 10.11.12.76:6053
[07:02:10][C][wifi_info:009]: WifiInfo IPAddress 'refridgerator ip'
[07:02:10][C][wifi_signal.sensor:009]: WiFi Signal 'refridgerator wifi signal'
[07:02:10][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[07:02:10][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[07:02:10][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[07:02:10][C][wifi_info:010]: WifiInfo SSID 'refridgerator ssid'
[07:02:10][D][hlw8012:071]: Got power=125.1W, voltage=228.2V
[07:02:10][D][sensor:092]: 'refridgerator voltage': Sending state 228.24896 V with 1 decimals of accuracy
[07:02:10][D][sensor:092]: 'refridgerator power': Sending state 0.12507 kW with 5 decimals of accuracy
[07:02:10][D][sensor:092]: 'refridgerator daily energy': Sending state 0.64085 kWh with 7 decimals of accuracy
```

**It is shown as offline, but when pressing show logs, it appears**
https://www.screencast.com/t/nbaFJAsFgi

It does appear in mDNS.

OttoWinter, 2019-11-26:
Unfortunately, I could not view the screencast as it requires flash player, which I can't install.

As for the issue, if HA can't connect to it this will require logs from the Home Assistant side. Please enable verbose logs in HA and post those logs here (you can filter by `esphome`)

fribse, 2019-11-27:
hi @OttoWinter , ahh, ok, sorry about that, the cast in itself is not interesting, I've illustrated it on a few screenshots here:
[image]
Note the 'adrianentertainent' is shown as offline, but opening logs shows:
[image]
It is only shown as offline in ESPHome Dashboard for 1.14.3, HA has it's sensors as online.

fribse, 2020-02-09:
I've replaced the AP's, and I've enabled IGMP Snooping, and IGMPv3, and also removed multicast limitations. This makes it show up stable on the HA. So it's a circumvention

closed 2020-02-09 by fribse