esphome/issues #885: Xiaomi LYWSDCGQ sensor not found (dev-branch)

kanttti, 2019-11-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io add-on
**ESP (ESP32/ESP8266, Board/Sonoff):**

platform: ESP32
board: Generic Chinese Devkit V1 ESP32
**Affected component:**

https://esphome.io/components/sensor/xiaomi_lywsdcgq.html

**Description of problem:**
The Bluetooth tracker is unable to find Xiaomi LYWSDCGQ in ESPHome dev-build.

I installed dev-version of ESPHome Hass.io add-on to try the new RuuviTag sensor. The RuuviTag sensor itself seems to be working perfectly, but the Xiaomi LYWSDCGQ sensors do not popup anymore.  However the Xiaomi hhccjcy01 (miFlora) seems to be working, so this is not related to the whole xiaomi_ble.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ruuvitagreader
  platform: ESP32
  board: esp32doit-devkit-v1

wifi:
  ssid: "xxx"
  password: "xxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "xxx"
    password: "xxx"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

esp32_ble_tracker:

#sensor:
#- platform: ruuvitag
#  mac_address: F9:39:82:A6:9D:0F
#  humidity:
#    name: "RuuviTag Humidity"
#  temperature:
#    name: "RuuviTag Temperature"
#  pressure:
#    name: "RuuviTag Pressure"
#  acceleration:
#    name: "RuuviTag Acceleration"
#  acceleration_x:
#    name: "RuuviTag Acceleration X"
#  acceleration_y:
#    name: "RuuviTag Acceleration Y"
#  acceleration_z:
#    name: "RuuviTag Acceleration Z"
#  battery_voltage:
#    name: "RuuviTag Battery Voltage"
#  tx_power:
#    name: "RuuviTag TX Power"
#  movement_counter:
#    name: "RuuviTag Movement Counter"
#  measurement_sequence_number:
#    name: "RuuviTag Measurement Sequence Number"
#- platform: xiaomi_lywsdcgq
[... 374 lines cut ...]
[21:13:28][D][xiaomi_ble:164]:   Humidity: 38.7%
[21:13:29][D][xiaomi_ble:158]: Got Xiaomi LYWSDCGQ (58:2D:34:36:B8:FF):
[21:13:29][D][xiaomi_ble:161]:   Temperature: 28.0°C
[21:13:29][D][xiaomi_ble:164]:   Humidity: 27.1%
[21:13:37][D][xiaomi_ble:158]: Got Xiaomi HHCCJCY01 (C4:7C:8D:65:61:A6):
[21:13:37][D][xiaomi_ble:161]:   Temperature: 19.2°C
[21:14:01][D][xiaomi_ble:158]: Got Xiaomi HHCCJCY01 (C4:7C:8D:65:61:A6):
[21:14:01][D][xiaomi_ble:176]:   Moisture: 6%
[21:14:17][D][xiaomi_ble:158]: Got Xiaomi HHCCJCY01 (C4:7C:8D:65:61:A6):
[21:14:17][D][xiaomi_ble:161]:   Temperature: 19.3°C
[21:14:26][D][xiaomi_ble:158]: Got Xiaomi HHCCJCY01 (C4:7C:8D:65:61:A6):
[21:14:26][D][xiaomi_ble:173]:   Illuminance: 37lx
[21:14:28][D][xiaomi_ble:158]: Got Xiaomi LYWSDCGQ (58:2D:34:35:D2:8E):
[21:14:28][D][xiaomi_ble:161]:   Temperature: 21.2°C
[21:14:29][D][xiaomi_ble:158]: Got Xiaomi LYWSDCGQ (58:2D:34:36:B8:FF):
[21:14:29][D][xiaomi_ble:161]:   Temperature: 25.6°C
[21:14:29][D][xiaomi_ble:164]:   Humidity: 28.4%
```

**Additional information and things you've tried:**
I've been using version 1.14.3 before with no problems. And I have crosschecked a couple of times between v1.14.3 and dev-branch that it really works differently. 

I've also tried with both RuuviTag and Xiaomi sensors defined in YAML-file and without defining it. however, I assume this shouldn't have an effect as the print in the log will be printed anyway?

I also tried to change the scan_parameters: in esp32_ble_tracker: with no effect.

Alex9779, 2019-11-28:
See https://github.com/esphome/esphome/commit/ad76709d00e757907e3241f43e1cf9626de8032b

kanttti, 2019-11-28:
Thanks @Alex9779.
With Active: false in ble_tracker scan_parameters the sensors can be found again.

In my understanding the ble_scan_type = active (true), means that everytime the BLE tracker receives a message from any MAC it will send a reply back to that device. This causes some BLE devices to send an additional message back to the BLE tracker.
So this means that both of the data are correct, but the data from LYWSDCGQ should be parsed differently depending on which of Active/Passive (true/false) is set. But still I do not understand why the UUID is not found as it is in the 6th and 7th byte in both cases. But I did not check the code for the data parsing so there might be some obvious reason for this.

This should be fixed in the code of course, but my question is why ble_scan_type is defaulted to true?
What are the devices that requires the scan to be active?

I assume that most of the temperature etc. sensors like Xiaomis, Ruuvitags, Cleargrasses, etc... should be only listened in passive mode to increase the battery life? Pls. correct me if I'm wrong.

Is the active mode needed for using this as an actual BLE_Tracker? Do some beacons/smartphones/watches need the scan to be active to populate some needed additional data? Are there cases where the MAC address is not enough for BLE_Tracking?

Alex9779, 2019-11-28:
Regarding active I have no idea too...
To parsing I just can say that the component is checking the length of the data which is totally different if using active after that change so parsing does not even start because the data is not considered valid initially...

Alex9779, 2019-12-09:
I think this got fixed with esphome/esphome#904

kanttti, 2019-12-09:
> I think this got fixed with [esphome/esphome#904](https://github.com/esphome/esphome/pull/904)

Seems to be working from dev head now. Great work. 👍

closed 2019-12-09 by kanttti