esphome/issues #886: YAML anchors

Edgesystems, 2019-11-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip install esphome
arch linux

**ESP (ESP32/ESP8266, Board/Sonoff):**

shelly 2.5 
esphome:
  platform: ESP8266
  board: esp01_1m

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml

binary_sensor:
  - &button1
    platform: gpio
    name: "button 1"
    pin:
      number: GPIO2

switch:
  <<: *button1
  name: "override"
```

Output:
> ERROR Error while reading config: Invalid YAML syntax. Please see YAML syntax reference or use an online YAML syntax validator:
> 
> duplicate key: "name"

Expected result: the latter should override the formet.

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

brandond, 2019-11-28:
I don't believe the esphome YAML loader supports key merge.

edge90, 2019-11-28:
The normal python yaml module does. This is caused by yaml_util.py

closed 2019-12-04 by OttoWinter