esphome/issues #887: climate bangbang, no way to change the units shown for use with humidity?

tiswi, 2019-11-27:
Hi,

For the bangbang climate controller there is the Note: 
"While this platform uses the term temperature everywhere, it can also be used for other values. For example, controlling humidity is also possible with this platform."

There is no way to change the units used accordingly? Like for example humidity in % or temperature in °F

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

brandond, 2019-11-28:
Yeah, furthermore the schema validation for the setpoints seems to require a temperature unit (C/K/F). While that doesn't prevent you from controlling (for example) humidity, it might be a bit confusing to have to pretend that you're managing a 0-100 degree temperature range when it's really something else.

OttoWinter, 2019-11-29:
That is a limitation of Home Assistant - the "climate" model in HA only allows temperature setting - there's no way to select a generic unit of measurement.

For °F, you need to set the HA unit system - then the frontend will automatically convert the values to fahrenheit (at least it should).

I understand it's not very intuitive, but I don't see any good way to handle this within the system we work in.

jaysalazar, 2020-06-11:
I too am running in to the same problem with Humidity Control. Not only is it hard to pretend you are managing humidity with degree values. I use °F system so the ESP32 sends info in °C in turn HA converts the values on the front-end to °F. It seems to work for Temperature control albeit I have to input my temp values in °C which I am not used too. I have to use a converter online (annoying).

So then when it reads the the Humidity percentage it converts that to °F also, so 40% humidity value shows up as 104 °F on the card. 

I really wanted to use this however I don't think its going to work out until humidity control is worked out :(

jaysalazar, 2020-06-11:
> Hi,
>
> For the bangbang climate controller there is the Note:
> "While this platform uses the term temperature everywhere, it can also be used for other values. For example, controlling humidity is also [...]

What did you end up doing for a Humidity Thermostat?

tiswi, 2020-06-11:
It's not a big issue for me, for me its just a "set and forget" thermostat, I don't have to look at it often enough to bother.

closed 2020-12-05 by stale[bot]