esphome/issues #888: Pulse counter not counting pulses from PWM fans

BarBaar44, 2019-11-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hassio running ESPhome beta (1.14.3) on an Odroid 

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32

**Affected component:**
Pulse counter -> https://esphome.io/components/sensor/pulse_counter.html?highlight=pulse_counter

**Description of problem:**
The pulse counter is not counting the pulses from a PWM fan.

**Problem-relevant YAML-configuration entries:**
```yaml
  - platform: pulse_counter
    pin:
      number: 27
      inverted: false
      mode: INPUT
    name: "DBE Kantoor Speed"
    unit_of_measurement: 'RPM'
    filters:
      - multiply: 0.5
    count_mode:
      rising_edge: DISABLE
      falling_edge: DISABLE
    update_interval: 30s
    internal_filter: 1us

```

**Logs (if applicable):**
[21:49:20][C][pulse_counter:147]: Pulse Counter 'DBE Kantoor Speed'
[21:49:20][C][pulse_counter:147]:   Unit of Measurement: 'RPM'
[21:49:20][C][pulse_counter:147]:   Accuracy Decimals: 2
[21:49:20][C][pulse_counter:147]:   Icon: 'mdi:pulse'
[21:49:20][C][pulse_counter:148]:   Pin: GPIO27 (Mode: INPUT)
[21:49:20][C][pulse_counter:149]:   Rising Edge: DISABLE
[21:49:20][C][pulse_counter:150]:   Falling Edge: DISABLE
[21:49:20][C][pulse_counter:151]:   Filtering pulses shorter than 1 µs
[21:49:20][C][pulse_counter:152]:   Update Interval: 30.0s
``
```
[21:29:48][D][pulse_counter:159]: 'DBE Kantoor Speed': Retrieved counter: 0.00 pulses/min
[21:29:48][D][sensor:092]: 'DBE Kantoor Speed': Sending state 0.00000 RPM with 2 decimals of accuracy

```

**Additional information and things you've tried:**

Tried with the following fans:
- Artic F9 PWN
- Intel intel e97379

Tried the following config changes:
- Tried different pins on the ESP board.
- Tried different pins
- Tried setting the mode (INPUT or INPUT_PULLUP)
- Lowered the internal filter
- Check that the fan is actually running (pulled the PWN connector)

The pulse counter sometimes does initially work! (which only makes this behavior stranger)

OttoWinter, 2019-11-28:
```yaml
count_mode:
  rising_edge: DISABLE
  falling_edge: DISABLE
```

You have disabled counting on both rising and falling edge - this means the counter will never update. You have to increase/decrease the value on at least one of them.

BarBaar44, 2019-11-28:
That easy. Seems to be working, thanks!

closed 2019-11-28 by BarBaar44