esphome/issues #889: AttributeError: 'SConsEnvironment'

crood58, 2019-11-29:
[image]

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Running ESPHome in a docker container on a Raspberry Pi 4

**ESP (ESP32/ESP8266, Board/Sonoff):**

Generic ESP8266

**Affected component:**

N/A

**Description of problem:**
Keep getting a SConsEnvironment error when trying to compile. Refer to the attached screenshot

**Problem-relevant YAML-configuration entries:**
```yaml
# MASTER BATHROOM MAIN LIGHT
# ##################################
# Martin Jerry Light Switch MJ-S01
# ##################################
# D2 GPIO4: red led (inverted)
# D1 GPIO5: blue led (inverted)
# D6 GPIO12: relay
# D7 GPIO13: button (inverted, input_pullup)
# ##################################

substitutions:
  # https://esphome.io/guides/configuration-types.html#substitutions
  device_name: master_bath_main_light
  friendly_name: Master Bathroom Main Light 
  mqtt_topic_prefix: home/master/bathroom/main_light

  
esphome:
  name: ${device_name}
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid_upstairs
  password: !secret wifi_pass_upstairs 

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${friendly_name}
    password: !secret esphome_ap_password

web_server:
#   # https://esphome.io/components/web_server.html
  port: 80
  auth:
    username: admin
    password: !secret esphome_webserver_password

logger:
  # https://esphome.io/components/logger

ota:
  password: !secret esphome_ota_password
  # https://esphome.io/components/ota
  
# Enable MQTT
# https://esphome.io/components/mqtt.html
mqtt:
  broker: !secret mqtt_broker 
  username: !secret mqtt_user 
  password: !secret mqtt_password
  client_id: ${device_name}
  topic_prefix: ${mqtt_topic_prefix}
  on_message:
    - topic: command
      payload: "ON"
      then:
        - switch.turn_on: relay
    - topic: command
      payload: "OFF" 
      then:
        - switch.turn_off: relay
    
  
status_led: 
  # https://esphome.io/components/status_led
  pin:
    number: GPIO4   # Red LED
    inverted: True

binary_sensor:
  - platform: gpio
    # https://esphome.io/components/switch/gpio
    name: ${friendly_name} Button
    id: button
    pin:
      number: GPIO13
      mode: INPUT_PULLUP
      inverted: True
    internal: True
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    # https://esphome.io/components/switch/gpio
    name: ${friendly_name} Blue LED
    id: blue_led
    pin:
      number: GPIO5
      inverted: True
    internal: True

  - platform: gpio
    # https://esphome.io/components/switch/gpio
    name: ${friendly_name} Relay
    id: relay
    pin:
      number: GPIO12
    # Tie Blue LED operation to relay
    on_turn_on:
      - switch.turn_off: blue_led
      - mqtt.publish:
          topic: state
          payload: "ON"

    on_turn_off:
      - switch.turn_on: blue_led
      - mqtt.publish:
          topic: state
          payload: "OFF"

```

**Logs (if applicable):**

```
No Logs available from dashboard related to this issue

```

**Additional information and things you've tried:**

I have deleted the docker container and image along with the my volumes and recreated the container, but the error keeps occurring every-time.

OttoWinter, 2019-11-30:
Please post how you invoke ESPHome with docker, it looks like there's something seriously messed up with your install.

Anyway, this is an issue in platformio (or the docker build), see also https://community.platformio.org/t/solved-cannot-do-nearly-anything-sconsenvironment-object-has-no-attribute-getextrascripts-error/3797

It partially looks like you don't use the latest version of esphome, some lines like "You can ignore this message ..." are actively filtered with recent versions of ESPHome and should not appear in the logs.

crood58, 2019-12-01:
Here is my docker compose for the container. 

```
  esphome:
    container_name: esphome
    restart: unless-stopped
    image: esphome/esphome-armv7:1.14.3
    volumes:
      - /mnt/dietpi_userdata/home-automation/ESPHome:/config:rw
    ports:
      - 6052:6052
      - 6123:6123
    user: 0:0
    environment:
      - HOME="/"
    command: config/ dashboard 
```

OttoWinter, 2019-12-03:
The `user: 0:0` and `environment: [HOME="/"]` mapping could be the issue (especially remapping HOME, since platformio saves files in ~/.platformio)

Please explain why you set those and try without them.

crood58, 2019-12-04:
That was the issue was user: 0:0 and environment: [HOME="/"] mapping . I tested it quick and got the program to compile. I followed someone elses docker compose file and thought we had to use them. That was my fault and I do appreciate your help with this. 

Chris

closed 2019-12-04 by crood58