esphome/issues #890: AMS AS3935 Franklin Lightning Sensor | error Disturber was detected - try increasing the spike rejection value!

josephdyland, 2019-12-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io supervisor V 192
System	Ubuntu 18.04.3 LTS
Home Assistant 0.102.2.
ESPhome 1.14.3
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 D1 Mini 3.1
**Affected component:**

https://esphome.io/components/sensor/as3935.html
**Description of problem:**
as3935 Sensor component outputs thousands of message per minute;
[14:48:15][I][as3935:041]: Disturber was detected - try increasing the spike rejection value! 
14:48:22][W][i2c:073]: Received NACK on transmit of data for address 0x03
[14:48:22][W][as3935_i2c:029]: Read failed!
[14:48:15][I][as3935:041]: Disturber was detected - try increasing the spike rejection value!
**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: node_sensor1
  platform: ESP8266
  board: d1_mini
  includes:
    - sht30_custom_sensor.h
  libraries:
    - "WEMOS SHT3x"

wifi:
  ssid: 'redact'
  password: 'redact'
  fast_connect: true
      # ...
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.1.122
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1
    dns2: 192.168.1.1

# Enable logging
logger:
  level: VERBOSE
i2c:
  scan: True
as3935_i2c:
  irq_pin: GPIO13
#  noise_level: 2
  spike_rejection: 5
binary_sensor:
- platform: as3935
  name: "Storm Alert"
sensor:
- platform: as3935
  lightning_energy:
    name: "Lightning Energy"
  distance:
    name: "Distance Storm"

```

**Logs (if applicable):**

```
Level: Debug
[14:48:15][I][as3935:041]: Disturber was detected - try increasing the spike rejection value!
[14:48:15][I][as3935:041]: Disturber was detected - try increasing the spike rejection value!
[14:48:15][I][as3935:041]: Disturber was detected - try increasing the spike rejection value!
14:48:22][W][i2c:073]: Received NACK on transmit of data for address 0x03
[14:48:22][W][as3935_i2c:029]: Read failed!
[14:48:15][I][as3935:041]: Disturber was detected - try increasing the spike rejection value!

Level: Very-Verbose
[17:03:51][I][as3935:041]: Disturber was detected - try increasing the spike rejection value!
[17:03:51][V][as3935:175]: Calling read_interrupt_register_
[17:03:51][VV][i2c:056]: Beginning Transmission to 0x03:
[17:03:51][VV][i2c:093]:     Writing 0b00000011 (0x03)
[17:03:51][VV][i2c:061]:     Transmission ended. Status code: 0x00
[17:03:51][VV][i2c:083]: Requesting 1 bytes from 0x03:
[17:03:51][VV][i2c:113]:     Received 0b00100100 (0x24)

```

**Additional information and things you've tried:**
Sensor Model: GY-AS3935
https://www.aliexpress.com/item/32974749991.html?spm=a2g0s.9042311.0.0.b10e4c4dioYj6u

Connected via I2C
V > 3.3V
G > Ground
CL > D1
DA > D2
IRQ >GPIO13|D7
SI > Vdd (for I2C)

OttoWinter, 2019-12-03:
Hmm, not sure what it is.

The warning is just what the sensor itself reports - it has the error code for NOISE in the interrupt register. That error is generated directly from the sensor so the issue would be in the sensor itself.

josephdyland, 2019-12-03:
Unable to resolve.

closed 2019-12-03 by josephdyland

minsuke, 2019-12-06:
> Unable to resolve.

I got a same issue. and I have exactly same sensor you got 

I want to solve this problem. 

If I find something I'll share the solution.

josephdyland, 2019-12-06:
I did find that if I change indoor: false,   All of the disturber was detected message do disappear.  And things appear stable.

-No lightning around so I have not been able to actually test.  
- I was hoping a Piezo ignitor / bbq lighter would trip it off but I am unable to simulate 'lightnihg' to check if the sensor is working properly.

minsuke, 2019-12-07:
> I did find that if I change indoor: false, All of the disturber was detected message do disappear. And things appear stable.
>
> -No lightning around so I have not been able to actually test.
>
> * I was [...]

thank you for your information

I just checked then I can see Distance Storm is unknown, and Lightning Energy is unknown 

I think my location has no lighening. If it happened, I'll make it notice to me. 

thx again.

OttoWinter, 2019-12-07:
If it does work please post it here so that we can update the docs.