esphome/issues #891: Sonoff dual R2 fails to boot with GPIO08 button

Elkropac, 2019-12-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Esphome flashing

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff dual R2 version 1.4

**Affected component:**

boot

**Description of problem:**
When i try to add second button on GPIO08, board fails to boot. Log bellow.
This board is different from version 1.0 described on web. Physical button is connected to GPIO0 and button0 and button1 on addon header are connected separatetly

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
substitutions:
  dev_name: sonoff_dualr2_001

esphome:
  name: ${dev_name}
  platform: ESP8266
  board: esp01_1m

wifi:

# Enable logging
logger:
  level: DEBUG
  
# Enable Home Assistant API
api:
  
ota:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO10
      mode: INPUT_PULLUP
      inverted: True
    name: "${dev_name}_button"
    on_press:
      - switch.toggle: relay
      - switch.toggle: relay2
  - platform: gpio
    pin:
      number: GPIO08
      mode: INPUT_PULLUP
      inverted: True
    name: "${dev_name}_button0"
    on_press:
      - switch.toggle: relay
  - platform: gpio
    pin:
      number: GPIO09
      mode: INPUT_PULLUP
      inverted: True
    name: "${dev_name}_button1"
    on_press:
      - switch.toggle: relay2

switch:
  - platform: gpio
    name: "${dev_name}_relay"
    pin: GPIO12
    id: relay
  - platform: gpio
    name: "${dev_name}_relay2"
    pin: GPIO05
    id: relay2
  - platform: shutdown
    name: "${dev_name}_shutdown"
  - platform: restart
    name: "${dev_name}_restart"

status_led:
  pin:
    number: GPIO13
    inverted: yes

sensor:
  - platform: wifi_signal
    name: "${dev_name}_wifi_signal"
    update_interval: 30s
  - platform: uptime
    name: "${dev_name}_uptime"
    update_interval: 120s
  
text_sensor:
  - platform: version
    name: "${dev_name}_version"
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

[18:57:47]
[18:57:47] ets Jan  8 2013,rst cause:4, boot mode:(3,7)
[18:57:47]
[18:57:47]wdt reset
[18:57:47]load 0x4010f000, len 1384, room 16 
[18:57:47]tail 8
[18:57:47]chksum 0x2d
[18:57:47]csum 0x2d
[18:57:47]v8b899c12
[18:57:47]~ld
[18:57:47][I][logger:156]: Log initialized
[18:57:47][C][status_led:014]: Setting up Status LED...
[18:57:47][C][ota:364]: There have been 9 suspected unsuccessful boot attempts.
[18:57:47][I][app:028]: Running through setup()...
[18:57:47][D][binary_sensor:034]: 'sonoff_dualr2_001_button': Sending initial state OFF
[18:57:47][D][binary_sensor:034]: 'sonoff_dualr2_001_button0': Sending initial state OFF
[18:57:47][D][binary_sensor:034]: 'sonoff_dualr2_001_button1': Sending initial state OFF
```

**Additional information and things you've tried:**

Elkropac, 2019-12-01:
I'm using git version from dev branch

brandond, 2019-12-01:
Does it fail to boot when you run the firmware with the button configured, or when. You physically connect the button to gpio8?

Can you connect logs via serial?

Elkropac, 2019-12-01:
I don't have anything connected to header "button 1" , but i don't know, if it's really connected to GPIO08 - i was only testing it with multimeter 

Serial log is in issue, it repeats several times and then it reboots to safe mode

brandond, 2019-12-02:
You don't get a stack trace or anything, just the state messages and then back to the ets boot message?

Elkropac, 2019-12-02:
I think, i did not get any more messages on usb port. I will try again in the evening.

I will also try harder to identify the right pin connected to button1 header. (I don't see it well, it's too tiny :)

WDT reset is watchdog reset? Could it be caused because GPIO08 is connected to something which causes resets?

Could it be somehow related to: 
https://community.home-assistant.io/t/bricked-sonoff-devices-after-upgrade-to-1-14-2-1-14-3-the-solution-is-to-roll-back-to-1-13-6/148076

this thread?
I also managed to brick another device when probing GPIO ports
https://community.home-assistant.io/t/avatto-smart-plug-flashing/151938

OttoWinter, 2019-12-02:
GPIO8 is used by the flash interface on the sonoff dual r2, that's why it crashes. ESPHome also prints a warning about this when compiling.

Please see the generic sonoff page for the correct pin mapping - you should use GPIO10 instead.

closed 2019-12-02 by OttoWinter

Elkropac, 2019-12-02:
Hello. I have revision 1.4 of board and it has different pinout. It has physical button which is connected to GPIO10. But "button 0" on header is not connected to this pin (when i connect it with ground, relays don't click)

Elkropac, 2019-12-02:
You mean this warning ?
```
WARNING ESP8266: Pin 10 (6-11) might already be used by the flash interface. Be warned.
WARNING ESP8266: Pin 9 (6-11) might already be used by the flash interface. Be warned.
```
Yeah, i saw it and ignored it

Elkropac, 2019-12-02:
Ok, i tested it now 
 - GPIO10 - physical button
 - GPIO00 - button 0 pin on extension header
 - GPIO09 - button 1 pin on extension header

i was looking on some strangely labeled picture of ESP8266 yesterday