esphome/issues #892: ADS1115 needs locking?

balrog-kun, 2019-12-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip3

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32, lolin32

**Affected component:**

https://esphome.io/components/sensor/ads1115.html
https://esphome.io/components/sensor/ct_clamp.html

**Description of problem:**
I'm trying to measure current in 4 wires using 4 current transformers connected to ADS1115 inputs.  I configured the ct_clamp and ads1115 components.  I was getting either continuous ```[W][ads1115:116]: Reading ADS1115 timed out``` messages from boot-up, or if a conversion succeeded, the values would be wrong.  I read the code.  I think there may be two issues but this is the first time I'm using esphome or reading its code so I may be wrong.

Potential issue 1 is that there's no locking and with 4 sensors configured on top of the ads1115 component  the sensors may be interrupting each other's conversions, whether in continuous_mode or not.  This is based on the assumption that there's some multitasking in esphome which there seems to be.  There may be some locking implied by using PollingComponent in ct_clamp or somewhere else I don't know of, this and other facts would influence whether it's better if the locking is done at the ads1115 level or ct_clamp level, or both.

Potential issue 2 is that in ADS1115Component::request_measurement() we're waiting for CONFIG register's bit 15 to become 1 after a conversion but in continuous_mode, by my reading of the spec, this may never happen.  That loop will also end if the read_byte_16() of that register returns an error by the way.

An issue/feature-request 3 would be that in continuous mode we could, at least partially, take advantage of the latest conversion result already being there to avoid the delay(2).  With multiple sensors+ct_clamp components this would obviously require that each ct_clamp instance is able to exclusively use the ads1115 components for its sampling period.

**Problem-relevant YAML-configuration entries:**
```yaml
i2c:
  sda: 21
  scl: 22
  scan: True
  frequency: 100kHz

ads1115:
  address: 0x48
  continuous_mode: on
  id: ads1115_48

sensor:
  - name: "ADS1115 Channel A0-GND"
    id: ads1115_a0
    ads1115_id: ads1115_48
    multiplexer: 'A0_GND'
    gain: 6.144
    platform: ads1115
  - name: "ADS1115 Channel A1-GND"
    id: ads1115_a1
    ads1115_id: ads1115_48
    multiplexer: 'A1_GND'
    gain: 6.144
    platform: ads1115
  - name: "ADS1115 Channel A2-GND"
    id: ads1115_a2
    ads1115_id: ads1115_48
    multiplexer: 'A2_GND'
    gain: 6.144
    platform: ads1115
  - name: "ADS1115 Channel A3-GND"
    id: ads1115_a3
    ads1115_id: ads1115_48
    multiplexer: 'A3_GND'
    gain: 6.144
    platform: ads1115
  - name: "Dormitory 2 current"
    platform: ct_clamp
    sensor: ads1115_a0
    sample_duration: 200ms
    filters:
      calibrate_linear:
        # The sensor should measure 2.5V + I / 2500 * 220 / 1.44 ?
        - 0 -> 0
        - 2.3328 -> 15
  - name: "Living room socket 1 current"
    platform: ct_clamp
    sensor: ads1115_a1
    sample_duration: 200ms
    filters:
      calibrate_linear:
        # The sensor should measure 2.5V + I / 2500 * 220 / 1.44 ?
        - 0 -> 0
        - 2.3328 -> 15
...
```

**Logs (if applicable):**

**Additional information and things you've tried:**

OttoWinter, 2019-12-04:
As for 1)

No, that's not an issue. Timeouts/intervals are set on the component *instance*, so each ADS1115 is separate.

For 2)

Hm, potentially true. But we need to wait for *something* when changing the MUX, otherwise we're just reading old data.
Thinking about this more, it appears like the ads1115 can't be used in continuous mode when multiple settings from the MUX are used. Unless the ADS1115 has some registers for fast MUX switching.

3)

That's already the case when you have at most one MUX setting selected

balrog-kun, 2019-12-05:
Thanks for taking the time to reply!

> As for 1)
> 
> No, that's not an issue. Timeouts/intervals are set on the component instance, so each ADS1115 is separate.

The symptoms I have right now is that in single-shot mode I can mostly use four ct_clamp components but in continuous_mode I get these read timeout errors.  I guess this simply means the ADS1115 is not asserting that OS bit when in continuous mode.

But I also think in continuous mode the ADS1115 is automatically starting a new unneeded conversion on channel X after each conversion on channel X request by a .sample() call.  When another ct_clamp instance calls .sample() for channel Y, it will have to wait for that unneeded conversion on X to finish before ADS1115 starts a conversion on Y.

One thing I haven't tried debugging is that, in single-shot mode, the initial offset is way off for 1 or 2 of the 4 ct_clamp instances.  I was kind of thinking that, since there is a capacitor in the recommended AC measurement circuit, the capacitor might just need time to charge after power up but adding a delay() in ct_clamp's setup() didn't help.

> For 2)
> 
> Hm, potentially true. But we need to wait for _something_ when changing the MUX, otherwise we're just reading old data.
> Thinking about this more, it appears like the ads1115 can't be used in continuous mode when multiple settings from the MUX are used.

I can see some posibilities, but they're not all practical:

- whenever a conversion starts on an ads1115 component, note down the timestamp and the channel number (MUX setting).  If a new conversion is requested for the same channel, wait only 1.6s from the conversion start or from the last sample timestamp.  If the next conversion is on a different channel we exit the continuous mode and restart, or could wait 2x1.6s from the conversion start or last sample timestamp.
- use the ALERT/RDY pin, but it would need to be configured as edge interrupt source, I'm not sure if this can be done on both ESP8266 and ESP32... and also adds more burden for users by requiring another pin connection.
- maybe try writing all 0s to the CONVERSION register after reading a sample, so that we can find out when it gets overwritten by the chip?  The register is documented as read-only so this may be impossible, I'll probably test it later.

> Unless the ADS1115 has some registers for fast MUX switching.
> 
> 3.
>
> That's already the case when you have at most one MUX setting selected

Ideally a ct_clamp instance could mark the ads1115 as busy until its done with all its conversions but I'm not sure what the best API is for that.  Another ct_clamp instances would not start their measurements (would just exit loop()) until the sensor is released.  Ideally ct_clamp could detect whether its this->sensor_ implements that specific API so that not every sensor has to implement this, not sure what the best way to do this is.

OttoWinter, 2019-12-05:
> the initial offset is way off for 1 or 2 of the 4 ct_clamp instances

That is interesting - the initial offset is calculated by averaging the data over one `sample_duration` interval. So there should not be major error there. When did you last test that? The behavior changed in 1.14

For 2)

No, those are all not really solutions. The problem is that all of the ct_clamp instances expect a high sampling frequency (couple kHz) while they're in the sampling state. When the ADS1115 needs to constantly switch around config we have to wait for some time (probably in the range of 1-2ms), which makes the sampling much slower.

The solution would be to make ct_clamp instances not sample at the same time, but I do not see a good way of doing that without making the code a complete mess and breaking modularity principle.

balrog-kun, 2019-12-05:
> > the initial offset is way off for 1 or 2 of the 4 ct_clamp instances
> 
> That is interesting - the initial offset is calculated by averaging the data over one `sample_duration` interval. So there should not be major error there. When did you last test that? The behavior changed in 1.14

I'm running 1.14.3, I'll do some more testing to understand this.  For now I hardcoded an initial offset_ close to the expected value, then I let the calibration run.

> For 2)
>
> No, those are all not really solutions. The problem is that all of the ct_clamp instances expect a high sampling frequency (couple kHz) while they're in the sampling state. When the ADS1115 [...]

The most the ads1115 can achieve is 860Hz but the delay(2) limits this to 500Hz and on top of that ct_clamp has to wait for the next loop() call until it starts a new conversion (in single-shot mode) or until it issues the next delay(2) call (in continuous_mode).  That's even before considering multiple ct_clamp instances.  And I think both of these limitations can probably be fixed with some form of locking, while letting other tasks run too.

> The solution would be to make ct_clamp instances not sample at the same time, but I do not see a good way of doing that without making the code a complete mess and breaking modularity principle.

You have the VoltageSampler interface to add a single method, the locking primitives could simlarly be on an interface and could be just a pair of methods such as set_busy/get_busy, get/put, take/release, etc. and a C++ dynamic_cast could be used to see if it is implemented by the sensor.  There may be better ways to do this too.

balrog-kun, 2019-12-16:
So https://github.com/balrog-kun/esphome/commit/e23acc6285e4754d8c976fa29d320b83d80d403b illustrates one idea of doing this, where the ADS1115Component class inherits from SharedComponent and ADS1115Sensor inherits from SharedComponentUser.  CTClamp would then check if the source sensor implements SharedComponentUser and if so, use the request_time API to be assigned a period of time when it can use the ads1115 exclusively.  But, dynamic_cast<> is not supported when compiling with -fno-rtti so this isn't really usable.

Instead I came up with https://github.com/balrog-kun/esphome/commit/d8a97db26d13e8821b9a1d3ec3f0510de0c69ad8 where request_time is added directly to Component and may be overridden by SharedComponentUser.  CTClamp will then unconditionally use request_time and the code actually becomes simpler in some aspects.  The CTClamp changes are in https://github.com/balrog-kun/esphome/commit/cfee7b2d704065a4f2d75ad1a45b46332d75038b

To be honest I didn't notice much of an improvement in my current measurement accuracy here with these changes alone...  I definitely see the values react to the AC load connected but the values will jump between 100% up and 70% down from the expected value sometimes.

One unrelated question I have about CTClampSensor: since it inherits from PollingComponent and never calls set_interval, I understand the interval stays at 0.  Doesn't that mean that the polling is done at the highest possible frequency and the HighFrequencyLoopRequester usage doesn't add anything?

OttoWinter, 2019-12-17:
@balrog-kun Changing the base classes (Component here) for a small change needed for a component is a very big red flag. Any core architectural addition should be driven by use in many scenarios, not a specific hack.

For example, the new field in `Component` would add 4 bytes of space for each Component instance. It may not sound like a lot, but considering it's a base class you can easily loose half a kb of memory (and there's not much memory on ESPs). (and that's just to make this more clear, even if the field were not there it would be a red flag for me)

> since it inherits from PollingComponent and never calls set_interval, I understand the interval stays at 0

That's called from codegen

OT-metrics, 2020-06-09:
Hi 
I'm new in this forum, but I'm using the ESPHome platform for my devices. I now experience the same problem in using the ADS115 for two clamps in differential mode. Connecting one is working fine, connecting to give a kind of random results.
Following the discussion I get the problem and the problem implementing a solution. I have a question on that : would it be possible to assign each mux setup a “timeslot” based on the system time ? so using blocks of  (sample_duration  + delta) and then alinine to the ms of the system clock so in diff mode 2 blocks in single mode 4 blocks or 3 blocks in mixed mode.

closed 2020-12-05 by stale[bot]

duckfullstop, 2021-05-05:
Can we consider reviving this one? I feel like this is a fairly fundamental issue with the ADS1115 module, considering a number of CT clamp readers (such as the popular [mottramlabs 4-channel board](https://github.com/Mottramlabs/4-Channel-Mains-Current-Sensor)) rely on it. I've just hit it myself and I really don't want to have to resort to switching to raw Arduino / MQTT to resolve it.

lanrat, 2021-06-29:
I'm also running into this issue and would like this to be re-visited.