esphome/issues #895: custom component subdirectory not recognize and compile

minsuke, 2019-12-03:
**Describe the problem you have/What new integration you would like**

When I put in custom components which has sub directory, it has a problem to compile 

**Please describe your use case for this integration and alternatives you've tried:**

I tried another pip install esphome machine then I copied custom components(include sub components) to ESPHOME main component folder then it can be compiled.  

**Additional context**

docker version has no way to compile so I installed another machine, then I called integrates ESPHOME node, then I assaigned IP,  it works. 

If custom components which includes sub component can't be compiled or have an error in hassio and docker version, please apply this error. 

I always thanks to you Otto and developers.

OttoWinter, 2019-12-03:
I'm not sure I understand. You mean you placed folders in the `custom_components` folder and they are not being recognized?

Could you post:
- what folder structures you tried, something like

```
config_dir/
  custom_components/
    my_custom_component/
      __init__.py
      something.h
      something.cpp
  my_config.yaml
```

- what error you're getting when using the custom component
- if possible, the code for the custom component as a link

minsuke, 2019-12-11:
> I'm not sure I understand. You mean you placed folders in the `custom_components` folder and they are not being recognized?
>
> Could you post:
>
> * what folder structures you tried, something like
>
> ``` [...]

Thank you for your reply 

I found your check so late. 

I explain my situation. 

1. my community's developer shared a custom components for my commuty members. 

  https://github.com/greays/esphome/tree/master/esphome/components/rs485

2. downloaded in /config/esphome/custom_components/rs485 

 then I compiled but it was failed in Dashboard. 

 can't compile can't find subdirectory components. 

----------------------------------------------------
solution. 

so I installed another machine then installed the components as develop version. 

then compiled it was successful. 

thanks for your interests.

closed 2020-08-07 by stale[bot]