esphome/issues #896: WDT timeout during dallas component setup

brandond, 2019-12-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
pip/venv

**ESP (ESP32/ESP8266, Board/Sonoff):**
wemos esp32

**Affected component:**
dallas

**Description of problem:**
I've connected the device to an old Dallas 1-Wire Weather station, which is an anemometer/wind direction/temperature sensor combo. You can find some documentation on it here: https://www.curioustimo.nl/2016/04/10/mqtt-retrofitted-onewire-weather-station/

There are three Dallas chips on the bus:
* DS18S20 (temperature)
* DS2423 (counter - wind speed)
* DS2450 (quad A/D converter - wind direction)

After a couple quick panics (usually 2 or 3) it seems to boot fine, but it doesn't pick up any devices other than the temperature sensor. I know they're not supported, but that's what I was going to work on.

**Problem-relevant YAML-configuration entries:**
```yaml
dallas:
  - id: dallas0
    update_interval: 10s
    pin: 13
```

**Logs (if applicable):**
```
[01:19:50][I][logger:156]: Log initialized
[01:19:50][C][ota:364]: There have been 2 suspected unsuccessful boot attempts.
[01:19:50][I][app:028]: Running through setup()...
[01:19:50][C][dallas.sensor:032]: Setting up DallasComponent...
[01:19:50]Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
[01:19:50]Core 1 register dump:
[01:19:50]PC      : 0x4008420c  PS      : 0x00060534  A0      : 0x80083ffe  A1      : 0x3ffb12f0
WARNING Decoded 0x4008420c: timer_overflow_happened at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/esp_timer_esp32.c:340
 (inlined by) esp_timer_impl_get_time at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/esp_timer_esp32.c:200
[01:19:50]A2      : 0x01a5d624  A3      : 0x00000000  A4      : 0x08000000  A5      : 0x00000000
[01:19:50]A6      : 0xffffffff  A7      : 0x3ffb1420  A8      : 0x00000000  A9      : 0x00000000
[01:19:50]A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000050  A13     : 0x3ffb2c24
[01:19:50]A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000013  EXCCAUSE: 0x00000006
[01:19:50]EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff
[01:19:50]
[01:19:50]Backtrace: 0x4008420c:0x3ffb12f0 0x40083ffb:0x3ffb1310 0x40081c8b:0x3ffb1330 0x40081ccb:0x3ffb1350 0x400d3341:0x3ffb1370 0x400d342e:0x3ffb1390 0x400d3513:0x3ffb13c0 0x400d2fdd:0x3ffb13f0 0x400debe6:0x3ffb1440 0x40168795:0x3ffb1490 0x400de9c5:0x3ffb14b0 0x400e199e:0x3ffb1500 0x400ee333:0x3ffb1fb0 0x400891c5:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4008420c: timer_overflow_happened at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/esp_timer_esp32.c:340
 (inlined by) esp_timer_impl_get_time at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/esp_timer_esp32.c:200
WARNING Decoded 0x40083ffb: esp_timer_get_time at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/esp_timer.c:443
WARNING Decoded 0x40081c8b: micros
WARNING Decoded 0x40081ccb: delayMicroseconds
WARNING Decoded 0x400d3341: esphome::dallas::ESPOneWire::read_bit()
WARNING Decoded 0x400d342e: esphome::dallas::ESPOneWire::search()
WARNING Decoded 0x400d3513: esphome::dallas::ESPOneWire::search_vec()
WARNING Decoded 0x400d2fdd: esphome::dallas::DallasComponent::setup()
WARNING Decoded 0x400debe6: esphome::PollingComponent::call_setup()
WARNING Decoded 0x40168795: esphome::Component::call()
WARNING Decoded 0x400de9c5: esphome::Application::setup()
WARNING Decoded 0x400e199e: setup()
WARNING Decoded 0x400ee333: loopTask(void*)
WARNING Decoded 0x400891c5: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c:355 (discriminator 1)
[01:19:51]
[01:19:51]Core 0 register dump:
[01:19:51]PC      : 0x401695e6  PS      : 0x00060134  A0      : 0x800fce5a  A1      : 0x3ffbc0d0
WARNING Decoded 0x401695e6: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/pm_esp32.c:487
[01:19:51]A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000001
[01:19:51]A6      : 0x00060120  A7      : 0x00000000  A8      : 0x800fed5a  A9      : 0x3ffbc0a0
[01:19:51]A10     : 0x00000000  A11     : 0x40086cf8  A12     : 0x00060120  A13     : 0x3ffbba40
[01:19:51]A14     : 0x00000000  A15     : 0x3ffbbdc0  SAR     : 0x00000000  EXCCAUSE: 0x00000006
[01:19:51]EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
[... 62 lines cut ...]
WARNING Decoded 0x401695e6: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/pm_esp32.c:487
WARNING Decoded 0x400fce57: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/freertos_hooks.c:108
WARNING Decoded 0x4008b0c5: prvIdleTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c:3537
WARNING Decoded 0x400891c5: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c:355 (discriminator 1)
[01:19:53]
[01:19:53]Rebooting...
[01:19:53]ets Jun  8 2016 00:22:57
[01:19:53]
[01:19:53]rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
[01:19:53]configsip: 0, SPIWP:0xee
[01:19:53]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[01:19:53]mode:DIO, clock div:2
[01:19:53]load:0x3fff0018,len:4
[01:19:53]load:0x3fff001c,len:952
[01:19:53]load:0x40078000,len:6084
[01:19:53]load:0x40080000,len:7944
[01:19:53]entry 0x40080310
[01:19:53][I][logger:156]: Log initialized
[01:19:53][C][ota:364]: There have been 4 suspected unsuccessful boot attempts.
[01:19:53][I][app:028]: Running through setup()...
[01:19:53][C][dallas.sensor:032]: Setting up DallasComponent...
[01:19:53][W][dallas.sensor:204]: DS18S20 doesn't support setting resolution.
[01:19:53][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[01:19:53][C][wifi:033]: Setting up WiFi...
```

OttoWinter, 2019-12-04:
Hmm, it's crashing during the search phase.

During the search phase interrupts are disabled (to get more accurate timing).

The interrupt WDT probably means that the search phase is taking a very long time. I think the default WDT timeout is 1-2 seconds, does the device hang before the WDT?

Potentially one of the devices on the bus has some weird onewire search characteristics and is returning true for all branches - making the search take a long time.

brandond, 2019-12-04:
It doesn't hang long enough for me to notice. It reboots several times in 2-3 seconds and then comes up normally.

brandond, 2019-12-04:
I saw you changed some stuff in the same commit that added support for new esp8266 Arduino core. Gave it a try and now it does something new, at least. Hangs for about 22 seconds, prints a bunch of 'unknown device type' errors (WAY more than there should be given just 3 devices on the bus) and then eventually gets whacked by the WDT while printing.

Unlike before, it does not seem to ever make it past this and set up the component successfully - it boot loops until it goes into safe mode.

```
[14:44:36]ets Jun  8 2016 00:22:57
[14:44:36]
[14:44:36]rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
[14:44:36]configsip: 0, SPIWP:0xee
[14:44:36]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x0                                                                                                                                          0
[14:44:36]mode:DIO, clock div:2
[14:44:36]load:0x3fff0018,len:4
[14:44:36]load:0x3fff001c,len:952
[14:44:36]load:0x40078000,len:6084
[14:44:36]load:0x40080000,len:7944
[14:44:36]entry 0x40080310
[14:44:36][I][logger:156]: Log initialized
[14:44:36][C][ota:364]: There have been 3 suspected unsuccessful boot attempts.
[14:44:36][I][app:028]: Running through setup()...
[14:44:36][C][dallas.sensor:032]: Setting up DallasComponent...
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[14:44:58][W][dallas.sensor:051]: Unknown device type 0x20.
[... 29 lines cut ...]
WARNING Decoded 0x400de5ae: esphome::PollingComponent::call_setup() at /mnt/media/Repos/hass-esphome/nodes/esp32/fermenter00/src/esphome/core/component.cpp:180
WARNING Decoded 0x40167a19: esphome::Component::call() at /mnt/media/Repos/hass-esphome/nodes/esp32/fermenter00/src/esphome/core/component.cpp:82
WARNING Decoded 0x400de359: esphome::Application::setup() at /mnt/media/Repos/hass-esphome/nodes/esp32/fermenter00/src/esphome/core/application.cpp:37
WARNING Decoded 0x400e145e: setup() at /mnt/media/Repos/hass-esphome/nodes/esp32/fermenter00/src/main.cpp:398
WARNING Decoded 0x400eed9f: loopTask(void*) at /mnt/media/Home/brandond/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:14
WARNING Decoded 0x40089955: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[14:44:59]
[14:44:59]Core 0 register dump:
[14:44:59]PC      : 0x40168452  PS      : 0x00060134  A0      : 0x800f2019  A1      : 0x3ffbc050
[... cut ...]
WARNING Decoded 0x400f2016: esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c:108
WARNING Decoded 0x4008b149: prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
WARNING Decoded 0x40089955: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[14:44:59]
[14:44:59]Rebooting...
```

OttoWinter, 2019-12-05:
Whoops yeah there was an error in the commit (for ESP32, the new InterruptLock would enable interrupts at the start, and disable them after exiting the lock; so the wrong way around; only tested on ESP8266 but should have also done so on ESP32). Anyway, fixed now.

The logs however confirm my suspicion, at least one of the one-wire devices on your bus is spewing out lots of addresses (or does not support the 1-wire address search protocol and returns true for each branching decision).

It may help to add some more info to the "Unknown device type" line in your local install to make it spit out the entire address, not just the first octet.

closed 2020-12-05 by stale[bot]