esphome/issues #897: Baud rate doesn't fall back to 115200 if 460800 fails

aalbul, 2019-12-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP32

**Affected component:**

**Description of problem:**
I'm using 1.14.3 and i saw this fix: https://github.com/esphome/esphome/pull/856
Unfortunately, it doesn't work for me. Please see logs.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/night_light_2.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/night_light_2
Processing night_light_2 (board: esp-wrover-kit; framework: arduino; platform: espressif32@1.11.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <AsyncTCP-esphome> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling /data/night_light_2/.pioenvs/night_light_2/src/main.cpp.o
Linking /data/night_light_2/.pioenvs/night_light_2/firmware.elf
Retrieving maximum program size /data/night_light_2/.pioenvs/night_light_2/firmware.elf
Checking size /data/night_light_2/.pioenvs/night_light_2/firmware.elf
Building /data/night_light_2/.pioenvs/night_light_2/firmware.bin
DATA:    [=         ]  13.0% (used 42640 bytes from 327680 bytes)
PROGRAM: [======    ]  56.5% (used 925494 bytes from 1638400 bytes)
========================= [SUCCESS] Took 79.28 seconds =========================
INFO Successfully compiled program.
INFO Running:  platformio run -d /config/esphome/night_light_2 -t upload --upload-port /dev/ttyUSB0
Processing night_light_2 (board: esp-wrover-kit; framework: arduino; platform: espressif32@1.11.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <AsyncTCP-esphome> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Retrieving maximum program size /data/night_light_2/.pioenvs/night_light_2/firmware.elf
Checking size /data/night_light_2/.pioenvs/night_light_2/firmware.elf
DATA:    [=         ]  13.0% (used 42640 bytes from 327680 bytes)
PROGRAM: [======    ]  56.5% (used 925494 bytes from 1638400 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, ftdi, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: /dev/ttyUSB0
Uploading /data/night_light_2/.pioenvs/night_light_2/firmware.bin
Serial port /dev/ttyUSB0
Connecting........_
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:6f:28:24:73:80
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 15872 bytes to 10319...

A fatal error occurred: Timed out waiting for packet content
*** [upload] Error 2
========================= [FAILED] Took 20.86 seconds =========================

```

**Additional information and things you've tried:**

brandond, 2019-12-04:
I think @OttoWinter mentioned that the fix in 1.14.3 only applied to esp8266. Have you tried the dev branch?

aalbul, 2019-12-05:
Hi, i didn't try that on dev branch as i don't know exactly how to go back once i move to dev.
I had some bad experience trying to downgrade to 1.13x from 1.14x so i decided not to take a risk :)

OttoWinter, 2019-12-05:
Duplicate of https://github.com/esphome/issues/issues/879

@brandond It's not been fixed yet :)

closed 2019-12-05 by OttoWinter